openwrt/packages #3247: cshark - if possible build with libustream-openssl ?

build000, 2016-09-28:
The question contained in the notification or reporting a bug related to the recent changes in opkg scripts and building (LEDE) - whether it makes sense depends on whether it is possible to build cshark and appropriate trap Lucia to the needs of openssl.
https://github.com/openwrt/luci/issues/806
https://bugs.lede-project.org/index.php?do=details&task_id=189

hnyman, 2016-09-28:
By looking at the cshark Makefile, I think that for your own build you could just edit the cshark Makefile and depend on libustream-openssl instead. (libustream is the wrapper hiding the actual SSL library from the app.)

BlingRabbit, 2016-11-08:
https://bugs.lede-project.org/index.php?do=details&task_id=189
i also have this problem, even i change libustream-openssl is not set and libustream-polarssl is not set
but i saw lede make it close so i want to know hot can it fix?
Thx

hnyman, 2016-11-08:
> i also have this problem, even i change libustream-openssl is not set and libustream-polarssl is not set
but i saw lede make it close so i want to know hot can it fix?

@BlingRabbit 
Based on your above sentence, I am not quite sure what your problem is and what your goal is...

cshark explicitly requires "libustream-polarssl": https://github.com/openwrt/packages/blob/master/net/cshark/Makefile#L31 .  Until somebody modifies cshark package to work with other SSL libraries, you need that "libustream-polarssl" for cshark.  But like I said earlier in this thread, I think that you can also edit the cshark Makefile in your own repo and switch it to use another SSL libustream library so that your build gets built with it.

LEDE issue 189 discussed mainly other things, conflicting dependencies in general. 

In any case, you should make sure that you have not selected two different libustream- variants at the same time on .config. Dependencies to them may be caused by various packages. so you need to go through your selections and check what causes that selection. (E.g. "luci-ssl" selects luci, px5g and libustream-polarssl. If you want to get rid of libustream-polarssl, select "luci-ssl-openssl" instead, or manually select luci, px5g and libustream-openssl) Note that de-selecting a package does not deselect its dependent packages, so the easy clean solution is to start from an empty .config, or alternatively go through menuconfig and manually de-select the unneeded packages)

build000, 2016-11-09:
@hnyman 
...and until all possible packages / programs do not arise as an option in "make menuconfig" the ability to choose which interests us wersjji libustreem- * will still be a similar problem - as mentioned earlier - this is a free software for the appropriate license GPL2 unless otherwise stated the individual library programs and indeed should be able to choose packages/libraries dependent on the level of compilation. Besides, it is simply a convenience for everyone Useful Links SDK/sources - without the need to resort to tricks described by you (eg. Manual modification of the Makefile in the cshark).
I think that over time, developers LEDE digest this topic in depth, and there will be corresponding changes in the relevant libraries.
Besides, practically speaking - anyone who touches the subject's own compilation certainly is (or at least should be) aware at how powerful, high-performance architecture puts LEDE and what is the cost to use (size, computational requirements, etc.) openssl or polarssl - the choice is even indicated if only for this reason.

BlingRabbit, 2016-11-10:
@hnyman ThankYou！
i just want to compile this bin. Fix this problem to get process to continue.....

Now this mean ssl and polarssl can not coexistence? if i cancel ssl or polarssl,the problem is fixed?

Thx

hnyman, 2016-11-10:
@BlingRabbit 
libustream-polarssl and libustream-openssl (or libustream-mbedtls or libustream-cyassl) can't be installed at the same time. Make sure that you have just one of them in your build, and then edit cshark Makefile to need that libustream variant.

Note that you can have the actual libopenssl, libpolarssl and libmbedtls SSL libraries at the same time in the build. It is just the libustream-... wrapper layer that creates the complication.

@build000 
Where did you get GPL2 into discussion? Being free/allowed to do something is different than being actually able to do that. You can use any SSL library as long at somebody has written support for that SSL library for the app you need the library for. That somebody could be you, as the packages are developed/fixed/changed by the community. If you need cshark-openssl variant, feel free to create that variant and submit a Pull request for the maintainer's review ( @lperkov ) to get it accepted into the source repo.

There are so many apps that require some SSL layer, many apps require a certain SSL library etc., so nobody has energy to enable all SSL libraries for all apps. But feel free to do your part in the effort.

closed 2018-08-18 by dibdot