openwrt/packages #3248: 6in4: tunnel enpoint update is totally broken

paulcarroty, 2016-09-28:
```
# logread | grep -i 6in4
Tue Sep 27 15:42:37 2016 daemon.notice netifd: IP tunnel '6in4-wan6' link is up
Tue Sep 27 15:42:38 2016 user.notice 6in4-wan6: update 1/3: uclient-fetch: local: line 2: not in a function
Tue Sep 27 15:42:38 2016 user.notice 6in4-wan6: update 1/3: timeout
Tue Sep 27 15:42:38 2016 user.notice firewall: Reloading firewall due to ifup of wan6 (6in4-wan6)
Tue Sep 27 15:42:43 2016 user.notice 6in4-wan6: update 2/3: uclient-fetch: local: line 2: not in a function
Tue Sep 27 15:42:43 2016 user.notice 6in4-wan6: update 2/3: timeout
Wed Sep 28 12:52:03 2016 user.notice 6in4-wan6: update 3/3: uclient-fetch: local: line 2: not in a function
Wed Sep 28 12:52:03 2016 user.notice 6in4-wan6: update 3/3: timeout
Wed Sep 28 12:52:08 2016 user.notice 6in4-wan6: update failed
```

Tested on latest ar71xx from 26.09 and 28.09. Stable releases is also affected because have the same bad URL format:

* https://forums.he.net/index.php?topic=1994.0
* https://dev.openwrt.org/browser/trunk/package/network/ipv6/6in4/files/6in4.sh#L94

L94 need to be changed to 

```
local url="http://ipv4.tunnelbroker.net/nic/update?username=$username&password=$password&hostname=$tunnelid"
```
Also (according to logs) problem with including functions.

paulcarroty, 2016-09-29:
Cast mantainer - @jow-

jow-, 2016-09-29:
@openwrt/owners need to merge https://github.com/lede-project/source/commit/864b2d113ad0815c2444575014e7cf51498944db

hnyman, 2016-09-29:
Not quite sure about that diagnosis, as Openwrt is not yet using Busybox 1.25.0. (I guess that the OP has a LEDE build compiled just before your fix.)

But I noticed that updating he.net may have timeout problems. On my two tries it picked the update only on the second try.:
```
Thu Sep 29 16:12:35 2016 user.notice 6in4-henet: update 1/3: timeout
Thu Sep 29 16:12:45 2016 user.notice 6in4-henet: update 2/3: nochg 87.92...
Thu Sep 29 16:12:45 2016 user.notice 6in4-henet: updated
...
Thu Sep 29 16:16:43 2016 user.notice 6in4-henet: update 1/3: timeout
Thu Sep 29 16:16:53 2016 user.notice 6in4-henet: update 2/3: nochg 87.92...
Thu Sep 29 16:16:53 2016 user.notice 6in4-henet: updated
```
When I increased the timeout to 8 seconds in ```/lib/netifd/proto/6in4.sh ```, it started updating on the first try. You might consider adding a few seconds to the timeout.

(And yeah, I know that this is the wrong place for a 6in4 discussion.)

hizukiayaka, 2016-09-29:
It won't work without update the link following the first post. I have tried update it manually.

paulcarroty, 2016-09-29:
@hnyman @jow- I try the latest `6in4.sh` from [LEDE](https://github.com/lede-project/source/blob/master/package/network/ipv6/6in4/files/6in4.sh) with the `8` timeout and have the same problem.

The URL is broken:

```
local url="$http://ipv4.tunnelbroker.net/nic/update?hostname=$tunnelid"

# wget "http://ipv4.tunnelbroker.net/nic/update?hostname=36X2X5"

Downloading 'http://ipv4.tunnelbroker.net/nic/update?hostname=36X2X5'
Connecting to 64.62.200.2:80
HTTP error 401
```

jow-, 2016-09-29:
You forgot to pass the `--username` and `--password` arguments to wget.

paulcarroty, 2016-09-29:
Yep, with `--username=id` and `--password=updatekey` it works with wget. Now I have no idea what the problem. I'll test it with newest LEDE snapshot.

diizzyy, 2017-06-09:
@paulcarroty 
Works in LEDE?

paulcarroty, 2017-06-09:
@diizzyy yep.

diizzyy, 2017-06-09:
Please close this issue then.

closed 2017-06-09 by paulcarroty