openwrt/packages #3249: mii-tool - bad compile (if not possible - bad path to dir/files)

build000, 2016-09-28:
make -j8 BUILD_LOG=1

cat [sources]/logs/package/feeds/packages/mii-tool/compile.txt:
```
bash: ./configure.sh: No such file or directory
Makefile:53: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/mii-tool-2016-07-10-115f1af2494ded1fcd21c8419d5e289bc4df380f/.configured_yyyyyyy' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/mii-tool-2016-07-10-115f1af2494ded1fcd21c8419d5e289bc4df380f/.configured_yyyyyyy] Error 127
```
in short - the source is extracted to the directory [sources]/build_dir/mii-tool-2016-07-10- and testing and commissioning (and run compile) of files in the source program is done in the [sources]/build_dir/mii-tool-2016-07- 10-115f1af2494ded1fcd21c8419d5e289bc.
It seems obvious syntax error in the file https://github.com/openwrt/packages/blob/master/net/mii-tool/Makefile.

hnyman, 2016-09-28:
cc to maintainer:  @Borromini 

Builtbot has picked an error, too: 
https://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/mii-tool/compile.txt

build000, 2016-09-28:
@hnyman:
I may lay the question but still are sending me to certain persons by them messages:
1) I do not know their e-mail or other forms of contact as long as you do not give in profile
2) writing on the mailing list dev-openwrt ... well - maybe I'm actually a layman - but I prefer to write directly - on github I can not write directly to the people or have not yet discovered (too little'm familiar with github.com)
Therefore, unless existing form of the "issue" for me it is the only and fully sufficient form of contact - if anyone actually interested in what intercede for repo packages (openwrt, Lede) this assumption probably should be interested ...
Like the others do not have the time to read the notification for errors in programs (their own ... good ... hehe) so I do not have so much jam to seek contact with them ...

hnyman, 2016-09-28:
@build000 
I am not sure if I understood what you are asking, but in Github the easiest way to find a person (e.g. a package's maintainer) is to look at the package's commit history and/or the PR/issue history. That tells you pretty quickly who has been active with the package and who has done the last commit possibly breaking things.

(Borromini's mail address in the git commit message does apparently not match the address related to his Github account, so Github does not show his username normally at the commit in the package history. Finding him requires looking at the PR, which is visible in the package history.)

Filing just "compile error" issues without using @ to inform the package maintainer is pretty inefficient. Most maintainers do not get automatically messages related to issues or PRs. They need to get attached to the discussion first, by using @ to target them. (People having with write access here (like me) do get the messages, but not the regular package maintainers.)

So, please spend a minute more to find out the maintainer before filing an issue.

build000, 2016-09-28:
@hnyman:
ok - I'll try to apply to your recommendations, although I do not think it would change anything in practice.

Borromini, 2016-09-28:
Hnyman put me in copy so I'm aware. Looking into it.

build000, 2016-09-28:
@Borromini 
like this #3249: its same if last LEDE r1710 -still the same error.

PR #3263 (2016-09-29): mii-tool: fix variable being declared too early
PKG_SOURCE_SUBDIR was being declared too early which broke the build. Fixes #3249
closed 2016-09-29 by hnyman