openwrt/packages #3251: netdata - new version/changes - program error build/compile

build000, 2016-09-28:
make -j8 BUILD_LOG=1

cat [sources]/logs/package/feed/packages/netdata/compile.txt:

```
make[4]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0'
make  all-recursive
make[5]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0'
Making all in charts.d
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/charts.d'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/charts.d'
Making all in conf.d
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/conf.d'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/conf.d'
Making all in node.d
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/node.d'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/node.d'
Making all in plugins.d
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/plugins.d'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/plugins.d'
Making all in python.d
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/python.d'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/python.d'
Making all in src
make[6]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/src'
powerpc-openwrt-linux-musl-gcc      -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0:netdata-1.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/lib -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro  -o netdata appconfig.o avl.o common.o daemon.o dictionary.o eval.o global_statistics.o health.o log.o main.o plugin_checks.o plugin_idlejitter.o plugin_nfacct.o plugin_proc.o plugin_tc.o plugins_d.o popen.o proc_diskstats.o proc_interrupts.o proc_softirqs.o proc_loadavg.o proc_meminfo.o proc_net_dev.o proc_net_ip_vs_stats.o proc_net_netstat.o proc_net_rpc_nfsd.o proc_net_snmp.o proc_net_snmp6.o proc_net_stat_conntrack.o proc_net_stat_synproxy.o proc_stat.o proc_self_mountinfo.o proc_sys_kernel_random_entropy_avail.o proc_vmstat.o sys_kernel_mm_ksm.o sys_fs_cgroup.o procfile.o registry.o rrd.o rrd2json.o storage_number.o unit_test.o url.o web_buffer.o web_buffer_svg.o web_client.o web_server.o -lm  -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib -lz -L/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/lib -luuid  
global_statistics.o: In function `finished_web_request_statistics':
global_statistics.c:(.text+0x9c): undefined reference to `__atomic_compare_exchange_8'
global_statistics.c:(.text+0xc8): undefined reference to `__atomic_fetch_add_8'
global_statistics.c:(.text+0xdc): undefined reference to `__atomic_fetch_add_8'
global_statistics.c:(.text+0xf0): undefined reference to `__atomic_fetch_add_8'
global_statistics.c:(.text+0x104): undefined reference to `__atomic_fetch_add_8'
global_statistics.c:(.text+0x118): undefined reference to `__atomic_fetch_add_8'
global_statistics.o:global_statistics.c:(.text+0x12c): more undefined references to `__atomic_fetch_add_8' follow
global_statistics.o: In function `global_statistics_copy':
global_statistics.c:(.text+0x334): undefined reference to `__atomic_compare_exchange_8'
collect2: error: ld returned 1 exit status
Makefile:575: recipe for target 'netdata' failed
make[6]: *** [netdata] Error 1
make[6]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/src'
Makefile:461: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0'
Makefile:400: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0'
Makefile:60: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/.built] Error 2
```

diizzyy, 2016-09-28:
Does pull request #3255 fix your issue?

build000, 2016-09-28:
confirm, this pull request #3255 fixed this problem.

make -j8 BUILD_LOG=1

cat [sources]/logs/package/feeds/packages/netdata/compile.txt:
```
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/aclocal -I /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/share/aclocal -I /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/autoconf --include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/share/aclocal --include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/share/aclocal --force
autoreconf: running: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/autoheader --include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/share/aclocal --include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/share/aclocal --force
autoreconf: running: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/automake --add-missing --force-missing
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/automake line 3938.
autoreconf: Leaving directory `.'
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /home/sj/Pulpit/Projekty/WNDR4700-LEDE/include/site/powerpc
checking whether to enable maintainer-specific portions of Makefiles... no
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for powerpc-openwrt-linux-strip... powerpc-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... powerpc-openwrt-linux-gnu
checking for powerpc-openwrt-linux-gcc... powerpc-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether powerpc-openwrt-linux-musl-gcc accepts -g... yes
checking for powerpc-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether powerpc-openwrt-linux-musl-gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of powerpc-openwrt-linux-musl-gcc... gcc3
checking pkg-config is at least version 0.9.0... yes
checking how to run the C preprocessor... powerpc-openwrt-linux-musl-gcc -E
checking for grep that handles long lines and -e... /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/grep
checking for egrep... /home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/host/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for accept4... yes
checking if compiler needs -Werror to reject unknown flags... no
checking for the pthreads library -lpthreads... no
checking whether pthreads work without any flags... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking if more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking for uint8_t... yes
checking for uint16_t... yes
checking for uint32_t... yes
checking for inline... inline
checking for sin in -lm... yes
[... 424 lines cut ...]
 /bin/mkdir -p '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-install/usr/share/netdata/web/lib'
 /usr/bin/install -c -m 644 lib/dygraph-combined.js lib/dygraph-smooth-plotter.js lib/jquery-1.12.0.min.js lib/jquery.peity.min.js lib/jquery.sparkline.min.js lib/morris.min.js lib/raphael-min.js lib/jquery.easypiechart.min.js lib/jquery.nanoscroller.min.js lib/bootstrap.min.js lib/ElementQueries.js lib/ResizeSensor.js lib/bootstrap-toggle.min.js lib/c3.min.js lib/d3.min.js lib/gauge.min.js '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-install/usr/share/netdata/web/lib'
 /bin/mkdir -p '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-install/usr/share/netdata/web/old'
[... cut ...]
rstrip.sh: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-powerpc_464fp/netdata/usr/sbin/netdata: executable
rstrip.sh: /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-powerpc_464fp/netdata/usr/lib/netdata/plugins.d/apps.plugin: executable
Packaged contents of /home/sj/Pulpit/Projekty/WNDR4700-LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/netdata-1.3.0/ipkg-powerpc_464fp/netdata into /home/sj/Pulpit/Projekty/WNDR4700-LEDE/bin/packages/powerpc_464fp/packages/netdata_1.3.0-1_powerpc_464fp.ipk
```
issues to by closed.

PR #3258 (2016-10-03): netdata: disable atomic instructions
Maintainer: @nitroshift
Compile tested: powerpc_464fp, Netgear Centria N900 WNDR4700/WNDR4720, LEDE 875cddd94cea002fae1b7eeb20ef2098410c44c1
Run tested: N/A

Description:

The auto detection is a bit dodgy and sometimes fails to identify support, disable atomic instruction for now.

Fixes #3251 and bumps PKG_RELEASE
commit 4b122db8 (2016-10-03): netdata: disable atomic instructions

The auto detection is a bit dodgy and sometimes fails to identify support, disable atomic instruction for now.

Fixes #3251 and bumps PKG_RELEASE
closed 2016-09-28 by build000