openwrt/packages #3257: bind-libs: failed to build for mips for 9.9.9-P3

hizukiayaka, 2016-09-28:
Maintainer: @nmeyerhans
Environment: WNDR3800 mips build on fedora x86_64
openwrt  c96b95cbab50ccaa6a31fad8ca814280e524ff63
Description:
When I try to build bind server as a module, I met the following program. It sounds like some bug in build rules or could be the toolchains?
Only the bind failed, the other base packages build successfully.
```
format -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -W -Wall -Wmissing-prototy[16/1936]
t-qual -Wwrite-strings -Wformat -Wpointer-arith -fno-strict-aliasing -fno-delete-null-pointer-checks  -c stats.c
OpenWrt-libtool: compile:  mips-openwrt-linux-musl-gcc -I/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bi
nd-9.9.9-P3 -I./unix/include -I./nothreads/include -I./mips/include -I./include -I./include -DOPENSSL -D_GNU_SOURCE -Os -pipe -mno
-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unuse
d-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3:bind-9.9.9-P3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -W -Wall 
-Wmissing-prototypes -Wcast-qual -Wwrite-strings -Wformat -Wpointer-arith -fno-strict-aliasing -fno-delete-null-pointer-checks -c 
stats.c  -fPIC -DPIC -o .libs/stats.o
{standard input}: Assembler messages:
{standard input}:641: Error: unrecognized opcode `ll $3,4($2)'
{standard input}:642: Error: unrecognized opcode `add $7,$0,$3'
{standard input}:643: Error: unrecognized opcode `add $3,$3,$6'
{standard input}:644: Error: unrecognized opcode `sc $3,4($2)'
{standard input}:655: Error: unrecognized opcode `ll $3,0($5)'
{standard input}:656: Error: unrecognized opcode `add $2,$0,$3'
{standard input}:657: Error: unrecognized opcode `add $3,$3,$6'
{standard input}:658: Error: unrecognized opcode `sc $3,0($5)'
{standard input}:730: Error: unrecognized opcode `ll $3,4($2)'
{standard input}:731: Error: unrecognized opcode `add $7,$0,$3'
{standard input}:732: Error: unrecognized opcode `add $3,$3,$6'
{standard input}:733: Error: unrecognized opcode `sc $3,4($2)'
{standard input}:742: Error: unrecognized opcode `ll $3,0($5)'
{standard input}:743: Error: unrecognized opcode `add $2,$0,$3'
{standard input}:744: Error: unrecognized opcode `add $3,$3,$6'
{standard input}:745: Error: unrecognized opcode `sc $3,0($5)'
Makefile:241: recipe for target 'stats.lo' failed
make[6]: *** [stats.lo] Error 1
make[6]: Leaving directory '/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3/lib/isc'
Makefile:98: recipe for target 'subdirs' failed
make[5]: *** [subdirs] Error 1
make[5]: Leaving directory '/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3/lib'
Makefile:102: recipe for target 'subdirs' failed
make[4]: *** [subdirs] Error 1
make[4]: Leaving directory '/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3'
Makefile:209: recipe for target '/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3/.built' fail
ed
make[3]: *** [/home/ayaka/src/openwrt/openwrt/build_dir/target-mips_34kc_musl-1.1.15/bind-9.9.9-P3/.built] Error 2

```

nmeyerhans, 2016-09-29:
Confirmed. This will require deeper investigation.

nmeyerhans, 2016-09-29:
The problem occurs when `ISC_STATS_USEMULTIFIELDS` is defined by the preprocessor in stats.c. There's some conditional code that runs when that symbol is defined, which itself is dependent on certain platform features.

This code has also changed in bind 9.10, so it's possible that the issue is no longer present. I have held off on moving to 9.10 in OpenWRT, but it's worth revisiting this decision if the bug is fixed there.

nmeyerhans, 2016-09-29:
The problem is reproducible in bind 9.10.3-p3 as well. Will investigate further and likely follow up with upstream.

jow-, 2016-09-29:
Did you try `PKG_USE_MIPS16:=0` yet?

nmeyerhans, 2016-09-30:
Thanks @jow- that seems to have done it. @hizukiayaka please pull and rebuild.

closed 2016-09-30 by nmeyerhans