openwrt/packages #3268: adblock: 1.4.10 adblock don't work without wget

gendouhydeist, 2016-09-30:
@dibdot 

> adblock[14248] info : all adblock related services stopped
adblock[14394] info : please install 'uclient-fetch' or 'wget' with ssl support to use adblock                                                                                                                                                       
root@lede:~# opkg install uclient-fetch
Package uclient-fetch (2016-07-30-ccb314db2e99463de15eed9d7626360d82aec4d2) installed in root is up to date.

dibdot, 2016-09-30:
not a bug.
Please check the online documentation (https://github.com/openwrt/packages/blob/master/net/adblock/files/README.md#prerequisites) ... the default default ustream ssl backend 'libustream-polarssl' is not supported, switch to mbedtls or openssl backend.

thess, 2016-09-30:
As Dirk says...

closed 2016-09-30 by thess