openwrt/packages #3271: ttyd listens by default on tcp 7681 on ALL interfaces, IPv4/IPv6, without any firewall rules.

Aditza2015, 2016-10-01:
Maintainer: @tsl0922

i installed LEDE r1725 built with ttyd and... surprise, surprise... discovered it listens on ALL interfaces by default, with no firewall rules defined and it doesn't even use HTTPS/TLS encryption... all data is transferred via plain http.. it's basically a telnet server with a web interface.

please make it work only via https TLS 1.2/1.3, without http and change it so it only listens only on local-network and only on IPv4 reserved local addresses.

`# netstat -a -n -p | grep -i tty
tcp        0      0 :::7681                 :::*                    LISTEN      8062/ttyd
tcp        0      0 2a02:2f0d:__REMOVED_THIS___:7681 2a02:2f0d:___AND_THIS___:56195 ESTABLISHED 8062/ttyd`

and there are no dedicated firewall rules defined for it:
`# iptables-save | grep 7681
#`

`# iptables-save  | grep -i tty
#`

Package: ttyd
Version: 1.0.0-1
Depends: libc, libopenssl, libjson-c, libpthread, libwebsockets-openssl
Source: feeds/packages/utils/ttyd
License: MIT
LicenseFiles: LICENSE
Section: utils
Maintainer: Shuanglei Tao <tsl0922@__removed__>
Architecture: mips_24kc
Installed-Size: 113122
Description:  ttyd is a command-line tool for sharing terminal over the web.

tsl0922, 2016-10-01:
@Aditza2015 Thanks for pointing out this 👍 

 ttyd can accept command line options to make itself secure, so I'd like to remove the default init script and let users to configure it manually or provide a luci interface. What do you think?

Aditza2015, 2016-10-01:
sounds much better this way

tsl0922, 2016-10-02:
@Aditza2015 The insecure init script is removed, can we close this issue now?

Aditza2015, 2016-10-02:
yes, thank you.

closed 2016-10-02 by hnyman