openwrt/packages #3273: Reaver : "WARNING: Failed to associate with xx:xx:xx:xx:xx"

tomudo, 2016-10-01:
When run reaver by command "reaver -i mon0 -b xx:xx:xx:xx -vv",it continously print error line message, "WARNING: Failed to associate with xx:xx:xx:xx:xx".  

The reaver works on openwrt "Barrier breaker".

Testing router is TL-MR3020 .

yousong, 2016-10-02:
Are you testing in a controlled environment?  It's possible that the target has intrusion detection and turned down the assoc req.

tomudo, 2016-10-03:
The controlled env. is 
- Router(Reaver package installed): TP-Link TL-MR3020 
- Firmware: Barrier breaker r48549 / Lede r1771
- AP: WPA2 / D-Link 600 router / signal strenght about 50-60 dBm

yousong, 2017-01-09:
@tomudo , can you give the answer at link [1] a try?

[1] http://unix.stackexchange.com/questions/133974/how-to-associate-using-aireplay-ng

diizzyy, 2017-06-13:
@thess @hnyman 
Please close this, reporter timeout.

closed 2017-06-13 by yousong