openwrt/packages #3292: ddns forces updating forever since last updates

duvi, 2016-10-03:
Since the update to ddns-scripts a couple of days ago, an update is forced every 10 minutes, even though the IP does not change.

```
root@OpenWrt:~# cat /etc/config/ddns
config ddns 'global'
        option date_format '%F %R'
        option log_lines '250'
        option allow_local_ip '0'

config service 'myddns_ipv4'
        option enabled '1'
        option service_name 'changeip.com'
        option domain 'mydns.changeip.net'
        option username 'user'
        option password 'pass'
        option interface 'wan'
        option ip_source 'network'
        option ip_network 'wan'
        option retry_count '0'
        option lookup_host 'mydns.changeip.net'
```
```
root@OpenWrt:~# cat /var/log/ddns/myddns_ipv4.log
 224837       : Detect registered/public IP
 224837       : #> /usr/bin/nslookup mydns.changeip.net  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 224837       : Registered IP '89.133.69.82' detected
 224837  info : Rerun IP check at 2016-10-03 22:48
 224837       : Detect local IP on 'network'
 224837       : Local IP '89.133.69.82' detected on network 'wan'
 224837       : Forced Update - L: '89.133.69.82' == R: '89.133.69.82'
 224837       : #> /bin/uclient-fetch -q -O /var/run/ddns/myddns_ipv4.dat -Y off 'http://user:pass@nic.changeip.com/nic/update?u=user&p=pass&cmd=update&hostname=mydns.changeip.net&ip=89.133.69.82' 2>/var/run/ddns/myddns_ipv4.err
 224838       : DDNS Provider answered:
200 Successful Update
 224838 ERROR : IP update not accepted by DDNS Provider
 224838       : Waiting 600 seconds (Check Interval)
 225838       : Detect registered/public IP
 225838       : #> /usr/bin/nslookup mydns.changeip.net  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 225838       : Registered IP '89.133.69.82' detected
 225838  info : Rerun IP check at 2016-10-03 22:58
 225838       : Detect local IP on 'network'
 225838       : Local IP '89.133.69.82' detected on network 'wan'
 225839       : Forced Update - L: '89.133.69.82' == R: '89.133.69.82'
 225839       : #> /bin/uclient-fetch -q -O /var/run/ddns/myddns_ipv4.dat -Y off 'http://user:pass@nic.changeip.com/nic/update?u=user&p=pass&cmd=update&hostname=mydns.changeip.net&ip=89.133.69.82' 2>/var/run/ddns/myddns_ipv4.err
 225839       : DDNS Provider answered:
200 Successful Update
 225839 ERROR : IP update not accepted by DDNS Provider
 225839       : Waiting 600 seconds (Check Interval)
```
and this goes on forever.
Note that regardless of `200 Successful Update` there's still an error message `ERROR : IP update not accepted by DDNS Provider`.
And the update is of course successful always.

System has LEDE r1733, with ddns-scripts - 2.7.5-2

hnyman, 2016-10-03:
cc to DDNS maintainer: @chris5560

duvi, 2016-10-04:
Thank you, it'f fixed now.

duvi, 2016-12-11:
Need to reopen this.
The recent ddns changes brought back the old behaviour. The answer of changeip.com has to be fixed again.
Thank you!

closed 2018-08-17 by dibdot