openwrt/packages #3308: python-pyopenssl:warning exposed by cffi-1.8.3

earthGavinLee, 2016-10-06:
.../lib/python2.7/site-packages/OpenSSL/crypto.py:1108:
UserWarning: implicit cast from 'char *' to a different pointer type: will be
forbidden in the future (check that the types are as you expect; use an
explicit ffi.cast() if they are correct)

fixed these issues
001-fix_cffi_warnings_for_proper_buffer_types.patch
diff --git a/src/OpenSSL/SSL.py b/src/OpenSSL/SSL.py
index 3f97ccb..2d76516 100644
--- a/src/OpenSSL/SSL.py
+++ b/src/OpenSSL/SSL.py
@@ -1613,7 +1613,7 @@ class Connection(object):
             return None
         length = _lib.SSL_get_server_random(self._ssl, _ffi.NULL, 0)
         assert length > 0
-        outp = _ffi.new("char[]", length)
+        outp = _ffi.new("unsigned char[]", length)
         _lib.SSL_get_server_random(self._ssl, outp, length)
         return _ffi.buffer(outp, length)[:]
 
@@ -1629,7 +1629,7 @@ class Connection(object):
 
         length = _lib.SSL_get_client_random(self._ssl, _ffi.NULL, 0)
         assert length > 0
-        outp = _ffi.new("char[]", length)
+        outp = _ffi.new("unsigned char[]", length)
         _lib.SSL_get_client_random(self._ssl, outp, length)
         return _ffi.buffer(outp, length)[:]
 
@@ -1645,7 +1645,7 @@ class Connection(object):
 
         length = _lib.SSL_SESSION_get_master_key(session, _ffi.NULL, 0)
         assert length > 0
-        outp = _ffi.new("char[]", length)
+        outp = _ffi.new("unsigned char[]", length)
         _lib.SSL_SESSION_get_master_key(session, outp, length)
         return _ffi.buffer(outp, length)[:]
 
diff --git a/src/OpenSSL/crypto.py b/src/OpenSSL/crypto.py
index 116cc51..7918877 100644
--- a/src/OpenSSL/crypto.py
+++ b/src/OpenSSL/crypto.py
@@ -1100,7 +1100,7 @@ class X509(object):
         if digest == _ffi.NULL:
             raise ValueError("No such digest method")
 
-        result_buffer = _ffi.new("char[]", _lib.EVP_MAX_MD_SIZE)
+        result_buffer = _ffi.new("unsigned char[]", _lib.EVP_MAX_MD_SIZE)
         result_length = _ffi.new("unsigned int[]", 1)
         result_length[0] = len(result_buffer)
 
diff --git a/src/OpenSSL/rand.py b/src/OpenSSL/rand.py
index 087b243..cce1327 100644
--- a/src/OpenSSL/rand.py
+++ b/src/OpenSSL/rand.py
@@ -54,7 +54,7 @@ def bytes(num_bytes):
     if num_bytes < 0:
         raise ValueError("num_bytes must not be negative")
 
-    result_buffer = _ffi.new("char[]", num_bytes)
+    result_buffer = _ffi.new("unsigned char[]", num_bytes)
     result_code = _lib.RAND_bytes(result_buffer, num_bytes)
     if result_code == -1:
         # TODO: No tests for this code path.  Triggering a RAND_bytes failure

jefferyto, 2016-10-16:
Thanks for investigating this. Can you submit a pull request to upstream ([pyOpenSSL](https://github.com/pyca/pyopenssl))? As this is just a warning I think we can wait for upstream to merge.

commodo, 2017-05-26:
@earthGavinLee 
can we close this issue ?
or is there anything else ?

[ for reference, I'm just butt-ing in, because I'm looking through all issues on Github that have any `python` keyword in them, and trying to find closure for some, and see if i missed any ]

dibdot, 2018-08-17:
no feedback - close this by now.

closed 2018-08-17 by dibdot