openwrt/packages #3319: luasec: 'unsupported relocation type 10' error on powerpc

maz-1, 2016-10-09:
Maintainer: @MikePetullo 
Environment: LEDE Reboot r0

Description:
luasec cannot be built with libssp disabled, it complains about “undefined reference to `__stack_chk_fail_local’”
 I modified luasec makefile and append -fno-stack-protector to CFLAGS. It builts after modification. But when i try to load ssl module, it gives errors:
```
Lua 5.1.5 Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> require “ssl” error loading module ‘ssl.core’ from file ‘/usr/lib/lua/ssl.so’:
      Error relocating /usr/lib/lua/ssl.so: unsupported relocation type 10
stack traceback:
      [C]: ?
      [C]: in function 'require'
      /usr/lib/lua/ssl.lua:7: in main chunk
      [C]: in function 'require'
      stdin:1: in main chunk
      [C]: ?
```
As @chunkeey has mentioned, replace -fPIC with -fpic in luasec makefile should solve the problem
https://bugs.lede-project.org/index.php?do=details&task_id=186

diizzyy, 2019-03-28:
@maz-1 
Is this still an issue and if so, can you do a PR?

neheb, 2019-03-28:
http://downloads.openwrt.org/snapshots/faillogs/powerpc_464fp/packages/luasec/compile.txt

neheb, 2020-01-12:
This is fixed.

closed 2020-01-12 by neheb