openwrt/packages #3320: minidlna: minidlna won't detect video files on WNDR4700

maz-1, 2016-10-09:
Maintainer: @antonlacon @chunkeey
Environment: LEDE Reboot r0 ( WNDR4700, powerpc )

Description:
I compiled minidlna and libffmpeg-full for wndr4700, 
audios and pictures can be detected just fine
however minidlna won't detect any video files, and it crashes as soon as I move video files into the media directory
here is the debug log
```
[2016/10/09 03:09:58] getifaddr.c:327: info: Enabled interface 192.168.1.1/255.255.255.0
[2016/10/09 03:09:58] minidlna.c:984: warn: Starting MiniDLNA version 1.1.0.
[2016/10/09 03:09:58] minidlna.c:354: warn: Creating new database at /tmp/run/minidlna/files.db
[2016/10/09 03:09:58] scanner.c:703: warn: Scanning /mnt/ssd/MEDIA/Video
[2016/10/09 03:09:58] minidlna.c:1024: warn: HTTP listening on port 8200
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minissdp.c:734: maxdebug: Sending ssdp:byebye
[2016/10/09 03:09:58] minidlna.c:1065: debug: Sending SSDP notifies
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:58] minissdp.c:294: maxdebug: Sending ssdp:alive
[2016/10/09 03:09:59] inotify.c:159: debug: Add watch to /mnt/ssd/MEDIA/Video
[2016/10/09 03:09:59] inotify.c:166: debug: Add watch to /mnt/ssd/MEDIA/Video
[2016/10/09 03:10:09] inotify.c:757: debug: The file /mnt/ssd/MEDIA/Video/test.mp3 was moved here.
[2016/10/09 03:10:09] inotify.c:373: debug: Adding: /mnt/ssd/MEDIA/Video/test.mp3
Not a JPEG file: starts with 0x89 0x50
[2016/10/09 03:10:13] inotify.c:757: debug: The file /mnt/ssd/MEDIA/Video/test.mp4 was moved here.
[2016/10/09 03:10:13] inotify.c:373: debug: Adding: /mnt/ssd/MEDIA/Video/test.mp4
Illegal instruction

```
and the configuration file
```
port=8200
network_interface=br-lan
friendly_name=OpenWrt DLNA Server
db_dir=/var/run/minidlna
log_dir=/var/log
inotify=yes
enable_tivo=no
strict_dlna=no
album_art_names=Cover.jpg/cover.jpg/AlbumArtSmall.jpg/albumartsmall.jpg/AlbumArt.jpg/albumart.jpg/Album.jpg/album.jpg/Folder.jpg/folder.jpg/Thumb.jpg/thumb.jpg
notify_interval=900
serial=12345678
model_number=1
root_container=.
media_dir=/mnt/ssd/MEDIA/Video
```

build000, 2016-10-09:
Using default install mininidlna in LuCI
r1814 (WNDR4700)
```
ssh root@wndr4700.lan -p 54123
Warning: the RSA host key for '[wndr4700.lan]:54123' differs from the key for the IP address '[192.168.1.1]:54123'
Offending key for IP in /home/sj/.ssh/known_hosts:2
Matching host key in /home/sj/.ssh/known_hosts:4
Are you sure you want to continue connecting (yes/no)? yes

BusyBox v1.25.0 () built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (HEAD, r1814)
    \________\/    -----------------------------------------------------------

 | Machine: Netgear WNDR4700/WNDR4720 Series                                 |
 | Uptime: 0d, 01:29:22                                                      |
 | Load: 0.36 0.13 0.18                                                      |
 | Flash: total: 6.8MB, free: 6.3MB, used: 2%                                |
 | Memory: total: 249.3MB, free: 195.9MB, used: 21%                          |
 | WAN: 89.70.80.230, proto: dhcp                                            |
 | LAN: 192.168.1.1                                                          |
 | radio0: mode: ap, ssid: LEDE-2, channel: 11, conn: 1                      |
 | radio1: mode: ap, ssid: LEDE-5, channel: 128, conn: 2                     |
 | radio1: mode: sta, ssid: LEDE-ORANGE, channel: 128, conn: 1               |
 -----------------------------------------------------------------------------
root@wndr4700:~# opkg update
Downloading file://mnt/sdb1/guest/WNDR4700/targets/apm821xx/nand/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_core.
Downloading file://mnt/sdb1/guest/WNDR4700/targets/apm821xx/nand/packages/Packages.sig.
Signature check passed.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_base.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/base/Packages.sig.
Signature check passed.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/telephony/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_telephony.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/telephony/Packages.sig.
Signature check passed.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_packages.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/Packages.sig.
Signature check passed.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/routing/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_routing.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/routing/Packages.sig.
Signature check passed.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/mycompile_reboot_luci.
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/luci/Packages.sig.
Signature check passed.
root@wndr4700:~# opkg install luci-app-minidlna libffmpeg-audio-dec
Installing luci-app-minidlna (git-16.280.54886-e06c60a-1) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/luci/luci-app-minidlna_git-16.280.54886-e06c60a-1_all.ipk.
Installing minidlna (1.1.5-1) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/minidlna_1.1.5-1_powerpc_464fp.ipk.
Installing libexif (0.6.21-1) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/libexif_0.6.21-1_powerpc_464fp.ipk.
Installing libjpeg (9a-1) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/libjpeg_9a-1_powerpc_464fp.ipk.
Installing libsqlite3 (3130000-2) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/libsqlite3_3130000-2_powerpc_464fp.ipk.
Installing libffmpeg-mini (2.8.8-1) to root...
Downloading file://mnt/sdb1/guest/WNDR4700/packages/powerpc_464fp/packages/libffmpeg-mini_2.8.8-1_powerpc_464fp.ipk.
[... 66 lines cut ...]
album_art_names=Cover.jpg/cover.jpg/AlbumArtSmall.jpg/albumartsmall.jpg/AlbumArt.jpg/albumart.jpg/Album.jpg/album.jpg/Folder.jpg/folder.jpg/Thumb.jpg/thumb.jpg
notify_interval=900
serial=12345678
model_number=1
root_container=.
media_dir=/mnt/sdb1
#################################
from LuCI:
miniDLNA Status

The miniDLNA service is active, serving 2 audio, 0 video and 0 image files.
#################################
ls -l /mnt/sdb1/media
-rw-r--r--    1 root     root         31591 Oct  9 13:28 5506197276.png
-rw-r--r--    1 root     root         31151 Oct  9 13:28 5506199475.png
-rw-r--r--    1 root     root         31931 Oct  9 13:28 5532010915.png
-rw-r--r--    1 root     root     1371182649 Oct  9 13:34 Rosewood.S02E03.720p.HDTV.x264-KILLERS.mkv
-rwx------    1 root     root     670464774 Oct  9 13:39 Sunrise Mass - Ola Gjeilo.mp4
-rwxr--r--    1 nobody   nogroup      74496 Jan  5  2014 sb16.mp3
-rwxr--r--    1 nobody   nogroup     102144 Jan  5  2014 sb25.mp3
-rwxr--r--    1 nobody   nogroup   33078366 Jan 30  2016 sunrise_mass-ola_gjeilo.mka
-rwx------    1 root     root     419805468 Oct  9 13:37 sunrise_mass-ola_gjeilo.mkv
###############################
```
Generally its the same problem.

thess, 2016-10-09:
@maz-1 - I need to investigate your issue a bit more - However...

@build000 - Without verifying this myself - I believe the problem you are seeing here has to do with which ffmpeg library package is actually installed. These packages are very different in configuration and capabilities even though they each provide libffmpeg to satisfy other packages dependencies.

minidlna should correctly function with either libffmpeg-full or libffmpeg-mini (which is an explicit configuration for support of minidlna). libffmpeg-audio-dec is ONLY audio decoders and has no video capability at all. The unfortunate issue is that libffmpeg-audio-dec is alphabetically first of the 3 packages and it provides ffmpeg.

I think if you first install either libffmpeg-full or libffmpeg-mini before installing minidlna, things should work a whole lot better. You might needs to remove libffmpeg-audio-dec first if already installed.

thess, 2016-10-09:
@antonlacon - Any other thoughts on the original problem?

build000, 2016-10-09:
@thess 
What I showed is like the accumulation of two approaches to the problem:
1) First I installed the machine without libffmpeg-audio-dec - nothing at all to see the available media
2) then only added libffmpeg-audio-dec - the effect was that he saw only that 2 mp3 files.

Generally do not see any movies - does not help separate installation in the first place libffmpeg (whether full, mini or custom), and then the rest.

chunkeey, 2016-10-09:
Well, I asked Google and the first hit was: Altivec. The APM82181-SoC doesn't support it, so I just
compiled libffmpeg-mini with --disable-altivec and gave it a try with a sample .mp4 video of "Big Buck Bunny". minidlna found it and I could watch it in VLC. I also compiled libffmpeg-full, but didn't test it.
Note: If you have already downloaded libffmpeg-* from the repository, you need to use --force-checksum for opkg install.
  
[compiled libffmpeg-mini and libffmpeg-full](https://drive.google.com/drive/folders/0BwOU1CguSZuBODNjV0phTHZ2OGM)

Can somebody make a patch, otherwise this will never get fixed? I'm busy with something else (automatic wifi detection). As for patch suggestions: I think it would be enough to just look if the arch is 464fp and add --disable-altivec to FFMPEG_CONFIGURE in packages/multimedia/ffmpeg/Makefile.
However, this should also be posted to ffmpeg, so they can improve their altivec detection.

thess, 2016-10-09:
@build000 - was merely suggesting a work-around until I have time to really fix this (If at all possible given the way dependencies work).

@chunkeey - Thanks the info. I was not able to actually look into the problem and fix(es) today. I will try to get it done tomorrow.

build000, 2016-10-10:
@chunkeey @thess 
confirm;
- compilation @chunkeey (--disable-alivec) workink properly (testing only libffmpeg-mini - actualy not planed chroot):

> miniDLNA Status

>The miniDLNA service is active, serving 2 audio, 3 video and 0 image files.

Only not wiew 3 *.png files (probably this situation if only libffmpeg-mini).

thess, 2016-10-10:
@build000 - You may very well be correct about the pruning of parsers & demuxers by Ian's last submisstion. IIRC, I added most of those a few years back due to similar media recognition issues with MPD. Perhaps this is also why PNG is not recognized. (libpng is not used by ffmpeg.)

thess, 2016-10-10:
Darn - wrong button...

antonlacon, 2016-10-10:
For the original @maz-1 issue, where did this minidlna makefile come from? Our oldest makefile in the github tree is for 1.1.3, and it's been at 1.1.5 since 2016 started. Which ffmpeg version is being used, too?

For the @build000 issue with images, my last commit didn't touch still images at all. jpegls and png are still in place under FFMPEG_MINI_DECODERS. No parsers were selected for these formats to be removed. Did image detection ever work for you with libffmpeg-mini?

build000, 2016-10-11:
@antonlacon 
Long ago, the openwrt work showing PNG version libffmpeg (when she was still only this one library without division into mini, full and custom) and himself have set (de) muxer's and linked with libpng burying in the libffmpeg (sources) - officially does not always work showing PNG images ... this little riddle ...
Now you can not set it at all in any way.

maz-1, 2016-10-11:
@antonlacon 
I can confirm that building ffmpeg with --disable-altivec will solve the problem
The minidlna version in the log is not correct. At first I ran into the problem with the official version, so I try to build minidlna from git repo ( which should be 1.1.6 ) but the problem still persists, so it is definitely a problem with ffmpeg.

maz-1, 2016-10-11:
@build000 
I think png support is always enabled
```
FFMPEG_MINI_DECODERS:= \
	aac \
	ac3 \
	flac \
	h264 \
	hevc \
	jpegls \
	mp3 \
	mpeg1video \
	mpeg2video \
	mpeg4 \
	mpegvideo \
	opus \
	png \
	vc1 \
	vorbis \
	wmav1 \
	wmav2 \
```

antonlacon, 2016-10-11:
The single variant version of libffmpeg handled png the same way the variants are now. See: https://dev.openwrt.org/browser/packages/multimedia/ffmpeg/Makefile?rev=28101

Minidlna does not appear to support PNG files. See: https://sourceforge.net/p/minidlna/patches/132/

thess, 2016-10-11:
Disabling building ffmpeg with altivec on hard-float powerpc fixes OP issue. Closing this with: e32715621118cca6a441c19eb1605cde30a2a663

Discussions regarding PNG and other minidlna/ffmpeg issues should be on a separate thread.

closed 2016-10-11 by thess