openwrt/packages #3325: luci-app-fwknopd: UI in luci shows errors

dkadioglu, 2016-10-09:
Maintainer: @jp-bennett 
Environment: (mvebu, WRT1900ACS, LEDE trunk)

fwknopd cannot be configured using luci because of the following error:

`Sorry, the server encountered an unexpected error.
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute cbi dispatcher target for entry '/admin/services/fwknopd'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
A runtime error occured: [string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:15: attempt to call field 'json_encode' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function`

Because of this the save button is not displayed.
This happened on my old router with OpenWRT and on my new router with LEDE.

If you need further information, please ask.

jp-bennett, 2016-10-09:
Thanks for the report, I'll dig into this and get it fixed.
--Jonathan

AlexKay88, 2016-10-12:
Same here with OpenWRT trunk on x86_64.
I made a dirty workaround which only helps to display the fwknop web page, but not completely:

deleted `s.dynamic=true` in 2 sections (line 16 and 47) of `/usr/lib/lua/luci/model/cbi/fwknopd.lua`

Error messages got less now:
```
Wed Oct 12 14:43:27 2016 daemon.err uhttpd[1467]: uci: Entry not found
Wed Oct 12 14:43:27 2016 daemon.err uhttpd[1467]: uci: Entry not found
Wed Oct 12 14:43:27 2016 daemon.err uhttpd[1467]: sh: : unknown operand
Wed Oct 12 14:43:27 2016 daemon.err uhttpd[1467]: sh: : unknown operand
```

jp-bennett, 2016-10-12:
I'm compiling LEDE trunk now.  What makes this challenging is that I don't know what changed that suddenly broke the application.  

@jow- or @hnyman I have tried a handful of approaches to try to get this working, with no success.  Any hints would be appreciated.  At this point, I'm not sure where to be looking for the problem.  It is beginning to look like a bug in Luci.

hnyman, 2016-10-12:
That sounds like a script launched by LuCI is mishaving. uhttpd is just documenting the errors from the underlying script (first uci is used and probably fails, then .

In similar situations I have tried to enter debug logging statements to the scripts, trying to find the culprit.
E.g. in #2798 caused usage of an uninitialised variable, solved by https://github.com/openwrt/luci/pull/740

hnyman, 2016-10-12:
I briefly tested it and, yes the page looks somehow broken. 

There seems to be actually two different errors:
* uci & sh errors come from a somehow faulty QR script (called via  "fwknopd-qr"). Disabling that template call removes the errors from system log.
* The ``` 'json_encode' (a nil value) ``` error seems to come from the "dynamic" attribute. I have no experience with that. @jow- is needed for that. (Actually, luci-app-fwknopd seems to be almost the only user of that option., so the whole functionality may have been broken for some time...  (Luci-mod-admin-mini and luci-app-asterisk seem to reference it, but otherwise I find no reference to it)

hnyman, 2016-10-12:
The two uci & sh errors come from the uci complaining about non-existent values and then using later uninitialised variables.

This is with the default config right after installing the package.
```
root@lede:~# uci get fwknopd.@access[0].KEY_BASE64
Nli0U4gFE0+NSkV3sgJXQRq80pBCGc7JNPk3/uu+NNc=
root@lede:~# uci get fwknopd.@access[0].KEY
uci: Entry not found
root@lede:~# uci get fwknopd.@access[0].HMAC_KEY_BASE64
D4xnqXDn6PKSFvPfCKf6nck1BHW5mdwd80w/aI7wOIF2YUu1SZNdneONvG16UKjqCXyB+gUTmW8VQzYpDV4d0g==
root@lede:~# uci get fwknopd.@access[0].HMAC_KEY
uci: Entry not found
```

jp-bennett, 2016-10-12:
The qr script needs to be able to include either a KEY_BASE64 or a KEY entry, which is why it is checking the possibly nonexistant value.  I can clean that up to get rid of the error messages.  

The bigger issue is the broken luci page resulting from the dynamic attribute.  I believe the dynamic attribute allows for the addition of extra values in the UCI section.  Fwknop supports lots of flags, most of which do not need to be shown on the Luci page by default.  "Dynamic" allows the user to specify the option name and value of these extra options.

dkadioglu, 2016-10-30:
Hi again,

first, thank you very much for looking into this!
I would like to ask, if there are any news on this?

Thanks

jp-bennett, 2016-10-30:
@dkadioglu I'm fairly convinced that this is a bug in Luci.  I've opened a bug report and tagged @jow- but haven't heard anything yet.  We'll get it fixed eventually.

xcom169, 2017-01-04:
Hello All,
I'm dealing with the same issue. Do you know about any workaround here?
Is it possible to configure fwknopd by UCI -without Luci- ?

hnyman, 2017-01-04:
> Is it possible to configure fwknopd by UCI -without Luci- ?

Naturally yes. LuCI is just a graphical GUI modifying the underlying config file /etc/config/fwknopfd, which you can also edit manually from console.

xcom169, 2017-01-04:
Well yes, just in Luci it's much easier. I hope it will be fixed soon.

jp-bennett, 2017-01-04:
I'm planning to spend time on this this weekend.  I talked a bit with the Luci guys, and we came to conclusion about what needs to happen to fix it.  Hope to have a PR opened to fix this by Monday.

xcom169, 2017-01-09:
Really good news. Please keep us posted.

jp-bennett, 2017-01-09:
Have a PR open with the fix:  https://github.com/openwrt/luci/pull/927

Also snuck in a couple extra goodies.  This sets NAT_DNS to true by default, and also generates a QR code per access.conf stanza, whereas previously only the first stanza would generate a QR code.

hnyman, 2017-01-09:
This has likely been fixed with  https://github.com/openwrt/luci/pull/927

closed 2017-01-09 by hnyman

xcom169, 2017-01-09:
Super! Do you think it will be also implemented to LEDE? It uses the same source for Luci as OpenWRT?

hnyman, 2017-01-09:
> Do you think it will be also implemented to LEDE? It uses the same source for Luci as OpenWRT?

Luci and Packages feed repos are common to both Openwrt and LEDE, so all changes are visible in both of them.

dkadioglu, 2017-01-09:
Great, it works now. Thank you very much for the fix!