openwrt/packages #3333: reaver: segmentation fault

psyborg55, 2016-10-11:
trunk, kernel 4.1.4, ramips

reaver output:

>     [+] Switching mon0 to channel 1
    [+] Waiting for beacon from XX:XX:XX:XX:XX:XX
    [!] WARNING: Failed to associate with XX:XX:XX:XX:XX:XX (ESSID: xxxxxxxx)
    [+] Associated with XX:XX:XX:XX:XX:XX (ESSID: xxxxxxxx)
    [+] Trying pin 12345670
    [+] Sending EAPOL START request
    [+] Received identity request
    [+] Sending identity response
    [+] Received identity request
    [+] Sending identity response
    [+] Received identity request
    [+] Sending identity response
    [+] Received identity request
    [+] Sending identity response
    [+] Received M1 message
    Segmentation fault

syslog output:

>     Tue Oct 11 14:27:16 2016 kern.info kernel: [64510.933931] 
    Tue Oct 11 14:27:16 2016 kern.info kernel: [64510.933931] do_page_fault(): sending SIGSEGV to reaver for invalid read access from 00000004
    Tue Oct 11 14:27:16 2016 kern.info kernel: [64510.942600] epc = 0041ac30 in reaver[400000+49000]
    Tue Oct 11 14:27:16 2016 kern.info kernel: [64510.947561] ra  = 004073e0 in reaver[400000+49000]
    Tue Oct 11 14:27:16 2016 kern.info kernel: [64510.952698]

neheb, 2016-10-16:
reaver has not been publicly updated in several years. It would probably be better to use the forked version at https://github.com/t6x/reaver-wps-fork-t6x

yousong, 2016-10-16:
Hi @neheb , proposal of using t6x fork of reaver has been put forward before and I think it's the right direction to go as the current upstream package seems unmaintained at all.  But I hope someone other than me will step up, do the transition, then take the maintainership of reaver package here.  The reasons have been given at https://github.com/openwrt/packages/issues/2872#issuecomment-242100214

Regards

psyborg55, 2016-10-21:
wait. forked version has new features that could bring more bugs. we have this old version that is not very stable but should work on LE devices. how about working BE patch? have the guys working on forked version found a fix for BE? i've tried latest trunk and now got errors: WARNING: Failed to associate with...

yousong, 2016-10-21:
@psyborg55 , just for your information, the t6x fork do have a branch named `Big_endian` (https://github.com/t6x/reaver-wps-fork-t6x/commits/Big_endian)

psyborg55, 2016-10-21:
didn't think it actually works because of failed association warnings, but seems like that problem is related to something else, maybe mac80211?

psyborg55, 2016-10-23:
happens with both reaver versions

>     [  1423.197749] do_page_fault(): sending SIGSEGV to reaver for invalid read access from 00000004
    [  1423.206432] epc = 0041acf0 in reaver[400000+49000]
    [  1423.211373] ra  = 00407460 in reaver[400000+49000]
    [  1423.216299] 
    [  1438.806961] 
    [  1438.806961] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000000
    [  1438.815616] epc = 777b62b0 in libc.so[7778e000+91000]
    [  1438.820907] ra  = 7775a0f3 in libpcap.so.1.3.0[77756000+36000]
    [  1438.826936] 
    [  1446.398974] 
    [  1446.398974] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1446.407571] epc = 77ae5e20 in libc.so[77abe000+91000]
    [  1446.412787] ra  = 77ae6c8c in libc.so[77abe000+91000]
    [  1446.417973] 
    [  1451.210009] 
    [  1451.210009] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1451.218613] epc = 77a2be20 in libc.so[77a04000+91000]
    [  1451.223849] ra  = 77a2cc8c in libc.so[77a04000+91000]
    [  1451.229013] 
    [  1461.230711] 
    [  1461.230711] do_page_fault(): sending SIGSEGV to reaver for invalid read access from 00000000
    [  1461.239409] epc = 7708dadf in libpcap.so.1.3.0[7708a000+36000]
    [  1461.245634] ra  = 7708e495 in libpcap.so.1.3.0[7708a000+36000]
    [  1461.251644] 
    [  1485.280480] 
    [  1485.280480] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1485.289058] epc = 77ef5e20 in libc.so[77ece000+91000]
    [  1485.294263] ra  = 77ef6c8c in libc.so[77ece000+91000]
    [  1485.299427] 
    [  1498.262818] 
    [  1498.262818] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1498.271444] epc = 77493e20 in libc.so[7746c000+91000]
    [  1498.276669] ra  = 77494c8c in libc.so[7746c000+91000]
    [  1498.282367] 
    [  1506.140389] 
    [  1506.140389] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1506.148967] epc = 776a1e20 in libc.so[7767a000+91000]
    [  1506.154152] ra  = 776a2120 in libc.so[7767a000+91000]
    [  1506.159354] 
    [  1612.802557] 
    [  1612.802557] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1612.811182] epc = 774e9e20 in libc.so[774c2000+91000]
    [  1612.816410] ra  = 774eac8c in libc.so[774c2000+91000]
    [  1612.821600] 
    [  1636.746282] 
    [  1636.746282] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000000
    [  1636.754911] epc = 77ab42b0 in libc.so[77a8c000+91000]
    [  1636.760099] ra  = 77a580f3 in libpcap.so.1.3.0[77a54000+36000]
    [  1636.766107]

neheb, 2016-10-25:
Send a bug report to the fork

yousong, 2016-12-08:
I was notified a recent merge into reaver-t6x that could possibly fix the sigsev issue here: https://github.com/t6x/reaver-wps-fork-t6x/pull/101

Before doing any runtest, I would recommend doing a tcpdump first and see if the related packets are oversized in your environment.  The change is from `BUFSIZ` defined by musl as 1024 to const 65536.

You can do this on your device with the following steps (adapt it to your mileage ;)

```
iw phy phy0 interface add mon0 type monitor
ip link set mon0 up
tcpdump -i mon0 -nne  -w /tmp/p.cap 'type mgt and ( subtype beacon or subtype proberesp ) '
```

Check the actual frame length to see if any of them is bigger than 1024.

yousong, 2016-12-12:
I am closing this one for the moment.  Please report back if the issue still persists.  Thanks for the feedback.

closed 2016-12-12 by yousong