openwrt/packages #3338: ddns-scripts: fails updating DDNS as long as 2 IPv6 addresses assigned

AlexKay88, 2016-10-12:
-------------------------------
Maintainer: @chris5560 
Environment: x86, 64, OpenWRT trunk

Description:
After getting a new IPv6 prefix from ISP I have 2 prefixes assigned until old prefix becomes deprecated.
ddns-scripts detects (via interface) 2 IPv6 addresses, but concatenates both IPv6 to e.g.
"2001:db8:4:4::1**2001:db8:5:5::1**" Trying to update DDNS provider with this address fails obviously.

On my system I added some additional code to use `ip `command instead of `ifconfig `and choose the right IPv6 depending on highest "preferred_lft" which corresponds to latest IPv6.

```
ip -o -6 addr show dev eth1 scope global primary | grep -v "inet6 f" | sed 's/^.* inet6 //' | sed 's/scope.*valid/valid/'
2001:db8:4:4::1/64 valid_lft 6739sec preferred_lft 3139sec
2001:db8:5:5::1/64 valid_lft 10000sec preferred_lft 5000sec   <------ right IPv6 to choose
```

The code addition works fine in my environment after observing it for a few days.
Code added in file `dynamic_dns_functions.sh` , function `get_local_ip ()` (line 880) under section `interface` between line 929 and 930.

```
							}' $DATFILE
						)
####################### begin of my debug code ###########################################################
#						write_log 6 "DEBUG: start of debug section"
						if [ $(echo "$__DATA" | grep -c . ) -gt 1 ]; then
#							write_log 6 "DEBUG: __DATA has more than one line: $__DATA"
							MY_IP_CMD=$(ip -o -6 addr show dev $ip_interface scope global primary | grep -v "inet6 f" | awk '/inet6/ && /: [0-9a-eA-E]/ {FS="/"; $0=$0; $2=""; FS=" "; $0=$0; print $4;}')
							if [ $(echo "$MY_IP_CMD" | grep -c . ) -gt 1 ]; then
#								write_log 6 "DEBUG: IP_CMD has more than one line: $MY_IP_CMD"
								MY_IP_RAW1=$(ip -o -6 addr show dev $ip_interface scope global primary | grep -v "inet6 f" | sed -n 1p | sed 's/^.* inet6 //' | sed 's/scope.*valid/valid/')
								MY_IP_PLFT1=$(echo "$MY_IP_RAW1" | awk -F '[ \t]+|/' '{print $6}' | sed s'/sec//'g)
								if [ ! $MY_IP_PLFT1 -eq $MY_IP_PLFT1 ] 2>/dev/null ; then
									write_log 6 "DEBUG: IP_CMD: PARSE ERROR: MY_IP_PLFT1 is not a number: $MY_IP_PLFT1 , RAW1: $MY_IP_RAW1"
									MY_IP_PLFT1=1
								fi
								MY_IP_RAW2=$(ip -o -6 addr show dev $ip_interface scope global primary | grep -v "inet6 f" | sed -n 2p | sed 's/^.* inet6 //' | sed 's/scope.*valid/valid/')
								MY_IP_PLFT2=$(echo "$MY_IP_RAW2" | awk -F '[ \t]+|/' '{print $6}' | sed s'/sec//'g)
								if [ ! $MY_IP_PLFT2 -eq $MY_IP_PLFT2 ] 2>/dev/null ; then
									write_log 6 "DEBUG: IP_CMD: PARSE ERROR: MY_IP_PLFT2 is not a number: $MY_IP_PLFT2 , RAW1: $MY_IP_RAW2"
									MY_IP_PLFT2=0
								fi
								if [ $MY_IP_PLFT1 -gt $MY_IP_PLFT2 ]; then
									MY_IP_L1=$(echo "$MY_IP_CMD" | sed -n 1p)
									__DATA="$MY_IP_L1"
									write_log 6 "DEBUG: IP_CMD: PLFT1 won, choosed: $MY_IP_L1 , PFLT1/PLFT2: $MY_IP_PLFT1/$MY_IP_PLFT2 , RAW1/RAW2: $MY_IP_RAW1/$MY_IP_RAW2"
								else
									MY_IP_L2=$(echo "$MY_IP_CMD" | sed -n 2p)
									__DATA="$MY_IP_L2"
									write_log 6 "DEBUG: IP_CMD: PLFT2 won, choosed: $MY_IP_L2 , PFLT1/PLFT2: $MY_IP_PLFT1/$MY_IP_PLFT2 , RAW1/RAW2: $MY_IP_RAW1/$MY_IP_RAW2"
								fi
							else
								write_log 6 "DEBUG: IP_CMD: __DATA has more than one line, but not MY_IP_CMD: $__DATA , $MY_IP_CMD"
							fi
#						else
#							write_log 6 "DEBUG: __DATA has one line as expected: $__DATA"
						fi
#						write_log 6 "DEBUG: end of debug section"
####################### end of my debug code ###########################################################
					fi
					[ -n "$__DATA" ] && write_log 7 "Local IP '$__DATA' detected on interface '$ip_interface'"
```
You may easily check my code by manually adding a second IPv6 address with a preferred_lifetime of zero seconds to an interface:
`ip -6 addr add 2001:db8:beef:beef:beef::1/64 dev eth1 valid_lft 60 preferred_lft 0`
It does not confuse your test system, because the address will never be used and disappear after 60 seconds.

It would be great if you could review and improve my code, since I am not familiar in writing robust code.

Many thanks,
Alex

chris5560, 2016-10-12:
@AlexKay88

Thanks for your hint, I will definitely look on it.
Have you check what happen if you use ip_source network?

AlexKay88, 2016-10-13:
@chris5560 
I can select ip_source `network `instead of `interface`, but it seems to be ignored when applying...it doesn't become active.

chris5560, 2016-10-13:
@AlexKay88
you not only need to set
option ip_source	"network"
also you need to set (i.e. wan or wan6)
option ip_network	"wan"

AlexKay88, 2016-10-15:
@chris5560 
With _doesn't get applied_, I meant changing by `luci-app-ddns` web page. After I changed /etc/config/ddns manually, it gets active.
Here is what I tested:

```
config service 'WAN6_kay6_test'
	option use_ipv6 '1'
	option interface 'wan6'
	option ip_source 'network'
	option lookup_host 'my.dyndns.name'
	option check_interval '20'
	option force_interval '360'
	option service_name 'afraid.org'
	option password 'blabla'
	option use_https '1'
	option cacert '/etc/ssl/certs/'
	option use_syslog '1'
	option use_logfile '0'
	option enabled '1'
```
`ddns-scripts` should send my IPv6 address to dyndns provider, but fails getting the right IPv6 address.
It wrongly choosed my Unique Local Address (ULA) Prefix (`fdXX::blabla`) address instead of the right `2001::` IPv6 address.
Log messages:
```
user.err ddns-scripts[9843]: WAN6_kay6_test: No or private or invalid IP 'fdXX::blabla' given! Please check your configuration
user.err ddns-scripts[9843]: WAN6_kay6_test: No update send to DDNS Provider
```
So,` ip_source network` seems to be broken in a environment with additional ULA Prefix addresses.
Now I remember again why I switched to `ip_source interface` ... ;-)

chris5560, 2016-10-16:
@AlexKay88

If internal functions (network_get_ipaddr6) of OpenWrt (/lib/functions/network.sh) don't report the correct IPv6, ddns-scripts' have no chance because they depend on it.
Please report the problem to OpenWrt/LEDE development. This should be an ubus problem because network.sh is using ubus to read current IP's.
The reason why the current version of ddns-scripts don't use "ip" command is, that first time "ip" is available inside OpenWrt build is DD (trunk).
Please have patience with me to implement "ip" in stead of "ifconfig". I'm currently working on another project.

dibdot, 2018-09-29:
let's close dated ddns tickets without further input.

closed 2018-09-29 by dibdot