openwrt/packages #3339: Python - HOST_CFLAGS uses absolute path to the build system (/usr/include/ncursesw)

zyxmon, 2016-10-12:
https://github.com/openwrt/packages/blob/master/lang/python/Makefile#L216

It looks like a bug

commodo, 2016-10-12:
Well, yes & no.
Python can build without libncurses(w) [if I remember correctly] ; I think it's there as a copy-paste fluke.
In that sense, if libncursesw is present it will use that, otherwise libncurses, and then finally it won't build it.

I will admit, it's not particularly correct.
We could remove it, or fix it [add a host-side libncurses build].
Though, I am a bit more inclined to remove it.

What error do you see ?

commodo, 2016-10-12:
ah ; just noticed the rename of the topic

closed 2016-10-12 by hnyman

zyxmon, 2016-10-13:
@commodo - we have removed these lines in Entware. No problems.