openwrt/packages #3342: tvheadend: Please add hdhomerun support

janwerner, 2016-10-14:
@sairon 
Could you enable "hdhomerun_client" build option, so that we can use HDHomeRun network tuners on Turris Omnia? I'm not sure why the feature is still disabled by default, since it's been working reliably for a while now.

sjjgo, 2018-07-01:
This is fixed in this PR: https://github.com/openwrt/packages/pull/6386

It's turned off by default, but can be added from the build configuration menu.

neheb, 2020-01-25:
ping @BKPepe I think you guys have this package updated?

BKPepe, 2020-01-25:
Yes, we do, but I don't have HDHomeRun network tuner, so I can not test it on that device. But before doing that, I will need to add two packages for that. However, I don't want to promise anything, but I will try to look at it and send it here as I am using it almost on daily basis.

walterav1984, 2021-07-22:
Can this still be enabled by default for 21.02 release?

M95D, 2021-08-30:
> ping @BKPepe I think you guys have this package updated?

Updated to v4.2.8 with significant changes in OpenWRT package features. See #16475.
(v4.3.0 appears to be significantly older than v4.2.8.)

HDHomeRun is enabled by default, but I didn't test it. I don't have the hardware.

walterav1984, 2021-08-30:
Nice I saw that P.R. this weekend and noticed the HDHomerun was set on default. 

Just testing snapshot OpenWrt SNAPSHOT, r17454-f08b76b32f on a Raspberry Pi and the HDhomerun HDHR3-EU is detected as tuner in "Configuration>DVB Inputs>Tv adapters" section! 

```
opkg install tvheadend
Installing tvheadend (4.2.8-2) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/packages/tvheadend_4.2.8-2_arm_arm1176jzf-s_vfp.ipk
Installing libopenssl1.1 (1.1.1l-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/base/libopenssl1.1_1.1.1l-1_arm_arm1176jzf-s_vfp.ipk
Installing zlib (1.2.11-3) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/base/zlib_1.2.11-3_arm_arm1176jzf-s_vfp.ipk
Installing libffi (3.4.2-1) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/packages/libffi_3.4.2-1_arm_arm1176jzf-s_vfp.ipk
Configuring libffi.
Configuring libopenssl1.1.
Configuring zlib.
Configuring tvheadend.
```

However it was detected as DVB-T, but after setting it to DVB-C both tuners are now scanning and operational tested  both on multiple muxes.

Questions:
Is it correct that the tvheadend webinterface has no username/password setup, pressing default login just signs me in?

Is satip-client in this setup functional since it won't detect the Fritzbox 6490 Satip server, this tuner is however detected on other machines/architectures running tvheadend? Or does it need extra upnp/avahi packages to operate? On other tvheadend instances it helps to discover by going in the tvheadend webinterface to the button "Configuration>General>SATIP-Server>Discover SATIP-Servers" but this also doesn't help.

Side note about the raspberry pi 1 512MB model, its ethernet port won't come up after it has been disconnected... probably unrelated to this.

walterav1984, 2021-08-30:
Feel free to close this issue since HDHomerun support has been verified using a HDHR3-EU using DVB-C with this commit #16475  multiple tuners and multiple muxes.

M95D, 2021-08-31:
> Questions:
> Is it correct that the tvheadend webinterface has no username/password setup, pressing default login just signs me in?
Yes. That is the default after (re)install. It creates a default user only after install. After that, the user can set up accounts from web interface.

sphr2k, 2021-08-31:
Thanks for adding.

closed 2021-08-31 by sphr2k

walterav1984, 2021-09-01:
@M95D will #16475 also be backported to the 21-02 branch or is more testing required?

M95D, 2021-09-01:
Is this my decision? I never did a backport before.
Since this was my first significant pull request, I would feel more comfortable if more people test it.