openwrt/packages #3343: libgcc  [package_install] Segmentation fault

David6P, 2016-10-14:
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
 According to https://wiki.openwrt.org/doc/devel/packages/libgcc?s[]=libgcc
 Maintainer is OpenWrt Developers Team openwrt-devel@openwrt.org

Environment: (put here arch, model, OpenWRT/LEDE version)
Image Building ar71xx-generic.Linux-x86_64 15.05.1 and 15.05 Profile TLMR3020

Description:
Imag Builder fails with Segmentation fault after downloading libgcc
make image PROFILE=TLMR3020
The same error occurs using either  
OpenWrt-ImageBuilder-15.05-ar71xx-generic.Linux-x86_64
or
OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64
as follows (with full path omitted)

Downloading file:packages/Packages.
Updated list of available packages in OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx//usr/lib/opkg/lists/imagebuilder.
make package_install
make[2]: Entering directory `OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64'

Installing packages...
IPKG_NO_SCRIPT=1 IPKG_TMP="OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/tmp/ipkgtmp" IPKG_INSTROOT="OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx" IPKG_CONF_DIR="OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/tmp" IPKG_OFFLINE_ROOT="OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx" OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/staging_dir/host/bin/opkg -f OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/repositories.conf --force-depends --force-overwrite --force-postinstall --cache OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/dl --offline-root OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx --add-dest root:/ --add-arch all:100 --add-arch ar71xx:200 install OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/packages/base/libc_0.9.33.2-1_ar71xx.ipk
Installing libc (0.9.33.2-1) to root...
Installing libgcc (4.8-linaro-1) to root...
Copying OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64/dl/libgcc_4.8-linaro-1_ar71xx.ipk.
make[2]: *** [package_install] Segmentation fault (core dumped)
make[2]: Leaving directory `OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64'
make[1]: *** [_call_image] Error 2
make[1]: Leaving directory `OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64'
make: *** [image] Error 2

robertek, 2016-12-06:
I have the same issue. Unfortunately the core file does not contain any symbols, so debugging is not easy.

Since staging_dir/host/bin/opkg is static executable it would not be cause with system libraries, so there are only two possibilities the opkg itself or some unsupported kernel syscall/instruction.

robertek, 2016-12-06:
So after some debugging and searching opkg code, it is caused by time syscall. It fail with segmentation fault when vsyscall=none on kernel commandline or configured in kernel. It works fine using vsyscall=emulate.

xcb567, 2016-12-29:
For future reference, @robertek solved it for me, adding vsyscall=emulate to my grub config

closed 2019-03-30 by neheb