openwrt/packages #3363: fping on openWRT

keulu, 2016-10-20:
My 2 cents for OpenWRT community

```
#
# Copyright (C) 2012-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=fping
PKG_VERSION:=3.13
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=http://fping.org/dist/
PKG_MD5SUM:=a39134baca6bc6ced56191cb51fb0dc2

PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)-$(PKG_VERSION)

PKG_INSTALL:=1

include $(INCLUDE_DIR)/package.mk

define Package/fping
  SECTION:=net
  CATEGORY:=Network
  TITLE:=Program to ping multiple hosts in parallel
  URL:=http://fping.org/
endef

define Package/fping/description
 Program to ping multiple hosts in parallel
endef

define Build/Configure
	( cd $(PKG_BUILD_DIR) ; rm -rf config.cache; \
		./configure \
			$(TARGET_CONFIGURE_OPTS) \
			CFLAGS="$(TARGET_CFLAGS)" \
			./configure \
			--build=$(GNU_HOST_NAME) \
			--host=$(GNU_TARGET_NAME) \
			--target=$(GNU_TARGET_NAME) \
			--prefix=/usr \
			--exec-prefix=/usr \
			--bindir=/usr/bin \
			--sbindir=/usr/sbin \
			--libexecdir=/usr/lib \
			--sysconfdir=/etc \
			--datadir=/usr/share \
			--localstatedir=/var \
			--mandir=/usr/man \
			--infodir=/usr/info \
			$(DISABLE_LARGEFILE) \
	)
endef

define Build/Prepare
	$(call Build/Prepare/Default)
endef

define Build/Compile
	cd $(PKG_BUILD_DIR) && $(MAKE)
	mv $(PKG_BUILD_DIR)/src/$(PKG_NAME) $(PKG_BUILD_DIR)/src/$(PKG_NAME)4
	cd $(PKG_BUILD_DIR) && CFLAGS="$(TARGET_CFLAGS) -DIPV6=1" && $(MAKE)
	mv $(PKG_BUILD_DIR)/src/$(PKG_NAME) $(PKG_BUILD_DIR)/src/$(PKG_NAME)6
endef

define Package/fping/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/src/$(PKG_NAME)4 $(1)/usr/bin/$(PKG_NAME)
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/src/$(PKG_NAME)6 $(1)/usr/bin/
endef

$(eval $(call BuildPackage,fping))
```

wongsyrone, 2016-10-25:
There are some redundant rules in Build/Configure, you can check mine.

ref: https://github.com/wongsyrone/lede-1/blob/master/package/external/fping/Makefile

keulu, 2016-10-25:
yeah, this is my first makefile, i need more reading and i saw how do make your file. 

thanks dude

diizzyy, 2016-11-02:
@wongsyrone 
Does mips16 break things or is it just performance optimized?
Does the cflags and stuff actually give a performance boost or does it save space?

@keulu If you want to import this (wongsyrone's port seems pretty much complete), please use this branch as netdata needs it later on. https://github.com/ktsaou/fping

wongsyrone, 2016-11-02:
Hi, I didn't test mips16 compilation, if someone can confirm there are no issues found, I can remove this restriction. `-ffunction-sections -fdata-sections` and `-Wl,--gc-sections` are used to decrease binary size. LTO is used to boost performance; it can be removed.

diizzyy, 2016-11-27:
#3567

diizzyy, 2017-06-09:
@hnyman @thess 
fping is in the tree, please close.

closed 2017-06-09 by hnyman