openwrt/packages #3371: all-packages: move to right category and make new categories

bobafetthotmail, 2016-10-22:
@thess @jow- @hnyman 

I'd like to re-arrange a bit the package feeds to have packages in the right place and/or in new categories to make them easier to find. As it is now it is a bit messy.

Can I do this in one single commit or do I need to do it piecemeal for each single package (as stated by contribution rules)?

Thanks for any answer.

thess, 2016-10-22:
I think a single commit would be fine as long as all you change is the structure of the Makefiles and not to introduce any package changes which would require maintainer review. I agree that sometimes the long lists in **Libraries**, **Utilities** & **Network** are hard to navigate

bobafetthotmail, 2016-10-22:
Ok, I was also thinking about moving around a few package folders in the source (from` /utils`  to `/net` or wherever) where it makes sense too.

For example **bandwithd**, **bandwithhd-php**, **banhost** and **macchanger** are in` /utils` but they are networking tools or daemons and must go in` /net`, and there are also some luci packages around that should go in a (new)` /luci` folder.
There is also an **alsa-utils** in `/utils` that should go in `/sound`.

But yes, I'm not going to change the actual package in any way.

Can I do that too in the commit? The packages that need to be moved like this are only a few and going piecemeal will be enough.

mhei, 2016-10-22:
In general, I agree to @thess, on the other side, to give each maintainer an Acked-by chance for courtesy, a single commit for each package could make this easier. But I know this would require some more efforts on your side - don't know how many packages do you want to touch...

bobafetthotmail, 2016-10-22:
>to give each maintainer an Acked-by chance for courtesy,

I can @ them in the PR, so they get notified that there was a change in their package even if they aren't required to show up and ack the change before it is merged.

>But I know this would require some more efforts on your side
Not by much, for these one-line edits it is faster to edit/commit/PR from Github directly, at least for me.

But what I wanted to do is more "generic openwrt feeds housekeeping" than "I'm changing the package", so I thought I could avoid wasting everyone's time with a zillion PRs about changes that don't really require the mantainer's attention.

Also making a zillion PRs about minor changes might be sending the wrong message or something, so I asked first.

>don't know how many packages do you want to touch...

Heh, a lot.

I'd like to remove the huuuge list of random packages that you find in Network, Utilities  and Libraries, and have all those packages appear under subcategories so you don't need to scroll around for pages and pages.

**make menuconfig** shows more than a hundred packages, by eyeballing it's like 50-ish actual makefiles to edit.

thess, 2016-10-22:
Agreeing with @mhei that we should at least give the maintainers a chance to ACK the change. In fact, I noticed a slight (benign) glitch in the alsa-utils in PR #3368 which needs clarification/correction. That said, you should post your changes singly or perhaps in small category related batches and be prepared for corrections before merging.

Since there is no urgency to do any of this, the PR's can sit for a week waiting for comments before they are merged as well. I really don't want to see any proposed changes hanging around very long so we can minimize the amount and duration of churn to the menus.

bobafetthotmail, 2016-10-23:
ok, will send per-package PRs then. Will close this issue when I'm done with sending PRs.

Thanks for the clarifications so far.