openwrt/packages #3400: peervpn: Could this package return?

bdklahn, 2016-10-26:
First off, thanks for maintaining packages for OpenWRT! I have used it often, over the years.

It looks like peervpn might have been included in attitude adjustment (git://git.openwrt.org/12.09/packages.git).

I suppose I could try that package (https://downloads.openwrt.org/attitude_adjustment/12.09/brcm47xx/generic/packages/peervpn_0-023-1_brcm47xx.ipk), in my currently running image (https://downloads.openwrt.org/chaos_calmer/15.05.1/brcm47xx/legacy/openwrt-15.05.1-brcm47xx-legacy-linksys-wrtsl54gs-squashfs.bin), on my WRTSL54GS. But it looks like that Makefile might be circa 2013, and there's newer (2016-03-21) source code at https://peervpn.net/files/peervpn-0-044.tar.gz.

Might a newer makefile look close to something like this?

```
#
# Copyright (C) 2007-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=peervpn
PKG_VERSION:=0-044
PKG_RELEASE:=2

PKG_BUILD_DIR :=$(BUILD_DIR)/$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL :=http://www.peervpn.net/files

include $(INCLUDE_DIR)/package.mk

define Package/peervpn
  SECTION:=net
  CATEGORY:=Network
  TITLE:=PeerVPN the open source peer-to-peer VPN
  URL:=http://www.peervpn.net/
  DEPENDS:=+libopenssl +ip
endef

define Package/peervpn/description
  PeerVPN is software that builds virtual ethernet networks between multiple
  computers. Adding a new node to an existing network doesn't require the
  existing nodes to alter their configuration, because its IP address is
  automatically distributed across the whole virtual network.
endef

define Package/peervpn/install
        $(INSTALL_DIR) $(1)/sbin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/peervpn $(1)/sbin/
        $(INSTALL_DIR) $(1)/etc/peervpn
        $(INSTALL_CONF) $(PKG_BUILD_DIR)/peervpn.conf $(1)/etc/peervpn
endef

$(eval $(call BuildPackage,peervpn))
```

Thanks for taking a look!

jp-bennett, 2016-10-26:
@bdklahn If you're willing to take over as the maintainer for this package, it could be moved from the legacy oldpackages repository.  Your makefile needs to include the updated md5sum, and needs to be tested before getting pushed into the official repo.

Responsibility for maintaining is fairly simple, you should update the makefile and test it for new releases of PeerVPN, be responsive to opened bugs, and let the community know if you decide to step down as maintainer.

bobafetthotmail, 2016-10-27:
Being this a VPN, please add 

SUBMENU:=VPN

so it gets shown in the VPN submenu

bdklahn, 2016-10-27:
Still considering if I want to take this. :-)
Last night I tried installing the older package (peervpn_0-023-1_brcm47xx.ipk). I hadn't ever used peervpn, and thought I'd better at least see if/how it works. It did install and work. So that might be good enough for me, and others (as may be the defacto "policy" for "unsupported" packages).

Adding the md5sum and "SUBMENU:=VPN" are easy enough. I did come across mention of that submenu in, maybe, a commit message somewhere. I wondered about that, because I don't see an obvious VPN submenu in my web interface. I also wondered if that required additional GUI configuration fields/elements to be made, to make it useful, since configuration of this seems very different than, say, OpenVPN. Right now I'm just running the following, in "Local Startup"
```
((/bin/ash -c '/sbin/peervpn /root/peervpn.conf')&)&
```
So how does the maintainer thing work? Would I make pull requests, or would I get push access to this repo? Would it get pushed out to a/your build farm, and get tested to compile on all the chips/platforms there, or would I be expected to successfully cross-compile to all targets, on my machine, first.

Also, what is the normal way to acknowledge the original software maintainer? -put his/her name, and maybe e-mail, in the description? Or are there Makefile variables for those? I guess I could browse more other package Makefiles, to see more examples. Tobias has the code here, on GitHub (https://github.com/peervpn/peervpn), and an e-mail address on peervpn.net. Maybe it would be good for me to contact him.

bobafetthotmail, 2016-10-27:
>I wondered about that, because I don't see an obvious VPN submenu in my web interface.

That submenu is for the **make menuconfig** CLI interface.

jp-bennett, 2016-10-27:
> So how does the maintainer thing work? Would I make pull requests, or would I get push access to this repo? Would it get pushed out to a/your build farm, and get tested to compile on all the [...]

For those of us that only deal with one or two packages, we usually just use pull requests, rather than having full commit access.

The packages get built by the openwrt build system, so no need for you to compile for a bunch of platforms.  You would want to compile through the Openwrt/Lede build system for at least one platform to confirm it's working as expected.  I have started building Lede for x86 and running in a VM for testing package builds.

> Also, what is the normal way to acknowledge the original software maintainer? -put his/her name, and maybe e-mail, in the description? Or are there Makefile variables for those? I guess I could browse more other package Makefiles, to see more examples. 

I don't know of a makefile variable, but I would think including him in the description would be reasonable.

> Tobias has the code here, on GitHub (https://github.com/peervpn/peervpn), and an e-mail address on peervpn.net. Maybe it would be good for me to contact him.

It's usually good for maintainers and developers to be in contact, so emailing him would be good.

hnyman, 2016-10-27:
Usually the original software maintainer is not specifically mentioned (or even contacted). The Makefile item URL:=http://www.peervpn.net/ leads to the upstream site of the package and they have the needed references there. And you should define the LICENSE variable, too. See https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md 

Like jp-bennett already said, maintainership is about keeping the Makefile (and possible patches, init scripts etc.) here updated, so that the package follows new versions in upstream. Changes are usually submitted as pull requests.

bdklahn, 2016-10-27:
Thanks for the comment @hnyman.

Yeah, I meant to find/read the/a CONTRIBUTING.md doc.

Right, URL:=http://www.peervpn.net/, can be used to get the maintainer info. But that is not always clear to a less than curious openwrt end user. It can:
1) Let end users more explicitly know who wrote the actual code (credit).
2) Filter away potentially irrelevant questions to package maintainers.

I tend to like to give more visible/explicit credit, unless someone doesn't want the added visibility.

Of course I understand those who publish under GPL, etc., 
1) should know not to maybe expect this.
2) maybe don't want their contact info broadcast at too many additional places

I think pull requests are a reasonable/fine mechanism for contribution, and code review.

OK . . . The previous maintainer has commented to me, and pointed me to a newer version of the Makefile.
I'm thinking about it.

Thanks!

closed 2018-08-17 by dibdot