openwrt/packages #3403: freeradius2: create temp directory

sideeffect42, 2016-10-27:
Maintainer: @dangowrt 
Environment: mvebu, linksys-wrt1900acv2, LEDE Reboot (HEAD, r1953)

Description:

At the moment the freeradius2 package ([/etc/init.d/radiusd](https://github.com/openwrt/packages/blob/master/net/freeradius2/files/radiusd.init) to be precise) does not create a temporary directory for use by radiusd.

radiusd requires a temporary directory to be existent for certain operations, like verification of certificates.

Could you please adjust /etc/init.d/radiusd to create a temporary
directory (preferably /tmp/radiusd)?

Adding the following line after ([/etc/init.d/radiusd:19](https://github.com/openwrt/packages/blob/master/net/freeradius2/files/radiusd.init#L19) was worked well for me:
```
	mkdir -p /tmp/radiusd
```

Thanks for your amazing work!

bobafetthotmail, 2016-11-03:
Is that folder set already in the radiusd config file or does it need to be set?

sideeffect42, 2016-11-03:
It is specified in `freeradius2-mod-eap`'s `eap.conf` on line 395, but is commented by default.

Context:
``` /etc/freeradius2/eap.conf
			#
			#  As of version 2.1.10, client certificates can be
			#  validated via an external command.  This allows
			#  dynamic CRLs or OCSP to be used.
			#
			#  This configuration is commented out in the
			#  default configuration.  Uncomment it, and configure
			#  the correct paths below to enable it.
			#
			verify {
				#  A temporary directory where the client
				#  certificates are stored.  This directory
				#  MUST be owned by the UID of the server,
				#  and MUST not be accessible by any other
				#  users.  When the server starts, it will do
				#  "chmod go-rwx" on the directory, for
				#  security reasons.  The directory MUST
				#  exist when the server starts.
				#
				#  You should also delete all of the files
				#  in the directory when the server starts.
		#     		tmpdir = /tmp/radiusd

				#  The command used to verify the client cert.
				#  We recommend using the OpenSSL command-line
				#  tool.
				#
				#  The ${..CA_path} text is a reference to
				#  the CA_path variable defined above.
				#
				#  The %{TLS-Client-Cert-Filename} is the name
				#  of the temporary file containing the cert
				#  in PEM format.  This file is automatically
				#  deleted by the server when the command
				#  returns.
		#    		client = "/path/to/openssl verify -CApath ${..CA_path} %{TLS-Client-Cert-Filename}"
			}
```

bobafetthotmail, 2016-11-03:
hm, ok. It says "This configuration is commented out in the default configuration." so that's what it expects by default.

I'll send a pull request to add your line to the initscript then, the mantainer will eventually show up to merge it (if he likes the default folder, if not I'll have to change the default config file patches too).

dangowrt, 2016-11-17:
Please report back if the now merged fix solved the problem. If so, please close this ticket.

sideeffect42, 2016-11-21:
Works.

closed 2016-11-21 by sideeffect42