openwrt/packages #3405: chardet (probably not properly reading new HOST_BUILD to python)

build000, 2016-10-27:
make -j8 BUILD_LOG=1
[ERROR]

make -j1 V=s
```
(...)
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4300/LEDE/feeds/packages/lang/chardet'
. /home/sj/Pulpit/Projekty/WNDR4300/LEDE/include/shell.sh; gzip -dc /home/sj/Pulpit/Projekty/WNDR4300/LEDE/dl/chardet-2.3.0.tar.gz | tar -C /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0
touch /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.prepared_7a3bbd3ce34ae2dc96ad78e8e28d53c7
rm -f /home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.chardet_installed
(cd /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/./; if [ -x ./configure ]; then find /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ -name config.guess | xargs -r chmod u+w; find /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ -name config.guess | xargs -r -n1 cp --remove-destination /home/sj/Pulpit/Projekty/WNDR4300/LEDE/scripts/config.guess; find /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ -name config.sub | xargs -r chmod u+w; find /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ -name config.sub | xargs -r -n1 cp --remove-destination /home/sj/Pulpit/Projekty/WNDR4300/LEDE/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0:chardet-2.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0:chardet-2.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16 " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0:chardet-2.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16 " CPPFLAGS="-I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include " LDFLAGS="-L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   _python_sysroot="/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15" _python_prefix="/usr" _python_exec_prefix="/usr" ; fi; )
rm -f /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.configured_*
touch /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.configured_yyyyyyy
install -d -m0755 /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ipkg-install//usr/lib/python2.7/site-packages
if [ "" == "HOST" ]; then export PYTHONPATH="/lib/python2.7://lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/lib/python2.7:/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15//usr/lib/python2.7/site-packages:/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";   cd /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/; CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -fpic" CXX="mips-openwrt-linux-musl-g++" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0:chardet-2.3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" CPPFLAGS="-I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include -I/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/include/python2.7" LDFLAGS="-L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/lib -L/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"    /bin/python2.7   ./setup.py install --prefix=/usr --root=/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/ipkg-install ;
bash: /bin/python2.7: No such file or directory
Makefile:53: recipe for target '/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/chardet-2.3.0/.built] Error 127
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4300/LEDE/feeds/packages/lang/chardet'
package/Makefile:127: recipe for target 'package/feeds/packages/chardet/compile' failed
make[2]: *** [package/feeds/packages/chardet/compile] Error 2
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4300/LEDE'
package/Makefile:124: recipe for target '/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/sj/Pulpit/Projekty/WNDR4300/LEDE/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4300/LEDE'
/home/sj/Pulpit/Projekty/WNDR4300/LEDE/include/toplevel.mk:194: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

build000, 2016-10-27:
...and any other python packages, eq. python-crcmod:
```
cat logs/package/feeds/packages/python-crcmod
bash: /bin/python2.7: No such file or directory
Makefile:50: recipe for target '/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/crcmod-1.7/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4300/LEDE/build_dir/target-mips_24kc_musl-1.1.15/crcmod-1.7/.built] Error 127
```
Generally its the same error to other
```
bash: /bin/python2.7: No such file or directory
```

hnyman, 2016-10-28:
Relevant discussion in the PR:  https://github.com/openwrt/packages/pull/3297#issuecomment-256746969

build000, 2016-10-29:
@hnyman 
Significant discussion unless stuck in place - you still can not build a package python / dependent ... well unless first some patches would be appropriate to restore the ability to build - then you can consider which approach is appropriate.

hnyman, 2016-10-31:
This has likely been fixed by #3423

build000, 2016-10-31:
But perhaps something is not quite right?
```
make package/symlinks V=s
./scripts/feeds update -a
Updating feed 'packages' from 'https://git.lede-project.org/feed/packages.git' ...
remote: Counting objects: 60, done.
remote: Compressing objects: 100% (56/56), done.
remote: Total 60 (delta 38), reused 1 (delta 0)
Unpacking objects: 100% (60/60), done.
From https://git.lede-project.org/feed/packages
   066b631..07df9e3  master     -> origin/master
Updating 066b631..07df9e3
Fast-forward
 lang/python-urllib3/Makefile          |  6 +++---
 lang/python/files/python-host.mk      |  3 +++
 lang/python3/files/python3-package.mk |  3 +++
 lang/simplejson/Makefile              |  6 +++---
 lang/vala/Makefile                    | 36 ++++++++++++++++++------------------
 net/strongswan/Makefile               |  7 ++++---
 utils/luci-app-lxc/Makefile           |  4 ++--
 utils/luci-app-lxc/files/view/lxc.htm |  2 +-
 8 files changed, 37 insertions(+), 30 deletions(-)
Create index file './feeds/packages.index' 
Collecting package info: done
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.lede-project.org/feed/routing.git' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.lede-project.org/feed/telephony.git' ...
Already up-to-date.
Create index file './feeds/telephony.index' 
./scripts/feeds install -a
Collecting package info: done
Collecting target info: done
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
sj@slawek:~/Pulpit/LEDE-WNDR4300$ make dirclean
 make[1] dirclean
sj@slawek:~/Pulpit/LEDE-WNDR4300$ make menuconfig
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'libssl'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'md5sum'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
[... 174 lines cut ...]
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/download'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/download'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/prepare'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/prepare'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/compile'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/compile'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/install'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/install'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/update'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/update'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/refresh'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/refresh'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/prereq'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/prereq'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/dist'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/dist'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/distcheck'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/distcheck'
package/Makefile:127: warning: overriding recipe for target 'package/feeds/packages/python3/host/configure'
package/Makefile:127: warning: ignoring old recipe for target 'package/feeds/packages/python3/host/configure'

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

```

hnyman, 2016-10-31:
Yeah, I already mentioned a bit different python3 issues that to @commodo 
Apparently the PR  #3297 from @NeoRaider and the later fix #3423 are not yet perfect.

hnyman, 2016-11-01:
This should have been fixed by #3433

closed 2016-11-01 by hnyman