openwrt/packages #3407: freeradius crashes from requests from slaved AP

gruselbauer, 2016-10-28:
-------------------------------

Maintainer: @dangowrt 
Environment: TP-Link Archer C7 v2 (OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)) + TP-Link TL-WR841N/ND v10 (OpenWrt Chaos Calmer 15.05.1 / LuCI (git-16.018.33482-3201903))

Description:

Running freeradius on a TP-Link Archer C7 v2 with a TP-Link 841nd as a secondary AP behind it. The server works fine when auth Requests are coming from localhost by WiFi clients on the Archer C7 or when I test with radclient from clients connected directly to the AP. All fine so far.

*However*, when I set the 841nd up as an AP using PEAP with correct secret and everything - no iptables rules of its own, dumb AP as described in OpenWRT docu - and try to connect clients, freeradius crashes and doesn't respond to any client requests from any source anymore.

Reproducibly, radiusd -XX starts printing "Ready to process requests." a lot (~50x a second?) as soon as I try connecting to the server. Ultimately it stops responding to legitimate requests from anywhere in the network at this point.

When I connected a client to the AP via ethernet, I can also authenticate to the server without any trouble. Config files are as follows:

_clients.conf_
`client localhost {
	shortname = localhost
	ipaddr = 127.0.0.1
	secret = supersecretsecretgoeshere
	require_message_authenticator = no
	nastype = other
}
client 10.0.0.0/24 {
	shortname = lan-clients
	secret = thisisevenmoresecret
	require_message_authenticator = no
	nastype = other
}`
_radiusd.conf_
`prefix = /usr
exec_prefix = /usr
sysconfdir = /etc
localstatedir = /var
sbindir = /usr/sbin
logdir = /var/log
raddbdir = /etc/freeradius2
radacctdir = /var/db/radacct
name = radiusd
confdir = ${raddbdir}
run_dir = ${localstatedir}/run
db_dir = ${raddbdir}
libdir = /usr/lib/freeradius2
pidfile = ${run_dir}/${name}.pid
max_request_time = 30
cleanup_delay = 5
max_requests = 1024

listen {
	type = auth
	ipaddr = *
	port = 1812
}
hostname_lookups = no
allow_core_dumps = no
regular_expressions	= yes
extended_expressions	= yes
log {
	destination = files
	file = ${logdir}/radius.log
	syslog_facility = daemon
	stripped_names = no
	auth = yes
	auth_badpass = no
	auth_goodpass = no
}

checkrad = ${sbindir}/checkrad
security {
	max_attributes = 200
[... 650 lines cut ...]
	pap
}
authenticate {
	Auth-Type PAP {
		pap
	}
	Auth-Type CHAP {
		chap
	}
	Auth-Type MS-CHAP {
		mschap
	}
	eap
}
accounting {
	exec
}
session {
	radutmp
}
`

The radius.log file doesn't show any particular error. If run in debug mode, it simply starts firing "Ready to process requests" until it crashes.

Happy to supply any additional information needed.

nerilex, 2017-07-08:
According to this [forum post in german](https://forum.ubuntuusers.de/topic/freeradius-antwortet-nur-auf-lokalem-rechner/) regarding a similar problem on Debian / Ubuntu it seems to be a problem inside the kernels UDP handling, which is already fixed in newer kernels[1].

[1]:  #808293 - A regression in the UDP implementation prevented freeradius and some other applications from receiving data. (https://www.debian.org/security/2016/dsa-3434)

dangowrt, 2018-07-30:
Please reopen if the issue still exists with current builds

closed 2018-07-30 by dangowrt