openwrt/packages #3424: DDNS scripts infinite loop

yurtesen, 2016-10-31:
In luci it is possible to set ddns scripts to retry indefinitely (retry_count). However this option seems to set each action's retry number instead of setting the retry number for the whole script. (which is infinite by default and it never exits anyway).

If I have the retry number set to 0 (infinite retries) then the ddns script starts by first trying to get the current IP set in DNS by doing an nslookup.

The problem is, if the IP was not set previously (inactive/expired in DNS or first run) then nslookup returns error and script goes into an infinite loop. Running nslookup again and again, expecting a different result, which is impossible. It is impossible because it has to update the IP in DNS first for nslookup to succeed.

I do not get why it has to have a setting to run each action infinitely. Is there any point? because script will loop and come back to do same thing again when everything fails/succeeds. I tried to find some threads/reports about why this "feature" was implemented and could not find anything.

But if it has to that, then it should at least try update dns before trying nslookup again. Perhaps like 'if 2nd try, first do update, wait for a while, then try nslookup'. Otherwise it will sometimes enter an infinite loop. It is a bug if it can't exit this infinite loop without user intervention.

Thanks,

yurtesen, 2016-10-31:
Well can you put a link to this community request? I am interested to see exactly what was the request. Thank you.

Because I check the documentation and the feature has nothing to do with what you are explaining.
https://wiki.openwrt.org/doc/uci/ddns
"Since CC 15.05 Number of retries before the script terminates execution, when a communication error happen. :!: A setting of '0' will retry infinite."

I can't make any connection between that description and what you explained?

So, I still don't understand this so called feature, if you don't want to make updates for non existing dns records, you can put a checkbox saying "require dns record to exist before update". Then I can easily deduct, why my ddns is not working. On the other hand, it can also be implemented as "exit after X update rejections". Meanwhile it is in infinite loop now.

Now, I had to debug the issue to figure out why my ddns was not working. Yes, I had it working originally, but for an unknown reason it stopped after working for months. 

This is not user friendly, it is impossible to understand what is the purpose of this retry count. I find it really confusing.

chris5560, 2016-11-01:
The original idea was to stop i.e. after 5 retry's. 
For easy implementation of "0" for infinite the log message was not changed and it still count retry 1/0, 2/0 etc.
LuCI allows you to set all available configuration options including to show log file.
So nothing to debug.
I think there is nothing more possible to show in log. 
One of the next releases will retry not on every communication step.
It will retry/check on the whole check/update loop.
So retry timer will be removed because there is the check interval doing the job.
The retry counter will stay but it possibly will only show a final message on exit not on every error loop.
It will be still required to complete the setup at the DDNS provider (setup an initial ipv4/ipv6) so the world already knows about the domain before ddns-scripts will send out updates.
Something like a "force" option makes no sense because it's only needed once (the first update).
I will extend the "Requirements" section of OpenWrt Wiki the next days, that it is needed to fully complete the initial setup at the DDNS provider because ddns-scripts will not work correctly.

chris5560, 2016-12-15:
@yurtesen 
could be closed ?

yurtesen, 2016-12-15:
@chris5560 , sorry I full l forgot to reply. To be honest, I found the logs tab after your message. I still think that the way the script works is strange :) I can see how it is useful to not spam ddns providers. Anyway, you can close the ticket, the issue was not so important anyway.

closed 2016-12-15 by hnyman