openwrt/packages #3430: instsall openvswitch

Peng-Hua, 2016-11-01:
Input the command "./scripts/feeds install -p openvswitch"

Why does appear ：
package/Makefile:196: warning: overriding commands for target ‵package/feeds/packages/python3/host/clean'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/clean'
package/Makefile:196: warning: overriding commands for target `package/feeds/packages/python3/host/download'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/download'
package/Makefile:196: warning: overriding commands for target `package/feeds/packages/python3/host/prepare'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/prepare'
package/Makefile:196: warning: overriding commands for target `package/feeds/packages/python3/host/compile'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/compile'
package/Makefile:196: warning: overriding commands for target `package/feeds/packages/python3/host/install'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/install'
package/Makefile:196: warning: overriding commands for target `package/feeds/packages/python3/host/update'
package/Makefile:196: warning: ignoring old commands for target `package/feeds/packages/python3/host/update'

It did not appear this problem 3 days ago,
I would like to ask what the problem is,
Thanks a lot.

yousong, 2016-11-01:
It is caused by `host-build.mk` being included twice by python3 package and `BUILD_TYPES` have two duplicate values `host`.

yousong, 2016-11-01:
ping python3's current maintainer ;) @commodo

commodo, 2016-11-01:
yeah ; my bad for that too
will fix :)

[ when it rains, it pours [i guess] ]

Peng-Hua, 2016-11-01:
Thank you so much~ : )

yousong, 2016-11-01:
> [ when it rains, it pours [i guess] ]

Well, I am thinking of a similar saying in Chinese "屋漏偏逢连夜雨", which means literally "the rain pours for days without ending when the roof was just broken and leaky" ;)

That said, maybe we can add a include guard for host-build.mk just like what has been done for target.mk.  Even a warning can help if including twice the Makefile is not considered a good practice.

commodo, 2016-11-01:
was also thinking about a guard like that too ; tho, I am/was unsure whether to propose it, since it's in LEDE core
i think for python packages it may also make sense to do this 
so, i'll do them for python-*.mk files [and adapt it from target.mk, or maybe just wing it]

commodo, 2016-11-01:
so i was wondering, if we should reset all these python,python3 issues to moving 
```
HOST_BUILD_PREFIX:=$(if $(IS_PACKAGE_BUILD),$(STAGING_DIR)/host,$(STAGING_DIR_HOST))
```
in rules.mk  (away from host-build.mk) and not having to include host-build.mk in python-package.mk/python3-package.mk 

this story started with the change of HOST_BUILD_PREFIX;
and it looks like `STAGING_DIR`, `STAGING_DIR_HOST` & `IS_PACKAGE_BUILD` is part of `rules.mk`, which makes HOST_BUILD_PREFIX a shorthand for these 3

sorry, for not noticing this earlier

@hnyman @yousong @NeoRaider ?

commodo, 2016-11-01:
RFC: https://github.com/lede-project/source/pull/465

commodo, 2016-11-01:
referencing PR from @jow- https://github.com/openwrt/packages/pull/3433
when this gets merged, it should fix the issue

PR #3433 (2016-11-01): python, python3: use STAGING_DIR_HOSTPKG
Maintainer: @commodo 
Compile tested: LEDE 8bf4105a71387ef502731dd3e987e82f45daad89 / r2080
Run tested: -
Description:

Currently, the python host build framework relies on HOST_BUILD_PREFIX to
refer to the $(STAGING_DIR)/host directory but using the HOST_BUILD_PREFIX
variable requires the use of include/host-build.mk which in turn includes
the python host build makefiles, leading to target redefinition errors.

In order to provide a global, uniform way to refer to the host staging
directory, LEDE introduced a new variable STAGING_DIR_HOSTPKG which poin [... cut]
closed 2016-11-01 by jow-