openwrt/packages #3437: Unbound: cosmetic - whitespaces being printed

Borromini, 2016-11-01:
Maintainer: @EricLuehrsen 
Environment: ramips/mt7621, LEDE 2063

Hi  Eric,

Nice work on the recent updates.

Upon stopping (/etc/init.d/unbound stop), the init script prints a few white lines:

`# /etc/init.d/unbound stop

root@lede:~#`

All works as expected, I checked the init script but nothing obvious there pointing to any redundant whitespaces.

EricLuehrsen, 2016-11-01:
The current release is **only** to maintain "carryover" function, but protect users flash ROM from excessive wear out. The full boat UCI is structured in. Some script interpreters do not like empty function calls. **echo** is a stub. It will print to the command line, but as empty, should not print to syslog or create clutter when automated. I could do an incremental update with alternate stubs, or just wait for the full UCI. Let me know.

Note: _rootzone_update() is executed on a stop. The stop is a regular event during "hotplug ifup" from ISP renumbering, modem resets, netifd noise, and such. I have found that Unbound gets the trust anchor amazingly fast during the gaps will the network interfaces are being dis/re/assembled at boot. If you `touch -d 2016-01-01 /etc/unbound/root.key` and `reboot`, then you will see the update for ~300 days old in syslog and new data in /etc/~_

Borromini, 2016-11-02:
Okay, thanks for clarifying.

closed 2016-11-02 by Borromini