openwrt/packages #3438: gcc: Incorrect configure option

dl12345, 2016-11-01:
Maintainer:  gccmaintain@noblepepper.com
Environment: (trunk, all)

Description:

--with-local-prefix=/usr is explicitly tagged in the docs as an incorrect configure option. Please see [https://gcc.gnu.org/install/configure.html](url) for details.

> Do not specify /usr as the --with-local-prefix! The directory you use for --with-local-prefix must not contain any of the system's standard header files. If it did contain them, certain programs would be miscompiled (including GNU Emacs, on certain targets), because this would override and nullify the header file corrections made by the fixincludes script.

noblepepper, 2016-11-06:
What effects are you experiencing? Does this cause you problems or is just "not correct"? I am not having issues but haven't done extensive testing, this probably is because we aren't on a normal linux system and there aren't any other header files (that I can find anyway) to get messed up.

I have this and another -
"I do not think the flag --with-default-libstdcxx-abi=gcc4-compatible is needed. Toolchain does not use it, so gcc package does not need it." 
on my list but it will be a while before I can test these. What changes do you see with --with-local-prefix set to something else (and what do you use/suggest)?

Thank, Noble

dl12345, 2016-11-07:
I am not experiencing an issue with this flag since I'm using GCC 6.2.0 in my toochain and package. I backported the LEDE toolchain config into my CC system and then modified your GCC 5 package for GCC 6.2.0.

However, I noticed this option in the GCC 5.x package while building my own package and thought I would signal it because of the way it is so clearly labelled in the GCC docs as erroneous. I think for many packages it will be ok, but the GCC docs do clearly state that it will cause miscompilation in certain apps. I think anyway that having headers of site specific packages installed in /usr/local/ rather than /usr is a good thing. It separates system headers from site specific headers and lets them be removed easily if desired.

The ABI compatibility: agreed. Not necessary. It depends how the toolchain and libc is compiled. They should match. Changing the ABI in a way they don't match will cause linker errors. 

Something else to mention to you is that I suspect that this GCC 5 package will have some compilation problems for x86_64 build systems running Redhat / Centos / Fedora that are building for x86_64 targets because there is no with-build-sysroot or --with-headers option specified. This results in the gcc configure script enagaging in some build-target system contamination and pulling in a sdt.h header from the build system include folders. This then errors out in the build because the sdt.h header isn't available in the compilation tree. Christian Bruel previously had this problem because I noticed a couple of old email list messages referring to it 

I had to fix this via a patch to the gcc configure script, which you will probably need to modify for GCC 5.x (original patch credit to Christian - this is how he solved it as well).

```
--- a/gcc/configure
+++ b/gcc/configure
@@ -12274,6 +12274,8 @@ if test x$host != x$target || test "x$TA
   else
     target_header_dir="${with_sysroot}${native_system_header_dir}"
   fi
+elif test x$build != x$host && test "x$with_build_sysroot" != "x"; then
+    target_header_dir="${with_build_sysroot}${native_system_header_dir}"
 else
   target_header_dir=${native_system_header_dir}
 fi
```

For reference, these are the options I am using to configure GCC 6.2.0. Note also that I had to use the ldflags options because the GCC configure scripts insist on trying to link to libgcc statically. For OpenWrt systems, there is no static libgcc in the staging area, so I was getting a compilation error. 

```
define Build/Configure
    (cd $(PKG_BUILD_DIR); rm -f config.cache; \
        SHELL="$(BASH)" \
        $(TARGET_CONFIGURE_OPTS) \
        $(PKG_BUILD_DIR)/configure \
            $(CONFIGURE_ARGS) \
            --build=$(GNU_HOST_NAME) \
            --host=$(REAL_GNU_TARGET_NAME) \
            --target=$(REAL_GNU_TARGET_NAME) \
            --enable-languages=$(TARGET_LANGUAGES) \
            --with-bugurl=$(BUGURL) \
            --with-pkgversion="$(PKGVERSION)" \
            --with-build-sysroot="$(TOOLCHAIN_DIR)" \
            --enable-shared \
            --enable-threads \
            --enable-lto \
            --with-stage1-ldflags="-static-libstdc++ -shared-libgcc" \
            --with-boot-ldflags="-static-libstdc++ -shared-libgcc" \
            --with-gnu-ld \
            $(if $(CONFIG_LIBC_USE_GLIBC),--enable,--disable)-__cxa_atexit \
            --disable-nls \
            --disable-libmudflap \
            --disable-multilib \
            --disable-libgomp \
            --disable-libssp \
            --disable-libsanitizer \
            --disable-decimal-float \
            --disable-libstdcxx-pch \
            --with-host-libstdcxx=-lstdc++ \
            --prefix=/usr \
            --libexecdir=/usr/lib \
            $(SOFT_FLOAT_CONFIG_OPTION) \
            $(call qstrip,$(CONFIG_EXTRA_GCC_CONFIG_OPTIONS)) \
    );
endef
```

noblepepper, 2017-03-16:
Fixed by [3195](https://github.com/openwrt/packages/pull/3915) this issue needs to be closed

closed 2017-03-16 by Noltari