openwrt/packages #3449: mpd fails to install boost-fiber and boost-locale dependencies

Be-ing, 2016-11-02:
Maintainer: @thess 
Environment: Linksys WRT1200AC, LEDE build

Description:
```# opkg install mpd
Installing mpd-full (0.19.19-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/mpd-full_0.19.19-1_arm_cortex-a9_vfpv3.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mpd-full:
 *      boost-fiber *   boost-locale * 
 * opkg_install_cmd: Cannot install package mpd.```

Related: issue #3239 and PR #3266

ClaymorePT, 2016-11-02:
Boost.Fiber is marked as BROKEN due to issues in the Boost.Build. 
This has been reported upstream. https://github.com/boostorg/build/issues/147

thess, 2016-11-02:
MPD didn't require these before boost 1.62 update. I'll see what I can do short of rolling back boost to 1.61.

ClaymorePT, 2016-11-02:
Could it be this piece of code in the makefile?
`		$(foreach lib,$(BOOST_LIBS), \
			config PACKAGE_boost-$(lib)
			prompt "Boost $(lib) library."
			$(if $(findstring locale,$(lib)),depends on @BUILD_NLS,)
			$(if $(findstring fiber,$(lib)),depends on BROKEN,)
		)
`

thess, 2016-11-03:
@ClaymorePT - check out attempt to fix this #3458

thess, 2016-11-08:
MPD and BOOST now building and producing good packages - closing this

See 8a98b22a00b046c85a09076bff795b7cb3bc0e8f

closed 2016-11-08 by thess