openwrt/packages #3453: jamvm compile error

taeasy, 2016-11-03:
Making all in i386
make[8]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src/os/linux/i386'
/bin/bash ../../../../libtool  --tag=CC   --mode=compile i486-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../../src  -I../../../../src -I../../../../src -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include   -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0:jamvm-2.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT init.lo -MD -MP -MF .deps/init.Tpo -c -o init.lo init.c
OpenWrt-libtool: compile:  i486-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../../src -I../../../../src -I../../../../src -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0:jamvm-2.0.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT init.lo -MD -MP -MF .deps/init.Tpo -c init.c -o init.o
init.c:22:25: fatal error: fpu_control.h: No such file or directory
compilation terminated.
Makefile:377: recipe for target 'init.lo' failed
make[8]: *** [init.lo] Error 1
make[8]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src/os/linux/i386'
Makefile:436: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src/os/linux'
Makefile:357: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src/os'
Makefile:666: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src'
Makefile:421: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/src'
Makefile:388: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0'
Makefile:67: recipe for target '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/.built' failed
make[2]: *** [/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/jamvm-2.0.0/.built] Error 2
make[2]: Leaving directory '/home/main/x86/feeds/packages/lang/jamvm'
package/Makefile:196: recipe for target 'package/feeds/packages/jamvm/compile' failed
make[1]: *** [package/feeds/packages/jamvm/compile] Error 2
make[1]: Leaving directory '/home/main/x86'
/home/main/x86/include/toplevel.mk:192: recipe for target 'package/feeds/packages/jamvm/compile' failed
make: *** [package/feeds/packages/jamvm/compile] Error 2
main@debian:~/x86$

yousong, 2016-11-09:
Ping current maintainer of jamvm @danak6jq

danak6jq, 2016-11-09:
>
> Ping current maintainer of jamvm @danak6jq <https://github.com/danak6jq>

I'll take a look.

danak6jq, 2016-11-09:
I've kicked-off a build of the main dev trunk; in the meantime, what trunk/branch of OpenWRT are you building? Did you confirm that you have fpu_control.h installed? (it's part of glibc IIRC)

danak6jq, 2016-11-09:
Indeed, I've confirmed this error (building OpenWRT trunk on Ubuntu). It is a Linux-specific bit of code to switch the FPU into double precision in the base JamVM code. fpu_control.h is part of glibc, which OpenWRT is not built with (that I can tell), instead, uclibc is. I'll have to take a closer look at this and ping the JamVM author perhaps for some guidance.

taeasy, 2016-11-09:
musl patch:

--- a/src/os/linux/i386/init.c
+++ b/src/os/linux/i386/init.c
@@ -21,14 +21,14 @@
 
-#include <fpu_control.h>
+#include <stdint.h>
 
 /* Change floating point precision to double (64-bit) from
  * the extended (80-bit) Linux default. */
 
 void setDoublePrecision() {
-    fpu_control_t cw;
-
-    _FPU_GETCW(cw);
-    cw &= ~_FPU_EXTENDED;
-    cw |= _FPU_DOUBLE;
-    _FPU_SETCW(cw);
+    uint16_t cw;
+
+    __asm__ __volatile__ ("fnstcw %0" : "=m" (*&cw));
+    cw &= 0xF0C0;
+    cw |= 0x003F;
+    __asm__ __volatile__ ("fldcw %0" : : "m" (*&cw));
 }

closed 2016-11-09 by taeasy

hnyman, 2016-11-09:
This might be the needed musl compatibility patch for jamvm:
https://sourceforge.net/p/jamvm/patches/6/
https://patchwork.ozlabs.org/patch/626662/