openwrt/packages #3454: ruby-gems: gem install 'package' command fails, any ideas why?

valentt, 2016-11-03:
ruby-gems: gem install 'package' command fails, any ideas why?

Is this related to gems-openssl package or libopenssl package?

luizluca, 2016-11-03:
@valentt could you please give some more info? Which package are you trying to install and which ruby-* OpenWRT package you have already installed? For remote installation of gem, you might need ruby-openssl

valentt, 2016-11-03:
Hi @luizluca, sure and thanks for quick response. I'm trying to get puppet or chef installed from gems, so I first installed "ruby and ruby-gems" packages without problems, but when I try to install gem this is the response I get:

```
gem install puppet
ERROR:  Loading command: install (LoadError)
	failed to load encoding (Windows-31J)
ERROR:  While executing gem ... (NoMethodError)
    undefined method `invoke_with_build_args' for nil:NilClass
```

Searching online I found this [link](http://stackoverflow.com/questions/23350388/cant-install-gems-because-undefined-method-invoke-with-build-args-for-nilni#27740989)

luizluca, 2016-11-04:
It seems you need ruby-enc-extra. Ruby was split into tiny packages in order to allow its installation even on a few hundreds of kbytes. Dependencies between the OpenWRT ruby packages should be correct but something might be missing for external software as gem dependencies don't mix with OpenWRT ones. This division is not normal for a standard ruby installation.

If storage space is not a problem, simply install ruby-stdlib and you'll get full ruby environment.

Let me know if you still can't install what you need

valentt, 2016-11-04:
Thanks @luizluca, I really appreciate your quick answers. After installing ruby-enc-extra here is what I get as a response:

```
gem install puppet
ERROR:  Could not find a valid gem 'puppet' (>= 0), here is why:
          Unable to download data from https://rubygems.org/ - SSL_connect returned=1 errno=0 state=error: (null) (https://api.rubygems.org/specs.4.8.gz)

```

valentt, 2016-11-04:
After an hour it was still at 6% so I had to kill ruby2.2-bin process :(

Then I tried to install chef and got a very long backtrace: [http://pastebin.com/raw/vjpEPkQW](http://pastebin.com/raw/vjpEPkQW)

valentt, 2016-11-07:
@luizluca is https not working "normal" because of some missing component ?

luizluca, 2016-11-07:
@valentt , lets go step by step:

1) gem uses the Windows-31J encoding which is not provided by the very slim ruby-enc (but by ruby-enc-extra). If you downloaded the gem and installed it, it should work without it.

2) gem requires OpenSSL for any remote operation (even if source is http). So, install ruby-openssl. At this point, you should be able to install using '--source http://...' as you did.

3) SSL still fails because there is no ca certificate installed. Install ca-certificates and you might have all you need for "gem install xxx"

So: 

```
root@OpenWrt:/# opkg install ruby-gems ruby-enc-extra ruby-openssl ca-certificates`
...
root@OpenWrt:/# gem install puppet`
Fetching: json_pure-1.8.3.gem (100%)
Successfully installed json_pure-1.8.3
Fetching: hiera-3.2.2.gem (100%)
Successfully installed hiera-3.2.2
Fetching: facter-2.4.6.gem (100%)
Successfully installed facter-2.4.6
Fetching: puppet-4.8.0.gem (100%)
Successfully installed puppet-4.8.0
Parsing documentation for json_pure-1.8.3
Installing ri documentation for json_pure-1.8.3
Parsing documentation for hiera-3.2.2
Installing ri documentation for hiera-3.2.2
Parsing documentation for facter-2.4.6
Installing ri documentation for facter-2.4.6
Parsing documentation for puppet-4.8.0
Installing ri documentation for puppet-4.8.0
Done installing documentation for json_pure, hiera, facter, puppet after 73 seconds
4 gems installed
```

Now, about the ugly stack you got, it is something with ruby or the compiler. 'Illegal instruction at 0x000000' tells me something broke before this error was raised (like you jumped to a function pointer with null value). Not your fault at all. What was you trying to do when you got it? It might be something specially for mips as I couldn't reproduce it on x86.

Now, about chef, It'll fail at this point:

```
Building native extensions.  This could take a while...
ERROR:  Error installing chef:
        ERROR: Failed to build gem native extension.

    /usr/bin/ruby -r ./siteconf20161107-3375-qm4g3.rb extconf.rb
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /usr/lib/ruby/gems/2.2/gems/libyajl2-1.2.0 for inspection.
Results logged to /usr/lib/ruby/gems/2.2/extensions/x86-linux/2.2/libyajl2-1.2.0/gem_make.out
```

OpenWRT normally is not ready for building things inside it. There are no *-dev(el) packages. So, any native extension will fail to buid. Even with the compiler installed, currently OpenWRT ruby package does not provide the C headers (like an ruby-dev).

If you really need native gems, you might need to [precompile](http://stackoverflow.com/questions/3667918/how-do-you-precompile-the-native-extensions-for-a-ruby-gem-for-linux) while cross-compiling the needed native gems to mips arch. I never did it but it might not be impossible (just complicated).

Is there anything left for this issue? Can I close it?

valent-hyhelp, 2016-11-09:
Thanks, yes close it.

closed 2016-11-10 by luizluca