openwrt/packages #3466: rtl-ais compile failed

taeasy, 2016-11-05:
. /home/main/x86/include/shell.sh; gzip -dc /home/main/x86/dl/rtl-ais-20151213-b805be0164475e58a73b57763a8cbbf8a2bd0ded.tar.gz | tar -C /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/.. -xf - 
touch /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/.prepared_a13dcb475cfc9a683a70e7506e00b0a7
rm -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/stamp/.rtl-ais_installed
(cd /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/./; if [ -x ./configure ]; then find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/ -name config.guess | xargs -r chmod u+w; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/ -name config.guess | xargs -r -n1 cp --remove-destination /home/main/x86/scripts/config.guess; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/ -name config.sub | xargs -r chmod u+w; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/ -name config.sub | xargs -r -n1 cp --remove-destination /home/main/x86/scripts/config.sub; AR="i486-openwrt-linux-musl-gcc-ar" AS="i486-openwrt-linux-musl-gcc -c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=i486-openwrt-linux-musl-ld NM="i486-openwrt-linux-musl-gcc-nm" CC="i486-openwrt-linux-musl-gcc" GCC="i486-openwrt-linux-musl-gcc" CXX="i486-openwrt-linux-musl-g++" RANLIB="i486-openwrt-linux-musl-gcc-ranlib" STRIP=i486-openwrt-linux-musl-strip OBJCOPY=i486-openwrt-linux-musl-objcopy OBJDUMP=i486-openwrt-linux-musl-objdump SIZE=i486-openwrt-linux-musl-size CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include " LDFLAGS="-L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro "   ./configure --target=i486-openwrt-linux --host=i486-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   ; fi; )
rm -f /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/.configured_*
touch /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/.configured_yyyyyyyy
CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include " LDFLAGS="-L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro " make -j1 -C /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/. AR="i486-openwrt-linux-musl-gcc-ar" AS="i486-openwrt-linux-musl-gcc -c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=i486-openwrt-linux-musl-ld NM="i486-openwrt-linux-musl-gcc-nm" CC="i486-openwrt-linux-musl-gcc" GCC="i486-openwrt-linux-musl-gcc" CXX="i486-openwrt-linux-musl-g++" RANLIB="i486-openwrt-linux-musl-gcc-ranlib" STRIP=i486-openwrt-linux-musl-strip OBJCOPY=i486-openwrt-linux-musl-objcopy OBJDUMP=i486-openwrt-linux-musl-objdump SIZE=i486-openwrt-linux-musl-size CROSS="i486-openwrt-linux-musl-" ARCH="i386" ;
make[3]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213'
i486-openwrt-linux-musl-gcc -c rtl_ais.c -o rtl_ais.o -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include  -I./aisdecoder -I ./aisdecoder/lib -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include/
i486-openwrt-linux-musl-gcc -c convenience.c -o convenience.o -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213:rtl-ais-20151213 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include  -I./aisdecoder -I ./aisdecoder/lib -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include/
[... cut ...]
make[2]: *** [/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/rtl-ais-20151213/.built] Error 2
make[2]: Leaving directory '/home/main/x86/feeds/packages/utils/rtl-ais'
package/Makefile:196: recipe for target 'package/feeds/packages/rtl-ais/compile' failed
make[1]: *** [package/feeds/packages/rtl-ais/compile] Error 2
make[1]: Leaving directory '/home/main/x86'
/home/main/x86/include/toplevel.mk:192: recipe for target 'package/feeds/packages/rtl-ais/compile' failed
make: *** [package/feed

hnyman, 2016-12-31:
This is likely related to 32bit x86, as both Openwrt and LEDE buildbots build it ok for 64bit x86 and other platforms.

you might test similar solutions as mentioned in #3655 

cc to maintainer: @nunojpg

nunojpg, 2017-01-23:
@taeasy Since there is no testing automation for 32bits, and I don't run 32bits personally, there is no support for this package for x86 32bit.

closed 2017-01-23 by nunojpg