openwrt/packages #3482: vsftpd for x86 compile  failed

taeasy, 2016-11-06:
make[1]: Entering directory '/home/main/x86'
make[2]: Entering directory '/home/main/x86/package/libs/toolchain'
if [ -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install
if [ -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/main/x86/package/libs/toolchain'
make[2]: Entering directory '/home/main/x86/package/libs/ocf-crypto-headers'
make[2]: Leaving directory '/home/main/x86/package/libs/ocf-crypto-headers'
make[2]: Entering directory '/home/main/x86/package/libs/zlib'
make[2]: Leaving directory '/home/main/x86/package/libs/zlib'
make[2]: Entering directory '/home/main/x86/package/libs/openssl'
make[2]: Leaving directory '/home/main/x86/package/libs/openssl'
make[2]: Entering directory '/home/main/x86/feeds/packages/net/vsftpd'
. /home/main/x86/include/shell.sh; gzip -dc /home/main/x86/dl/vsftpd-3.0.3.tar.gz | tar -C /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/.. -xf - 

Applying ./patches/001-destdir.patch using plaintext: 
patching file Makefile

Applying ./patches/002-find_libs.patch using plaintext: 
patching file Makefile

Applying ./patches/003-chroot.patch using plaintext: 
patching file tunables.c

Applying ./patches/004-disable-capabilities.patch using plaintext: 
patching file sysdeputil.c

Applying ./patches/005-disable-pam.patch using plaintext: 
patching file builddefs.h

Applying ./patches/006-musl-compatibility.patch using plaintext: 
patching file sysdeputil.c

Applying ./patches/007-CVE-2015-1419.patch using plaintext: 
patching file ls.c
patching file str.c
patching file str.h
touch /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/.prepared_64a5b9c8f97362eea159ffb06f3aa127
rm -f /home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/stamp/.vsftpd.notls_installed
(cd /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/./; if [ -x ./configure ]; then find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/ -name config.guess | xargs -r chmod u+w; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/ -name config.guess | xargs -r -n1 cp --remove-destination /home/main/x86/scripts/config.guess; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/ -name config.sub | xargs -r chmod u+w; find /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/ -name config.sub | xargs -r -n1 cp --remove-destination /home/main/x86/scripts/config.sub; AR="i486-openwrt-linux-musl-gcc-ar" AS="i486-openwrt-linux-musl-gcc -c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=i486-openwrt-linux-musl-ld NM="i486-openwrt-linux-musl-gcc-nm" CC="i486-openwrt-linux-musl-gcc" GCC="i486-openwrt-linux-musl-gcc" CXX="i486-openwrt-linux-musl-g++" RANLIB="i486-openwrt-linux-musl-gcc-ranlib" STRIP=i486-openwrt-linux-musl-strip OBJCOPY=i486-openwrt-linux-musl-objcopy OBJDUMP=i486-openwrt-linux-musl-objdump SIZE=i486-openwrt-linux-musl-size CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include " LDFLAGS="-L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro "   ./configure --target=i486-openwrt-linux --host=i486-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   ; fi; )
rm -f /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/.configured_*
touch /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/.configured_yyyyyyyy
/home/main/x86/staging_dir/host/bin/sed -i -e 's/-lcrypt -lnsl/-lcrypt/' /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/Makefile
make -C /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3 CC="i486-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LDFLAGS="-L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro" vsftpd
make[3]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3'
i486-openwrt-linux-musl-gcc -c main.c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -idirafter dummyinc
i486-openwrt-linux-musl-gcc -c utility.c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -idirafter dummyinc
i486-openwrt-linux-musl-gcc -c prelogin.c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3:vsftpd-3.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -idirafter dummyinc
i486-openwrt-linux-musl-gcc -c ftpcmdio.c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/ [... cut ...]
make[2]: *** [/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/vsftpd-notls/vsftpd-3.0.3/.built] Error 2
make[2]: Leaving directory '/home/main/x86/feeds/packages/net/vsftpd'
package/Makefile:196: recipe for target 'package/feeds/packages/vsftpd/compile' failed
make[1]: *** [package/feeds/packages/vsftpd/compile] Error 2
make[1]: Leaving directory '/home/main/x86'
/home/main/x86/include/toplevel.mk:192: recipe for target 'package/vsftpd/compile' failed
make: *** [package/vsftpd/compile] Error 2
main@debian:~/x86$

yousong, 2016-11-09:
Hi, @taeasy , please try following the template when preparing the issue description, at least formatting buildlog properly!  Otherwise these issues will sit here rotting without people even wanting to take a second look.

DazzyWalkman, 2016-11-10:
Work around: removing all F_SETLK64 and F_SETLKW64 arguments in ptracesandbox.c of the vsftpd source.
It's been a long time issue. I always get around this by the above fix, and the resulting vsftpd binaries work like charm. 
It may introduce some bug. However, digging further into this issue is totally beyond me, let alone submitting a patch.
Hope it helps.

hnyman, 2016-12-31:
likely fixed by #3655

closed 2016-12-31 by hnyman