openwrt/packages #3487: wr740n v3/v4 China version.i have install openwrt ,but when i install pyton it says this :

1623628223, 2016-11-07:
Please make sure that the issue subject starts with `<package-name>: ` so that it's easily identifiable.

This repo here is ONLY for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt: https://dev.openwrt.org/newticket
 - LEDE: https://bugs.lede-project.org/
 - Most LuCI packages: https://github.com/openwrt/luci/issues

Thanks for your contribution
Please remove this text (before ---) and fill the following template
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWRT/LEDE version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

1623628223, 2016-11-07:
root@OpenWrt:/tmp# cd drcom_shell/
root@OpenWrt:/tmp/drcom_shell# opkg install *.ipk
Unknown package 'libffi'.
Unknown package 'python-mini'.
Unknown package 'zlib'.

hnyman, 2016-11-07:
To my knowledge there is no "drcom_shell" directory in the standard Openwrt distro. Looks like you are trying to install something from a private OEM repo and that package is missing dependencies. You might be able to solve it by "opkg update" if you are using the standard Openwrt package repo.

The issue report is so vague and unclear that I am closing it. Likely the is issue is not a problem with Openwrt packages themselves.

You might ask help in the forum, as it might be better suited for end-user support. Or if this is specific to "drcom", ask help at their forum (or whatever).

closed 2016-11-07 by hnyman