openwrt/packages #3495: boost-python and boost-python3 conflict

Be-ing, 2016-11-08:
Maintainer: @ClaymorePT 
Environment: Linksys WRT1200AC LEDE

Description:
```
# opkg install mpd
Installing mpd-full (0.19.19-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/mpd-full_0.19.19-1_arm_cortex-a9_vfpv3.ipk.
Installing glib2 (2.49.7-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/glib2_2.49.7-1_arm_cortex-a9_vfpv3.ipk.
Installing libffi (3.2.1-2) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libffi_3.2.1-2_arm_cortex-a9_vfpv3.ipk.
Installing libattr (20160302-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libattr_20160302-1_arm_cortex-a9_vfpv3.ipk.
Installing libflac (1.3.1-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libflac_1.3.1-3_arm_cortex-a9_vfpv3.ipk.
Installing libmad (0.15.1b-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libmad_0.15.1b-3_arm_cortex-a9_vfpv3.ipk.
Installing libvorbisidec (1.0.3-20150104-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libvorbisidec_1.0.3-20150104-1_arm_cortex-a9_vfpv3.ipk.
Installing libogg (1.3.2-2) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/libogg_1.3.2-2_arm_cortex-a9_vfpv3.ipk.
Installing boost (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-libs (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-libs_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-atomic (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-atomic_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-system (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-system_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-chrono (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-chrono_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-container (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-container_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-context (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-context_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-thread (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-thread_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-coroutine (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-coroutine_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-date_time (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-date_time_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-filesystem (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-filesystem_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-graph (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-graph_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-regex (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-regex_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-iostreams (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-iostreams_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-log (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-log_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-math (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-math_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-program_options (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-program_options_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-python (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-python_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Installing boost-python3 (1.62.0-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/boost-python3_1.62.0-3_arm_cortex-a9_vfpv3.ipk.
Configuring boost-system.
Configuring boost-chrono.
Configuring boost-date_time.
Configuring boost-atomic.
Configuring boost-thread.
Configuring boost-filesystem.
Configuring boost-regex.
Configuring boost-log.
Configuring libffi.
Configuring libattr.
Configuring glib2.
Configuring libogg.
Configuring libvorbisidec.
Configuring libflac.
Configuring boost-context.
Configuring boost-coroutine.
Configuring boost-container.
Configuring boost-graph.
Configuring boost-iostreams.
Configuring boost-math.
Configuring boost-program_options.
Configuring boost-python.
Configuring boost-libs.
Configuring boost.
Configuring libmad.
Collected errors:
 * check_data_file_clashes: Package boost-python3 wants to install file /usr/lib/libboost_python3.so
        But that file is already provided by package  * boost-python
 * check_data_file_clashes: Package boost-python3 wants to install file /usr/lib/libboost_python3.so.1.62.0
        But that file is already provided by package  * boost-python
 * opkg_install_cmd: Cannot install package mpd.
```

jow-, 2016-11-08:
Both `boost-python` and `boost-python3` should add something like `PROVIDES:=boost-python-binding` and `mpd` should depend on that `boost-python-binding` instead.

thess, 2016-11-08:
No - MPD does not need anything other than boost-container which in turn has no other boost dependencies. I don't know why boost-libs is being installed at all. I checked the package dependencies and don't see why this is happening. It used to behave simply before that latest boost updates.

I think @Be-ing 's config is polluted from previously broken packages? 

Curiously, mpd-full depends on ffmpeg libraries - where are they?

ClaymorePT, 2016-11-08:
mpd package is using `+boost +boost-container` as a dependency. 
There should be only `+boost`.

thess, 2016-11-08:
I disagree - boost-container is the only boost library needed by MPD. The reason that +boost is specified is to satisfy the build dependency on the BOOST package. Without it boost-container is not built in a custom config - go figure.

I believe the current set of BOOST packages are now correctly building and the dependencies are minimal as necessary unless boost-libs (roll-up pakcage) is specified. I do think that the boost-python/3 issue when installing boost-libs may be a problem as alluded to by @jow-. That I will look into.

thess, 2016-11-08:
OK - I see the problem. I don't know why all the boost libs get installed. Stay tuned.

thess, 2016-11-08:
An accident of wild-cards in the boost/Default/install procedure copies the python3 libs into the python package (and vice-versa). I think that to be the error originally reported in this thread. The fact that boost-libs is installing when only a single Boost library is required was another issue which fortunately was highlighted by this issue. That, too will be fixed.

thess, 2016-11-09:
@Be-ing  - This problem should be fixed by #3497 - Check out builds tomorrow. When you install MPD, the only boost library installed (plus empty meta package) should be boost-container.

thess, 2016-11-10:
Installation of mpd tested with latest build results (10-Nov-2016) - OK
Closing this.

closed 2016-11-10 by thess

ClaymorePT, 2016-11-10:
Great! :+1:

Be-ing, 2016-11-10:
I confirm that mpd now installs correctly. Thank you for fixing the myriad of issues!