openwrt/packages #3502: lighttpd: missing lib when using mod_auth

amaumene, 2016-11-10:
Maintainer: @MikePetullo  
Environment: LEDE Reboot r2145

Description:
When using mod_auth with lighttpd got this error:
```
2016-11-10 11:19:25: (plugin.c.227) dlopen() failed for: /usr/lib/lighttpd/mod_authn_file.so Error loading shared library /usr/lib/lighttpd/mod_authn_file.so: No such file or directory 
2016-11-10 11:19:25: (server.c.912) loading plugins finally failed
```
I'm using htpasswd auth.

MikePetullo, 2016-11-25:
I just submitted a pull request. See https://github.com/openwrt/packages/pull/3565.

MikePetullo, 2016-11-27:
This should be fixed as hnyman merged #3565.

gstrauss, 2017-01-05:
@MikePetullo I believe this is fixed and can be closed. :)

MikePetullo, 2017-01-05:
I agree, but it does not appear that GitHub has granted me the ability to close this issue.

closed 2017-01-05 by hnyman