openwrt/packages #3530: <ddns-scripts>: cloudflare script not able to handle router.home.example.com hostname

kduvekot, 2016-11-19:
Maintainer: @chris5560 
Environment: Netgear WNDR3700, Chaos Calmer 15.05.1 r49404

Description:
Hello,

I post this here as I do not find a better place to post this question.

I have my domain (example.com) DNS service with cloudlfare. And I'm using the home.example.com subdomain for all the hosts at home. So my router FQDN is router.home.example.com, and my nas is then nas.home.example.com etc etc.

in order to have my router accessible from the internet I would like to use the ddns cloudlfare scripts (https://github.com/openwrt/packages/tree/master/net/ddns-scripts).

I have entered the details correctly as far as I can tell via the Luci screens, and in the ddns logs I also see that the script is able to query the cloudflare API to get the correct rec_id for the router.home.example.com dns A record.

the following https call is made to cloudflare:
`https://www.cloudflare.com/api_json.html?
a=rec_load_all&
tkn=1234567890abcdef1234567890abcdef&
email=email@example.com&
z=example.com
`
and I see the correct details in the logs:

rec_id '1234567890' detected for host/domain 'router.home.example.com'

so far so good ...

then the update https call is created .. and there things go wrong:

`https://www.cloudflare.com/api_json.html?
a=rec_edit&
tkn=1234567890abcdef1234567890abcdef&
id=1234567890&
email=email@example.com&
z=example.com&
type=A&
name=router&
content=1.2.3.4&
service_mode=0&
ttl=1
`
so instead of setting the name variable to 'router.home' .. the script sets it to just 'router'

When looking into the scripts I see that in the first call the cloudflare the script is using the FQDN that you configured (router.home.example.com) to find the correct rec_id from the JSON output and it is using the variable `__DOMAIN` as the zone details (see line 67 on this page : `https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/update_cloudflare_com.sh#L67`)

so far so good again ..

in the update URL it is using the `__SUBDOM` variable in the 'name' field of the update so aparently that is set incorrect.

both the `__DOMAIN` and  `__SUBDOM` variables are populated by using the split_FQDN function from here
(https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/dynamic_dns_functions.sh#L1184)

and there I see nothing really wrong with how it gets split .. so why is the '__SUBDOM' variable (parameter $4 in the `split_FQDN` function) not populated with router.home?

any help would be appreciated

Kees D.

chris5560, 2016-11-20:
@kduvekot

please close your request at LEDE project because here is the right place to handle your problem.

I retested split_FQDN() and update_cloudflare_com.sh both working correct when looking on building the url.
I really used "host.sub.example.com" and it is split into "host.sub" "example" "com"
Also the final url has "...&name=host.sub&..." so no chance for me to see your problem

I ask you to send me your log file by direct mail.
Please only remove your email and token from the log.
So I can try to analyse and find out the problem.

Christian

kduvekot, 2016-11-20:
do I need to enable some additional logging?

chris5560, 2016-11-21:
No

Mosney, 2018-07-05:
same as you. 2 years pass, looks  still not resolve, I think.

Mosney, 2018-07-05:
so do you have any idea about this issue? I can not fix it because not professional.