openwrt/packages #3541: libopus: fails to build on kirkwood

nwf, 2016-11-22:
Maintainer: @psycho-nico or @antonlacon 
Environment: kirkwood (pogoplug v4), LEDE HEAD

Description:

When attempting to cross-compile from Linux amd64 to kirkwood, I am told

```
libtool: compile:  ccache_cc -DHAVE_CONFIG_H -I. -I./include -I./celt -I./silk -I./silk/float -I./silk/fixed -I/tank/openwrt/scratch/builder-darknetgw/staging_dir/target-arm_xscale_musl-1.1.15_eabi/usr/include -I/tank/openwrt/scratch/builder-darknetgw/staging_dir/target-arm_xscale_musl-1.1.15_eabi/include -I/tank/openwrt/scratch/builder-darknetgw/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl-1.1.15_eabi/usr/include -I/tank/openwrt/scratch/builder-darknetgw/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl-1.1.15_eabi/include/fortify -I/tank/openwrt/scratch/builder-darknetgw/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl-1.1.15_eabi/include -Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap /tank/openwrt/scratch/builder-darknetgw/build_dir/target-arm_xscale_musl-1.1.15_eabi/opus-1.1.3:opus-1.1.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fvisibility=hidden -W -Wall -Wextra -Wcast-align -Wnested-externs -Wshadow -Wstrict-prototypes -MT celt/kiss_fft.lo -MD -MP -MF celt/.deps/kiss_fft.Tpo -c celt/kiss_fft.c  -fPIC -DPIC -o celt/.libs/kiss_fft.o
{standard input}: Assembler messages:
{standard input}:307: Error: first transfer register must be even -- `ldrd r5,r6,[r0]'
{standard input}:442: Error: first transfer register must be even -- `ldrd r5,r6,[r0,r4]'
{standard input}:612: Error: first transfer register must be even -- `ldrd r9,r10,[r3]'
{standard input}:643: Error: first transfer register must be even -- `ldrd r7,r8,[r6]'

```

I am not sure what I am (or the package is) doing wrong.  What more can I say to be useful?

antonlacon, 2016-11-25:
Did this happen with libopus-1.1.0? The update to 1.1.3 included a switch to fixed-point math for machines using a software fpu. I'd be interested to know if the regular floating point built for you.

Upstream is here: https://github.com/xiph/opus

thess, 2016-11-25:
It happens with v1.1.0 also if you enable fixed-point math instead of soft-float. Fixed-point math with ASM optimizations on ARMv5te produce these errors. See my comment in the PR about whose problem it is. If you can fix the ASM code (or GCC/Binutils) to use aligned regs - go for it. I gave up.

nwf, 2016-11-25:
@thess: which PR, sorry?

nwf, 2016-11-25:
What a pain; I knew I didn't want to debug this.  It looks like there is no gcc constraint suitable to express the notion of "must be even register" (https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html doesn't list one, anyway) and it looks like gcc has trouble understanding this constraint within codegen (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=66964 is still open though apparently partially fixed in gcc5).  Blah!

PR #3556 (2016-12-08): opus: Disable ASM for xscale and arm926ej-s
Maintainer: @psycho-nico 
Compile tested: Several ARM configs
Run tested: None

Description:

ASM optimization fails on ARMV5TE processors requiring even numbered registers for `ldrd` opcode. I'm not sure if this is fixable in the opus source or is really a GCC problem. This patch is a workaround for this problem.

Fixes: #3541
closed 2016-12-08 by thess