openwrt/packages #3544: acme-1.2 - not renew and create only "CN=Fake LE Intermediate X1"

build000, 2016-11-22:
log for in create and renew another:
http://wklej.org/id/2965475/
effects (last firefox 50.0 - windows 10 x64 and ubuntu-16.04.* x64:
```
******.tk uses an invalid security certificate. The certificate is not trusted because the issuer certificate is unknown. The server might not be sending the appropriate intermediate certificates. An additional root certificate may need to be imported. Error code: SEC_ERROR_UNKNOWN_ISSUER
```

tohojo, 2016-11-22:
What's the bug? That log indicates that the certificate was created using the letsencrypt staging server, so it's no surprise that Firefox doesn't recognise it...

On 22 November 2016 13:03:36 CET, build000 <notifications@github.com> wrote:
>@tohojo 
>log for in create and renew another:
>http://wklej.org/id/2965529/
>effects (last firefox 50.0 - windows 10 x64 and ubuntu-16.04.* x64):
>```
>******.tk uses an invalid security certificate. The certificate is not
>trusted because the issuer certificate is unknown. The server might not
>be sending the appropriate intermediate certificates. An additional
>root certificate may need to be imported. Error code:
>SEC_ERROR_UNKNOWN_ISSUER
>```

build000, 2016-11-22:
Then how to make trivial saying, "it was green padlock"?

tohojo, 2016-11-22:
Stop using the staging server?

On 22 November 2016 13:08:38 CET, build000 <notifications@github.com> wrote:
>Then how to make trivial saying, "it was green padlock"?

build000, 2016-11-22:
hehe - I'll try and with "staging server" - Good thinking?

tohojo, 2016-11-22:
Try what? What are you trying to do, exactly?

On 22 November 2016 13:10:56 CET, build000 <notifications@github.com> wrote:
>hehe - I'll try and with "staging server" - Good thinking?

build000, 2016-11-22:
luci - that option

tohojo, 2016-11-22:
Yeah, you need to disable that option, and probably clear out the old state and start over. No Luci option for that, unfortunately, but if you log on to the router with SSH and remove everything in /etc/acme that should do the trick...

build000, 2016-11-22:
@tohojo 
There are not enough - I removed the / etc / acme, rebooted the router, after a few hours I repeated to create a new certificate, which happens automatically when koleinym "/etc/init.d/acme start" - still the same.
Disabling this option luci will only give rise to a self-signed certificate ... it seems that the "staging server" is essential ....
EDIT:
Also I got the same idea .... I wrote before this issue - it's just something kicked in general - now a question of whether to steak Lede/acme, or on the side of letsencrypt and the entire process ....

tohojo, 2016-11-22:
A debug log when you run it without the staging server enabled might be helpful...

build000, 2016-11-22:
@tohojo 
It was at this strange thing is, that restored the backup configuration and certification has enabled "staging_server" -Also there must be a problem somewhere else - except, of course, not crossed any limits described on pages letsencrypt.org ... Reg thing is for me not understood ....
Common sense it looks like the servers LE were somehow blocked on the creation of new certificates ... I do not know.
I have several IP and some routers and some domains.
I made attempts to create new certificates with the new IP addresses / e-mail / domains on different devices - each time the same failed result - as in the first post.

tohojo, 2016-11-22:
The log you posted shows a certificate being issued correctly, but from the staging server. If it fails when disabling that, something should show up in the log.

What's the contents of /etc/config/acme?

build000, 2016-11-23:
```
config acme
	option state_dir '/etc/acme'
	option account_email '******@gmail.com'
	option debug '1'

config cert '******'
	option enabled '1'
	option keylength '2048'
	option update_uhttpd '1'
	list domains '****.tk'
	option use_staging '1'
```

tohojo, 2016-11-23:
build000 <notifications@github.com> writes:

> config acme
> 	option state_dir '/etc/acme'
> 	option account_email '******@gmail.com'
> 	option debug '1'
>
> config cert '******'
> 	option enabled '1'
> 	option keylength '2048'
> 	option update_uhttpd '1'
> 	list domains '****.tk'
> 	option use_staging '1'

Hmm, that is still set to use the staging server. Did you uncheck
'staging' in luci and save changes?

Could you try manually setting use_staging to '0', remove the state and
re-run acme?

-Toke

build000, 2016-11-23:
@tohojo 
I prefer not to move, because at the moment this is the only operating certificate on my device (though only works on the device - the restoration described earlier backup on another router also does not work) - I do not want to risk that nothing will work ...

tohojo, 2016-11-24:
You don't need to discard the old acme state; you can just move it out of the way while testing.

Basically, edit the config file to set use_staging to 0, then do:

```
mv /etc/acme /etc/acme.bak
/etc/init.d/acme start
```

And then post the log file somewhere. The script will create a new /etc/acme directory with the new state in it. If it fails, you can just remove that and move the backup directory you created in the step above back in place.

build000, 2016-11-25:
@tohojo 
I wrote you before, because in an earlier post that disabling the "use_staging"
(Use_staging = 0) produces only incorrect, self-signed certificates - it seems that the staging server is necessary ... so that apparently properly developed and tested certificates (logs for other domains/ip/e-mail/etc. Are the key points of the same - correct) are not valid for any browser - here he really is the problem - because of this I am writing here, because maybe in the script or somewhere in Lede there been any significant changes to the process of creating/renew certificates make invalid the the basic point of view of their functionality.

tohojo, 2016-11-25:
Yes, well, if you want me to help figure out why that is, you're going to have to post a debug log (like you did in the initial post) of what happens when you run acme with use_staging disabled. Others (myself included) have had no problem issuing valid certificates...

build000, 2016-11-25:
default /etc/config/uhttpd and new /etc/config/acme (new e-mail/new domain) and pre-remove /etc/acme...and new run ```/etc/init.d/acme start```
```
Fri Nov 25 11:16:15 2016 daemon.info run-acme[11346]: Running pre checks.
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: uci: Entry not found
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  Lets find script dir.
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  _SCRIPT_='/usr/lib/acme/acme.sh'
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  _script='/usr/lib/acme/acme.sh'
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  _script_home='/usr/lib/acme'
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: touch: /etc/acme/account.conf: No such file or directory
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: grep: /etc/acme/account.conf: No such file or directory
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: grep: /etc/acme/account.conf: No such file or directory
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: /usr/lib/acme/acme.sh: line 4466: can't create /etc/acme/account.conf: nonexistent directory
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]: grep: /etc/acme/account.conf: No such file or directory
Fri Nov 25 11:16:15 2016 daemon.info run-acme[11346]: https://github.com/Neilpang/acme.sh
Fri Nov 25 11:16:15 2016 daemon.info run-acme[11346]: v2.6.3
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  Using api: 
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  DOMAIN_PATH='/etc/acme/mainrouter.tk'
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  No -i support in sed
Fri Nov 25 11:16:15 2016 daemon.err run-acme[11346]:  _on_before_issue
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  Le_LocalAddress
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  Check for domain='mainrouter.tk'
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  _currentRoot='no'
Fri Nov 25 11:16:16 2016 daemon.info run-acme[11346]:  Standalone mode.
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  _checkport='80'
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  _checkaddr
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  Using: netstat
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]: [Fri Nov 25 11:16:16 CET 2016] config file is empty, can not read CA_KEY_HASH
Fri Nov 25 11:16:16 2016 daemon.info run-acme[11346]:  Creating account key
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  length='4096'
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  Use length 4096
Fri Nov 25 11:16:16 2016 daemon.err run-acme[11346]:  Using RSA: 4096
Fri Nov 25 11:16:32 2016 daemon.err run-acme[11346]:  RSA key
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  AGREEMENT
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  url='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  payload='{"resource": "new-reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": ""}'
Fri Nov 25 11:16:36 2016 daemon.info run-acme[11346]:  Registering account
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  GET
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  url='https://acme-v01.api.letsencrypt.org/directory'
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  timeout
Fri Nov 25 11:16:36 2016 daemon.err run-acme[11346]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header '
Fri Nov 25 11:16:37 2016 daemon.err run-acme[11346]:  ret='0'
Fri Nov 25 11:16:37 2016 daemon.err run-acme[11346]:  POST
Fri Nov 25 11:16:37 2016 daemon.err run-acme[11346]:  url='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Fri Nov 25 11:16:37 2016 daemon.err run-acme[11346]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header '
Fri Nov 25 11:16:38 2016 daemon.err run-acme[11346]:  _ret='0'
Fri Nov 25 11:16:38 2016 daemon.err run-acme[11346]:  code='201'
Fri Nov 25 11:16:38 2016 daemon.info run-acme[11346]:  Registered
Fri Nov 25 11:16:38 2016 daemon.err run-acme[11346]:  _accUri='https://acme-v01.api.letsencrypt.org/acme/reg/6558676'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  _tos='https://letsencrypt.org/documents/LE-SA-v1.1.1-August-1-2016.pdf'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  AGREEMENT='https://letsencrypt.org/documents/LE-SA-v1.1.1-August-1-2016.pdf'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  Update tos: https://letsencrypt.org/documents/LE-SA-v1.1.1-August-1-2016.pdf
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  url='https://acme-v01.api.letsencrypt.org/acme/reg/6558676'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  payload='{"resource": "reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": "https://letsencrypt.org/documents/LE-SA-v1.1.1-August-1-2016.pdf"}'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  POST
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  url='https://acme-v01.api.letsencrypt.org/acme/reg/6558676'
Fri Nov 25 11:16:39 2016 daemon.err run-acme[11346]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header '
Fri Nov 25 11:16:40 2016 daemon.err run-acme[11346]:  _ret='0'
Fri Nov 25 11:16:40 2016 daemon.err run-acme[11346]:  code='202'
Fri Nov 25 11:16:40 2016 daemon.info run-acme[11346]:  Update success.
Fri Nov 25 11:16:41 2016 daemon.err run-acme[11346]:  Calc CA_KEY_HASH='KKjdZ2DGEzis3+wfLCZdEn29WjsPKlfGETLsgSNXK0I='
Fri Nov 25 11:16:41 2016 daemon.err run-acme[11346]:  Read key length:4096
[... cut ...]
-rw-r--r--    1 root     root           175 Nov 25 11:17 mainrouter.tk.csr.conf
-rw-r--r--    1 root     root          3243 Nov 25 11:17 mainrouter.tk.key
```
/etc/config/acme
```
root@WNDR4700:~# cat /etc/config/acme
config acme
	option state_dir '/etc/acme'
	option account_email 'tajemnyet@gmail.com'
	option debug '1'

config cert 'mainrouter'
	option use_staging '0'
	option enabled '1'
	option keylength '4096'
	option update_uhttpd '1'
	list domains 'mainrouter.tk'
```

/etc/config/uhttp not updated becouse valdation not properly and not create fullchain.cer

tohojo, 2016-11-25:
Right, so this indicates that the verification server cannot connect to your router. And that the acme debug script can't either.

Things to check:

1. Does the mainrouter.tk domain point to the public IP of your router?
2. Is there another firewall between your router and the public internet that may be blocking port 80?
3. Does the domain resolve on the router itself (i.e. can you `ping mainrouter.tk` when logged on to the router)?

build000, 2016-11-25:
Pleas run https://mainrouter.tk - this is...public adddress - no any other firewall - only in router

tohojo, 2016-11-25:
Hmm. Well it appears that the letsencrypt verification agent does actually connect; but then reports afterwards that it couldn't. That is odd...

What's the contents of /etc/acme/http.header?

build000, 2016-11-25:
```
root@WNDR4700:~# cat /etc/acme/http.header
HTTP/1.1 200 OK
Server: nginx
Content-Type: application/pkix-cert
Content-Length: 1199
Boulder-Request-Id: rvGELyDL4SYlWFsz2RqvPReYEKNWa7dhnvLpuQRvNsc
Replay-Nonce: LRLUtQvGWk77dmxozDXgvEWRd-T8jypRBTP0T1BhbIE
X-Frame-Options: DENY
Strict-Transport-Security: max-age=604800
Expires: Fri, 25 Nov 2016 12:17:55 GMT
Cache-Control: max-age=0, no-cache, no-store
Pragma: no-cache
Date: Fri, 25 Nov 2016 12:17:55 GMT
Connection: keep-alive
```

tohojo, 2016-11-25:
Right, uploaded a new package that handles the transition away from staging automatically, and also behaves better if things fail. Also updated the upstream acme.sh version. Not sure if it'll solve you problem, but at least it should give some more information as to what goes wrong.

Please try v 1.3: https://kau.toke.dk/lede/packages/acme_1.3-1_all.ipk (will also be in the upstream builds once the buildbot gets around to it).

build000, 2016-11-25:
install acme-1.3...and not visible any other changes - still bad certificate to any browser (option staging server or not staging).

tohojo, 2016-11-25:
Did you run /etc/init.d/acme start after installing? Could you post the debug output for that run, please?

build000, 2016-11-25:
remove /etc/acme
standard /etc/config/uhttpd (when use only https):
```
config uhttpd 'main'
	list listen_https '0.0.0.0:443'
	option redirect_https '1'
	option home '/www'
	option rfc1918_filter '0'
	option max_requests '3'
	option max_connections '100'
	option cgi_prefix '/cgi-bin'
	option script_timeout '60'
	option network_timeout '30'
	option http_keepalive '20'
	option tcp_keepalive '1'
	option ubus_prefix '/ubus'
	#option listen_http '0.0.0.0:80'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'

config cert 'defaults'
	option days '730'
	option bits '4096'
	option country 'PL'
	option state 'Masovian'
	option location 'Warsaw'
	option commonname 'LEDE'
```
...e-mail is the same, domain is the same (generally config the same and witchout staging server):

```
Fri Nov 25 16:16:03 2016 daemon.info run-acme[2665]: Running pre checks.
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]: uci: Entry not found
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]: v4 input_rule: Chain input_rule (1 references)
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  pkts bytes target     prot opt in     out     source               destination         
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:     0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]: v6 input_rule: Chain input_rule (1 references)
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  pkts bytes target     prot opt in     out     source               destination         
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:     0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 tcp dpt:80
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]: port80 listens: 
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Lets find script dir.
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _SCRIPT_='/usr/lib/acme/acme.sh'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _script='/usr/lib/acme/acme.sh'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _script_home='/usr/lib/acme'
Fri Nov 25 16:16:03 2016 daemon.info run-acme[2665]: https://github.com/Neilpang/acme.sh
Fri Nov 25 16:16:03 2016 daemon.info run-acme[2665]: v2.6.3
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Using api: 
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  DOMAIN_PATH='/etc/acme/mainrouter.tk'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  No -i support in sed
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _on_before_issue
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Le_LocalAddress
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Check for domain='mainrouter.tk'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _currentRoot='no'
Fri Nov 25 16:16:03 2016 daemon.info run-acme[2665]:  Standalone mode.
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _checkport='80'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  _checkaddr
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Using: netstat
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]: [Fri Nov 25 16:16:03 CET 2016] config file is empty, can not read CA_KEY_HASH
Fri Nov 25 16:16:03 2016 daemon.info run-acme[2665]:  Creating account key
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  length='4096'
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Use length 4096
Fri Nov 25 16:16:03 2016 daemon.err run-acme[2665]:  Using RSA: 4096
Fri Nov 25 16:16:37 2016 daemon.err run-acme[2665]:  RSA key
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  AGREEMENT
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  url='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  payload='{"resource": "new-reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": ""}'
Fri Nov 25 16:16:41 2016 daemon.info run-acme[2665]:  Registering account
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  GET
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  url='https://acme-v01.api.letsencrypt.org/directory'
Fri Nov 25 16:16:41 2016 daemon.err run-acme[2665]:  timeout
[... 74 lines cut ...]
Fri Nov 25 16:17:14 2016 daemon.info run-acme[2665]: 
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  checking
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  GET
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  url='https://acme-v01.api.letsencrypt.org/acme/challenge/QyEfZ51UY7wrXjaeeaRykwn4bmH6fTZH9gW3_E4XrFM/364154258'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  timeout
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header '
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  ret='0'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]: [Fri Nov 25 16:17:16 CET 2016] mainrouter.tk:Verify error:Could not connect to mainrouter.tk
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  Debug: get token url.
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  GET
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  url='http://mainrouter.tk/.well-known/acme-challenge/jVXGsaWw62rYpGQ-kX43s2Inej-AoV-YaToQI5fJLUg'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  timeout='1'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header  --connect-timeout 1'
[... cut ...]
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]: [Fri Nov 25 16:17:16 CET 2016] Please use add '--debug' or '--log' to check more details.
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]: [Fri Nov 25 16:17:16 CET 2016] See: https://github.com/Neilpang/acme.sh/wiki/How-to-debug-acme.sh
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]: Issuing cert for mainrouter.tk failed. Moving state to /etc/acme/mainrouter.tk.failed-1480087036
Fri Nov 25 16:17:16 2016 daemon.info run-acme[2665]: Running post checks (cleanup).
```

hnyman, 2016-11-25:
I have not used acme so this is pure guess, but it looks to me that you have explicitly disabled http = port 80 for uhttpd www server, and the script needs to connect to that for verification and fails that. You might test enabling http / port 80 while you generate the certificate.

```
config uhttpd 'main'
	list listen_https '0.0.0.0:443'
...
	#option listen_http '0.0.0.0:80'
```
```
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  url='http://mainrouter.tk/.well-known/acme-challenge/jVXGsaWw62rYpGQ-kX43s2Inej-AoV-YaToQI5fJLUg'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  timeout='1'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header  --connect-timeout 1'
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]: [Fri Nov 25 16:17:16 CET 2016] Please refer to https://curl.haxx.se/libcurl/c/libcurl-errors.html for error code: 7
Fri Nov 25 16:17:16 2016 daemon.err run-acme[2665]:  ret='7'
```

build000, 2016-11-25:
ok...enable it in /etc/config/uhttpd....

tohojo, 2016-11-25:
No, the acme script will disable uhttpd and listen on port 80 with netcat. Will look at the log later...

build000, 2016-11-25:
@tohojo 
So it does not matter if port 80 is enabled to listen to the /etc/config/uhttpd?
Anyway, whether incorporated or not, and does not work for me ...

tohojo, 2016-11-25:
No, it doesn't. And according to that log, everything is working. Only thing I can think of is that the letsencrypt server makes more than one request where the second one fails. 

Will see if I can replicate the error and figure out what's going on...

build000, 2016-11-26:
@tohojo 
According to the original instruction to the git project acme I remade minimum file /usr/lib/acme/run-acme on the more detailed debug mode:
```
(...)
config_get keylength "$section" keylength

    [ "$enabled" -eq "1" ] || return

    [ "$DEBUG" -eq "1" ] && acme_args="$acme_args --debug 2"

    set -- $domains
    main_domain=$1
(...)
```
...an new run ```/etc/init.d/acme start```. In fact, a number of additional lines of the entire process of creating and verifying the certificate, here's the log using the ```--debug 2``` :
```
Sat Nov 26 02:19:45 2016 daemon.info run-acme[24193]: Running pre checks.
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]: v4 input_rule: Chain input_rule (1 references)
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  pkts bytes target     prot opt in     out     source               destination         
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:     0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]: v6 input_rule: Chain input_rule (1 references)
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  pkts bytes target     prot opt in     out     source               destination         
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:     0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 tcp dpt:80
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]: port80 listens: 
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _is_idn_d='mainrouter.tk'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _idn_temp
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  Lets find script dir.
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _SCRIPT_='/usr/lib/acme/acme.sh'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _script='/usr/lib/acme/acme.sh'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _script_home='/usr/lib/acme'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  1:ACCOUNT_EMAIL='tajemnyet@gmail.com'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  Using api: 
Sat Nov 26 02:19:45 2016 daemon.info run-acme[24193]: https://github.com/Neilpang/acme.sh
Sat Nov 26 02:19:45 2016 daemon.info run-acme[24193]: v2.6.3
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  DOMAIN_PATH='/etc/acme/mainrouter.tk'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  1:Le_Domain='mainrouter.tk'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  2:Le_Alt='no'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  3:Le_Webroot='no'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  4:Le_PreHook=''
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  5:Le_PostHook=''
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  6:Le_RenewHook=''
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  options='s/^Le_LocalAddress.*$//'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  No -i support in sed
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  7:Le_API='https://acme-v01.api.letsencrypt.org'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _on_before_issue
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  'no' contains 'no'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  Le_LocalAddress
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  Check for domain='mainrouter.tk'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _currentRoot='no'
Sat Nov 26 02:19:45 2016 daemon.info run-acme[24193]:  Standalone mode.
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _checkport='80'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _checkaddr
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  Using: netstat
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  'no' does not contain 'apache'
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]: [Sat Nov 26 02:19:45 CET 2016] config file is empty, can not read CA_KEY_HASH
Sat Nov 26 02:19:45 2016 daemon.err run-acme[24193]:  _saved_account_key_hash
Sat Nov 26 02:19:46 2016 daemon.info run-acme[24193]:  Creating account key
Sat Nov 26 02:19:46 2016 daemon.err run-acme[24193]:  length='4096'
Sat Nov 26 02:19:46 2016 daemon.err run-acme[24193]:  Use length 4096
Sat Nov 26 02:19:46 2016 daemon.err run-acme[24193]:  Using RSA: 4096
Sat Nov 26 02:19:52 2016 daemon.err run-acme[24193]:  RSA key
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  AGREEMENT
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  url='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  payload='{"resource": "new-reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": ""}'
Sat Nov 26 02:19:57 2016 daemon.info run-acme[24193]:  Registering account
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  Use cached jwk for file: /etc/acme/ca/acme-v01.api.letsencrypt.org/account.key
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  Get nonce.
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  GET
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  url='https://acme-v01.api.letsencrypt.org/directory'
Sat Nov 26 02:19:57 2016 daemon.err run-acme[24193]:  timeout
[... 300 lines cut ...]
Sat Nov 26 02:20:17 2016 daemon.err run-acme[24193]:  response='{"type":"http-01","status":"invalid","error":{"type":"urn:acme:error:connection","detail":"Could not connect to mainrouter.tk","status": 400},"uri":"https://acme-v01.api.letsencrypt.org/acme/challenge/1WR352us9T_uNZ9BveAYrxw1hM-GfWZP4EvSL3B9gQA [... cut ...]
Sat Nov 26 02:20:17 2016 daemon.err run-acme[24193]: [Sat Nov 26 02:20:17 CET 2016] Please use add '--debug' or '--log' to check more details.
Sat Nov 26 02:20:17 2016 daemon.err run-acme[24193]: [Sat Nov 26 02:20:17 CET 2016] See: https://github.com/Neilpang/acme.sh/wiki/How-to-debug-acme.sh
Sat Nov 26 02:20:17 2016 daemon.err run-acme[24193]: Issuing cert for mainrouter.tk failed. Moving state to /etc/acme/mainrouter.tk.failed-1480123217
Sat Nov 26 02:20:17 2016 daemon.info run-acme[24193]: Running post checks (cleanup).
```
Can log in this form will be more helpful in understanding the problem...

thess, 2016-11-26:
I agree with @hnyman - Your server is not accepting connections directly on port 80. From what I see, it is redirecting (via "307 - Temporary Redirect") to port 443, LE will not succeed with this type of redirect.

>Connecting` to mainrouter.tk|89.70.80.230|:80... connected.
HTTP request sent, awaiting response...
  HTTP/1.0 307 Temporary Redirect
  Connection: close
  Location: https://mainrouter.tk/
Location: https://mainrouter.tk/ [following]
--2016-11-26 09:25:31--  https://mainrouter.tk/
Connecting to mainrouter.tk|89.70.80.230|:443... connected.

tohojo, 2016-11-26:
@thess - nope, the acme script will reconfigure uhttpd to not listen on port 80 for the duration of the certificate issuance, and listen using netcat instead, so you can't reason from what behaviour you observe when you connect. Besides, the letsencrypt client usually has no problem following redirects.

Thing is, these lines:
```
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: GET /.well-known/acme-challenge/GAnNtUpMRwKXqfHigXN7UzdioJcjPV41OIDxHnnWU_s HTTP/1.1
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: Host: mainrouter.tk
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: User-Agent: Mozilla/5.0 (compatible; Let's Encrypt validation server; +https://www.letsencrypt.org)
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: Accept: */*
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: Accept-Encoding: gzip
Sat Nov 26 02:20:15 2016 daemon.info run-acme[24193]: Connection: close
```

indicate that the letsencrypt client **does** connect successfully.

@build000 could you try running this:
```
iptables -I INPUT -m state --state -new -p tcp --dport 80 -j LOG
```

and then re-run acme and post the log? That should show any incoming connections during the test, which can tell us if it's because there are multiple of them.

build000, 2016-11-27:
Due to the fact that your statements were uncertain about the port 80 I thought I will give much more additional information ... because, as is often the case, "the devil may lie in the details".
```
root@WNDR4700:~# uname -a
Linux WNDR4700 4.4.32 #0 Sat Nov 26 21:39:27 2016 ppc GNU/Linux
root@WNDR4700:~# cat /etc/openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='HEAD'
DISTRIB_REVISION='r2331'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='apm821xx/nand'
DISTRIB_DESCRIPTION='LEDE Reboot r2331'
DISTRIB_TAINTS='no-all busybox'
```
1) .config compilation: http://wklej.org/id/2971998/
2). ```cat /etc/config/network```: http://wklej.org/id/2971999/
3). ```cat /etc/config/dhcp```: http://wklej.org/id/2972000/ (using dnscrypt-proxy:cisco)
4). ```cat /etc/config/firewall```: http://wklej.org/id/2972002/
5). using adblock (all rules - only to wan interface)
6). ```cat /etc/crontabs/root```: http://wklej.org/id/2972004/
If you need more please post - I might add as much as possible.
Due to the fact that I could not get any rfektu the new domain returned from a backup to a working certificate (and renewing automatically - in spite of the use of the "staging server"). So, during the test acme I bandaged appropriate shelves/folders "* .bak".

The current configuration acme: http://wklej.org/id/2972005/
The current configuration uhttpd: http://wklej.org/id/2972007/
Using proposes command:
```
```
iptables -I INPUT -m state --state NEW -p tcp --dport 80 -j LOG
```
...and new run /etc/config/acme start``` (--debug 2):

tohojo, 2016-11-27:
The path doesn't exist anywhere on the router. The acme script will start netcat listening on port 80, which will return a successful HTTP response when letsencrypt connects. As far as I can tell, this works correctly; see the parts of the log I posted above.

Could you do a run with the staging server **enabled**, but a clean /etc/acme dir and post the debug log for comparison?

Will open a bug entry at the upstream acme.sh as well and see if anyone there has an idea.

build000, 2016-11-28:
Ok - ```use_staging '1'```(rest is the same), LEDE upgrade to r2341, the same configuration system from previously main post( https://github.com/openwrt/packages/issues/3544#issuecomment-263102574 )...and new run ```/etc/init.d/acme start```:
```
Mon Nov 28 06:58:13 2016 daemon.info run-acme[3388]: Running pre checks.
Mon Nov 28 06:58:13 2016 daemon.info run-acme[3388]: Running post checks (cleanup).
Mon Nov 28 07:00:00 2016 cron.info crond[1273]: USER root pid 3663 cmd ping -c 5 192.168.2.1
Mon Nov 28 07:01:07 2016 daemon.info run-acme[3811]: Running pre checks.
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]: v4 input_rule: Chain input_rule (1 references)
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]:  pkts bytes target     prot opt in     out     source               destination         
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]:     0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]: v6 input_rule: Chain input_rule (1 references)
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]:  pkts bytes target     prot opt in     out     source               destination         
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]:     0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 tcp dpt:80
Mon Nov 28 07:01:07 2016 daemon.err run-acme[3811]: port80 listens: 
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _is_idn_d='mainrouter.tk'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _idn_temp
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Lets find script dir.
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _SCRIPT_='/usr/lib/acme/acme.sh'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _script='/usr/lib/acme/acme.sh'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _script_home='/usr/lib/acme'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  1:ACCOUNT_EMAIL='tajemnyet@gmail.com'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Using api: 
Mon Nov 28 07:01:08 2016 daemon.info run-acme[3811]: https://github.com/Neilpang/acme.sh
Mon Nov 28 07:01:08 2016 daemon.info run-acme[3811]: v2.6.3
Mon Nov 28 07:01:08 2016 daemon.info run-acme[3811]:  Using stage api:https://acme-staging.api.letsencrypt.org
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  DOMAIN_PATH='/etc/acme/mainrouter.tk'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  1:Le_Domain='mainrouter.tk'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  2:Le_Alt='no'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  3:Le_Webroot='no'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  4:Le_PreHook=''
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  5:Le_PostHook=''
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  6:Le_RenewHook=''
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  options='s/^Le_LocalAddress.*$//'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  No -i support in sed
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  7:Le_API='https://acme-staging.api.letsencrypt.org'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _on_before_issue
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  'no' contains 'no'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Le_LocalAddress
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Check for domain='mainrouter.tk'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _currentRoot='no'
Mon Nov 28 07:01:08 2016 daemon.info run-acme[3811]:  Standalone mode.
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _checkport='80'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _checkaddr
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Using: netstat
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  'no' does not contain 'apache'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]: [Mon Nov 28 07:01:08 CET 2016] config file is empty, can not read CA_KEY_HASH
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  _saved_account_key_hash
Mon Nov 28 07:01:08 2016 daemon.info run-acme[3811]:  Creating account key
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  length='4096'
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Use length 4096
Mon Nov 28 07:01:08 2016 daemon.err run-acme[3811]:  Using RSA: 4096
Mon Nov 28 07:01:22 2016 daemon.err run-acme[3811]:  RSA key
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  AGREEMENT
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  url='https://acme-staging.api.letsencrypt.org/acme/new-reg'
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  payload='{"resource": "new-reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": ""}'
Mon Nov 28 07:01:27 2016 daemon.info run-acme[3811]:  Registering account
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  Use cached jwk for file: /etc/acme/ca/acme-staging.api.letsencrypt.org/account.key
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  Get nonce.
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  GET
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  url='https://acme-staging.api.letsencrypt.org/directory'
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  timeout
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header  --trace-ascii /tmp/tmp.aNAfCK '
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  ret='0'
Mon Nov 28 07:01:27 2016 daemon.err run-acme[3811]:  _headers='HTTP/1.1 200 OK
[... cut ...]
Mon Nov 28 07:01:50 2016 daemon.info run-acme[3811]: Running post checks (cleanup).
```
Effect to FF is identical for 1. post (bad certificate):

> mainrouter.tk uses an invalid security certificate. The certificate is not trusted because the issuer certificate is unknown. The server might not be sending the appropriate intermediate certificates. An additional root certificate may need to be imported. Error code: SEC_ERROR_UNKNOWN_ISSUER

windows 10 edge, and other system/browser ... similar = (CN) Fake LE Intermediate X1 and "no green padlock"...

tohojo, 2016-11-28:
Huh, this is definitely weird. Almost seems like the reply doesn't make it to the live server, but does to the staging.

Your network config has 'mtu 9000' on the wan network. What's your uplink?

Could you try issuing these commands (instead of the previous iptables line; you can clear that by reloading the firewall), and then run acme with staging disabled and post another log?

```
iptables -I INPUT -p tcp --dport 80 -j LOG
iptables -I OUTPUT -p tcp --sport 80 -j LOG
```

Neilpang, 2016-11-28:
@tohojo @build000 

Hi,
I wonder if it's possible to stop the httpd server and use raw acme.sh to issue cert on commandline?

To make thing simple:

1.  stop your httpd server by hand.
2. install acme.sh.    `curl https://get.acme.sh | sh`
3. Use raw acme.sh to issue cert.

```
./$HOME/.acme.sh/acme.sh  --issue -d mainrouter.tk  --standalone  --debug 2
```

build000, 2016-11-28:
@tohojo 
...run again (use_staging '0'; rm -fr /etc/acme; restart network, firewall and uhttpd - change cert/key to default LEDE):
```
Mon Nov 28 16:11:34 2016 daemon.info run-acme[11583]: Running pre checks.
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]: v4 input_rule: Chain input_rule (1 references)
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  pkts bytes target     prot opt in     out     source               destination         
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:     0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]: v6 input_rule: Chain input_rule (1 references)
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  pkts bytes target     prot opt in     out     source               destination         
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:     0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 tcp dpt:80
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]: port80 listens: 
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  _is_idn_d='mainrouter.tk'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  _idn_temp
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  Lets find script dir.
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  _SCRIPT_='/usr/lib/acme/acme.sh'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  _script='/usr/lib/acme/acme.sh'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  _script_home='/usr/lib/acme'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  1:ACCOUNT_EMAIL='tajemnyet@gmail.com'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  Using api: 
Mon Nov 28 16:11:34 2016 daemon.info run-acme[11583]: https://github.com/Neilpang/acme.sh
Mon Nov 28 16:11:34 2016 daemon.info run-acme[11583]: v2.6.3
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  DOMAIN_PATH='/etc/acme/mainrouter.tk'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  1:Le_Domain='mainrouter.tk'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  2:Le_Alt='no'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  3:Le_Webroot='no'
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  4:Le_PreHook=''
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  5:Le_PostHook=''
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  6:Le_RenewHook=''
Mon Nov 28 16:11:34 2016 daemon.err run-acme[11583]:  options='s/^Le_LocalAddress.*$//'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  No -i support in sed
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  7:Le_API='https://acme-v01.api.letsencrypt.org'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  _on_before_issue
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  'no' contains 'no'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  Le_LocalAddress
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  Check for domain='mainrouter.tk'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  _currentRoot='no'
Mon Nov 28 16:11:35 2016 daemon.info run-acme[11583]:  Standalone mode.
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  _checkport='80'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  _checkaddr
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  Using: netstat
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  'no' does not contain 'apache'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]: [Mon Nov 28 16:11:35 CET 2016] config file is empty, can not read CA_KEY_HASH
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  _saved_account_key_hash
Mon Nov 28 16:11:35 2016 daemon.info run-acme[11583]:  Creating account key
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  length='4096'
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  Use length 4096
Mon Nov 28 16:11:35 2016 daemon.err run-acme[11583]:  Using RSA: 4096
Mon Nov 28 16:12:16 2016 daemon.err run-acme[11583]:  RSA key
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  AGREEMENT
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  url='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  payload='{"resource": "new-reg", "contact": ["mailto: tajemnyet@gmail.com"], "agreement": ""}'
Mon Nov 28 16:12:21 2016 daemon.info run-acme[11583]:  Registering account
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  Use cached jwk for file: /etc/acme/ca/acme-v01.api.letsencrypt.org/account.key
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  Get nonce.
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  GET
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  url='https://acme-v01.api.letsencrypt.org/directory'
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  timeout
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  _CURL='curl -L --silent --dump-header /etc/acme/http.header  --trace-ascii /tmp/tmp.HdHHoh '
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  ret='0'
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]:  _headers='HTTP/1.1 200 OK
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]: Server: nginx
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]: Content-Type: application/json
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]: Content-Length: 280
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]: Boulder-Request-Id: ETsQYKH888-ZZ92HvJcXDUKFvB743rw9x_rMD9WI5cQ
Mon Nov 28 16:12:21 2016 daemon.err run-acme[11583]: Replay-Nonce: _THCTrvMkoc-vXRhGlG2sNuDgoqz_k89t9_IhE1CjzE
[... cut ...]
Mon Nov 28 16:13:33 2016 daemon.info run-acme[11583]:  Your cert key is in  /etc/acme/mainrouter.tk/mainrouter.tk.key 
Mon Nov 28 16:13:34 2016 daemon.info run-acme[11583]:  The intermediate CA cert is in  /etc/acme/mainrouter.tk/ca.cer 
Mon Nov 28 16:13:34 2016 daemon.info run-acme[11583]:  And the full chain certs is there:  /etc/acme/mainrouter.tk/fullchain.cer 
Mon Nov 28 16:13:34 2016 daemon.info run-acme[11583]: Running post checks (cleanup).
```
Please check that you show up in your browser ...

tohojo, 2016-11-28:
Yup, I get a valid cert when I connect as well. Guess it was a transient network problem or something, then (unless you changed something else in the config?)

build000, 2016-11-28:
no, i not changed any rules, configuration, etc....only use your new iptables rules (to log) and usin this time "use_staging '0'"

tohojo, 2016-11-28:
Right, then I am going to put it down to goblins in the network and close this issue... :)

closed 2016-11-28 by tohojo

build000, 2016-11-28:
"Transient network problem ..." - lasting more than a week's time ... ???

tohojo, 2016-11-28:
*shrugs* - well, it's not unheard of... don't see anything else that has changed...

You're welcome to open a new issue if you have problems again, of course, but I really don't see any bugs to fix here...