openwrt/packages #3557: dump1090 - Problem fetching data

Strykar, 2016-11-23:
luci-dump1090 does not start the daemon with default or any options. Starting dump1090 manually works and it logs tracked craft to the console. I'm not sure if this is due to my custom build. The webui for dump1090 at port 8080 does not work and prints this:
```
Problem fetching data from dump1090.
AJAX call failed (parsererror: SyntaxError: Unexpected end of JSON input). Maybe dump1090 is no longer running?
The displayed map data will be out of date.

# ps -e | grep dump1090
14476 ttyS0    00:00:01 dump1090
```
`/usr/share/dump1090/data` points to `/var/run/dump1090` but `/var/run/dump1090` does not exist. Manually creating the directory and the file `aircraft.json` inside it fixes the issue.
See https://github.com/mutability/dump1090/issues/165

Noltari, 2016-11-24:
@Strykar please define what you mean by **custom build**

Strykar, 2016-11-24:
@Noltari Some GPIO, LED patches to LEDE nightlies from - https://github.com/riptidewave93/LEDE-APU2/tree/lede-upstream-port
It's currently in RFC 2 but it's slated to be merged and perhaps more importantly, they're not my patches :)
```
root@apu:/# uname -a
Linux apu2 4.4.27 #0 SMP Thu Oct 27 13:29:36 2016 x86_64 GNU/Linux
```
I can reproduce the issue by rebooting and `/var/run/dump1090` is gone.
Current fix - https://gist.github.com/Strykar/3dc71f5ae6e28330e2b706fcbf4d4ed3

Noltari, 2016-11-25:
@Strykar thanks for the info!
I will try to reproduce this as soon as possible.

Noltari, 2016-12-27:
Closing this since I couldn't reproduce it myself and it's working oob without any changes.

closed 2016-12-27 by Noltari