openwrt/packages #3579: Discuss contribution guidelines

diizzyy, 2016-11-27:
Hi, 

Where are you supposed to discuss guidelines?

@hnyman @thess @wigyori @commodo 

Best regards,
Daniel

hnyman, 2016-11-27:
> Where are you supposed to discuss guidelines?

Probably here, as the dev mailing lists are separate for Openwrt & LEDE.

diizzyy, 2016-11-27:
A few things I've noticed over some time...

Can we enforce SHA256 checksums on new pull commits? Pretty much all distros have dropped MD5 in favour of SHA1 and/or SHA256. Preferable we should have a correct variable (alias) for this instead of the rather ugly current hack.

Enforce URLs too? HTTPS --> HTTP --> FTP --> HTTPS (GIT/SVN) --> HTTP (GIT/SVN) --> native SCM protocol (SCM pulls should always be in xz format)

Have some kind of standard format of package version for SCM snapshots?

Prefer (and enforce) tarballs archives in the following order of preference: xz --> bz2 --> gz

Enforce some kind of maintainer and submitter timeout (including closing old pull reqs that needs work)? FreeBSD's is pretty sane IMHO
https://www.freebsd.org/doc/en/books/porters-handbook/makefile-maintainer.html

Makefile copyright standpoint should be clearly explained, (ie do not blindly copyright openwrt.org)?

commodo, 2016-11-27:
I don't have any strong preference on this matter, since I don't have a lot of experience [in the sense of past experience where to point for arguments & counter-arguments] with stuff.

As long as a consensus is reached, and I am aware of the guidelines, I will try to adhere.

hnyman, 2016-11-27:
I don't see any urgent necessity to enforce e.g. SHA256, .xz files or https preference. They might be added as "Recommended practices" section (or "Best practices" or something like that) to the contributing.md document. Same goes also for the copyright advice. It could be in contributing.md, so that it could be easily referred to.

There could be a timeout if no response from author for change requests for some time. I did some cleanup a while ago both for issues and old PRs, but there is no good official guideline.

Personally I see two issues as the urgent ones:
* disappearing maintainers. We have several old maintainers who have several PRs open for their packages, but no reaction for weeks/months. That includes also current "core developers" from both Openwrt and LEDE.
* some discussion & guidelines on what kind of new packages to adopt into the repo. Personally I have been rather reluctant to accept new packages, as in many cases they are just the author's need for his own build without much relevancy at least for any generic "router-like device" functionality. This applies e.g. to many Python add-ons, other languages, small utils. All the apps get compiled for all ~50 targets by buildbots, so adding new apps en masse could slow down buildbot compilation.

diizzyy, 2016-11-27:
SHA* is a security precaution which pretty much all other major distributions have adopted, I fail to see the benefit of not enforcing this.

Regarding compression I think you have to realize that far from all are on unrated connections and it's desirable to keep both HDD and bandwidth down on especially mirrors. This was the main reason for LEDE switching to xz on a bunch of packages and ports. It surely does take more processing power to compress but the gain is well worth it.

As far as https/http/ftp/git/svn goes it's much desirable to use https/http/ftp if you're behind lets say a corporate proxy. It doesn't apply to everyone for sure but we should try to be as business-friendly as possible. I know some lead developers have expressed interest in this subject as far as downloads goes.

Regarding disappearing maintainers I think one should adopt some kind of timeout system, otherwise you'll end up having both PRs is issues being around for ages and potential having no idea of the package's current state. Once maintainer timeout has been reached the package should be marked "Unmaintained" so ppl know that they can offer maintainership.

Regarding new packages, I don't think you can draw a line what's acceptable or not. Languages like Python are a pain in that regard since "base" is broken down into so many "random" variants and on top of that you have pypi which serves lots of packages.  Some are needed for larger applications which may not even be network related at all (home automation comes to mind) . If we don't see any obvious dependency I think it's fair to decline adding it. I do realize this may be the holy grail for some but having that in mind you'll also run into issues where you'll duplicate functionality, how do you device what makes the cut and what doesn't? I don't see the need to add languages without any actual application.

thess, 2016-11-28:
With regard to _enforcement_ of what I would consider minor issues such as .xz vs .bz2 (or .gz), SHA256 vs MD5 and HTTPS over HTTP, these topics should be addressed as preferred rather than mandatory. They can be documented in the seldom read (or followed) CONTRIBUTING.md guidelines. In fact, a PR from @diizzyy on that document with some cogent recommendations would be appreciated and reviewed by myself and others. Specifically - my $.02US:

- **Package compression**: Bandwidth and storage differences are not all that significant with respect to cost and availability. 
- **SHA** vs **MD5**: I see no real motivation of one over the other in our usage. Having NONE is far worse - hence the need for static (repeatable) archives.
- **HTTPS** vs **HTTP** vs **others**: no stong opinion. I do prefer reliable services better than private or low-bandwidth archives.

On the topic of MIA/hard to reach maintainers, I agree with Daniel E. and Hannu that we should have some rule-of-thumb timeouts and actions when we get no response to acceptable requests. The current active committers can handle those on a case-by-case basis.

A discussion on the subject of new package submissions is a good idea. I too am reluctant to accept packages that, at least to me, have little or no use to the community at large. Perhaps having a requirement to have some number of +1 ACKs before they are accepted. There seems to be enough special interest groups which do maintain private repos and some which provide package distributions. I don't think it is desirable for us to become a major, general purpose, package maintainer/distributor.

thess, 2016-11-28:
**Note:** Introducing a new build tag like SHA256SUM would require both build systems (LEDE and OpenWrt) be updated together. If you are thinking of doing this, please submit the PR's or [Patch] to both repositories.

karlp, 2016-11-28:
https://forum.lede-project.org/t/packaging-guidelines/65/2 has a bunch of clarifications and rewording on this, but it was still soliciting some more feedback.  It would be good to get it in as an update to the docs in this repo though.

EricLuehrsen, 2016-11-29:
> these topics should be addressed as preferred rather than mandatory.

I would agree, and BTW well put. There are motivators for these changes, but no strong rationale to forbid carry over practices.

The question of absentee maintainer is tough for a volunteer project. The issues need to be addressed appropriately and separately. Some packages are popular. Some packages are required by commercial interests that help fuel OpenWRT/LEDE. Some maintainers just fall away. Some alternate scale needs to be applied to more leisure packages with fits of genius at large intervals.  

On new package submissions, I will just drop this one in here for good measure. 'Packages' is big and growing. It may be time to consider adjusting the collective of add-on repo's to spread out categories of content. (Anything python just goes to openwrt/python). In fairness to my bias, I would consider entirely separating build-tools from the core embedded content in OpenWRT/LEDE.

diizzyy, 2016-11-29:
What's the view on threads and unicode, should this be encouraged? I personally we should have this enabled as much as possible but I know some are going to have grips about the (in general) relatively small size increase.

stintel, 2016-11-30:
As long as we are putting SHA256 checksums in a variable with MD5 in its name, I am completely against enforcing it. It is confusing and just wrong.

luizluca, 2016-11-30:
Can we add two checksum vars?

- PKG_MD5SUM:= (should only be used for MD5, but still accepts SHA256). Both OpenWRT and LEDE reads and uses it
- PKG_SHA256SUM:= (accepts only SHA256). It might be ignored by old SDK but I guess the absence of a checksum does not break the build.

Instead of PKG_SHA256SUM, we could create a new PKG_CHECKSUM which can be used for any supported checksum like:

`PKG_CHECKSUM:=SHA256:123456789abcdef...`

And build could err or warn (and ignore) when the used checksum is unknown.

diizzyy, 2016-11-30:
I'd propose that we create PKG_CHECKSUM (SHA256 only), leave MD5SUM as is and enforce new commits to use PKG_CHECKSUM instead.

diizzyy, 2016-12-01:
@thess 
Trying to avoid the wall of text, I think details should be linked to the wiki page rather than in this document. https://github.com/diizzyy/fresh-stuff/wiki/New-contribution.md
I think git related stuff should be in a separate document.

cshoredaniel, 2016-12-03:
I actually think that it would be good to reduce the number of packages in single feed (e.g. more of the openwrt-routing type repositories) except for packages that are of particular interest to mainstream uses of OpenWrt/LEDE.  The first obvious ones to separate out are the directories like admin and mail that are already really distinct from net/utils/libs.  Even the various languages could likely be split into their own repos since they are not part of the main distribution (except maybe perl and lua which are used by a larger number of packages).

Libs that are of wide general use should obviously be left in libs but special purposes ones only used by a small number of packages should probably moved into the repo for what they support.

Net and Utils are really a bit of a dumping ground and the moment and perhaps there ought to be a repo like openwrt-general-purpose which could have packages that are of enough interest to have people maintaining them and handling github, but aren't really mainstream (like cups, sane, maybe nut (which I intend to maintain)), and keep things like ddns-scripts and collectd and other packages for widely used parts of openwrt that just aren't maintained in the core system because they are not the focus of the core developers.

openwrt-virtualization would be a good place for handling lxc, and the recent qemu/kvm related efforts, adding docker, etc.

Just my 2¢

diizzyy, 2016-12-22:
Any comments? :-)

cshoredaniel, 2017-01-08:
On Mon, 28 Nov 2016 19:33:42 -0800
Eric Luehrsen <notifications@github.com> wrote:

> > these topics should be addressed as preferred rather than
> > mandatory.  
> 
> I would agree, and BTW well put. There are motivators for these
> changes, but no strong rationale to forbid carry over practices.

Agreed.

> The question of absentee maintainer is tough for a volunteer project.
> The issues need to be addressed appropriately and separately. Some
> packages are popular. Some packages are required by [...]

For more 'leisure' packages I'm wondering if as with the comments on
new packages, this should be a separate repo that gets built less often
(i.e. take one of the daily snapshots as a monthly snapshot and build
the 'leisure' packages then, and not necessarily as a condition of
release (i.e. as a secondary later release when a release happens).

That way the core effort goes to generally important packages, but the
community still gets goodies.

I've actually though about moving some of my packages to my own package
repo and setting up builds and releases, for example for my
enhancements to NUT and the 'msmtpq-ng-mta' tiny queuing MTA (depends
on MTA plus some tens of KB to do send-only MTA for packages that send
mail via sendmail or a localhost unauthed SMTP server (not enabled by
default).

I don't think that's a viable solution though for the majority of
package contributors, and I'm wondering if it would be possible to
have a separate 'leisure/experimental/community' package build system
that get's notifications when the core has built a new SDK and builds
on different infrastructure from core project.
 
> On new package submissions, I will just drop this one in here for
> good measure. 'Packages' is big and growing. It may be time to
> consider adjusting the collective of add-on repo's to spread out [...]

I totally agree; I at one time had a patch for a 'minimal SDK' build
which built just the kernel and packages like hostapd that were
required by the kernel/toolchain build, but that didn't seem to garner
a lot of interest.

I've mentioned elsewhere I'd like to see the core concentrate on things
like the kernel and wifi and separate out the 'integration' layer that
creates the collective firmware (for which different groups my have
different visions and goals, but not need to modify the core to achieve
different 'flavours').

Regards,

Daniel

commodo, 2017-05-26:
For stuff that's piling up (PRs & Issues).

Should we start adding a new procedure ?

Something like:
If an Issue or PR has timed out for X months" [ let's say X is 3, but can be whatever ], then
* add a comment to PR or Issue stating "is there anything to be done here ? if not, or no response in Y days, this will be closed"  [ let's say Y is 7, but can be whatever ]
* if the PR or Issue addresses a fix to a package that it's broken [ and times out ], maybe move the package under a "broken" tag, until it's picked up, or can be dropped from packages feed after Z time [ no idea how much Z time would be ]

I'd be happy to help [in the next couple of days] to cleanup the backlog [ if anyone wants/needs help ].
I am foreseeing a bit more time on my hands.

I'd like some thoughts/opinions on this.

EricLuehrsen, 2017-05-26:
In a product engineering setting, the process of pruning feature requests is cruticial. The businesses resources are diluted in just managing the baggage. A request will be done in a fair and reasonable time or the request is canceled forcefully.

diizzyy, 2017-12-10:
I think we can close this now....

closed 2017-12-10 by diizzyy