openwrt/packages #3590: <luasocket> fails tu build

derdigge, 2016-11-28:
On mpc85xx luasocket is fails to build. tried to post it @dev.openwrt.org but Post is "Spam" and my registration is "Spam". Tried several ways to do it but failed. so here is my report.

On a clean checkout:
`git clone git://github.com/openwrt/openwrt.git
./scripts/feeds update -a
./scripts/feeds install -a `

I think there is an "include accident" with lede?

`make package/luasocket/compile V=99
make[1]: Entering directory '/hdd/openwrt/autobuild/dd'
make[2]: Entering directory '/hdd/openwrt/autobuild/dd/package/libs/toolchain'
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/toolchain.default.install
make[2]: Leaving directory '/hdd/openwrt/autobuild/dd/package/libs/toolchain'
make[2]: Entering directory '/hdd/openwrt/autobuild/dd/package/utils/lua'
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install.clean; fi; echo "liblua" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install
if [ -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install.clean ]; then rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install.clean; fi; echo "lua" >> /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/pkginfo/lua.default.install
make[2]: Leaving directory '/hdd/openwrt/autobuild/dd/package/utils/lua'
make[2]: Entering directory '/hdd/openwrt/autobuild/dd/feeds/packages/lang/luasocket'
. /hdd/openwrt/autobuild/dd/include/shell.sh; bzcat /hdd/openwrt/autobuild/dd/dl/luasocket-3.0-rc1-20130909.tar.bz2 | tar -C /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/.. -xf -

Applying ./patches/0001-Add-interface-support.patch using plaintext:
patching file src/if.c
patching file src/if.h
patching file src/luasocket.c
patching file src/makefile
patching file src/options.c
touch /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/.prepared_f5b193f71d22e0a81b22c29fc24551a7
rm -f /hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/stamp/.luasocket_installed
rm -f /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/.configured_*
touch /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/.configured_ynnynyyyy
make -C /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/ LIBDIR="-L/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/usr/lib -L/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/lib -L/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/usr/lib -L/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro" CC="powerpc-openwrt-linux-musl-gcc -Os -pipe -mcpu=8540 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909:luasocket-3.0-rc1-20130909 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include/fortify -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include -std=gnu99" LD="powerpc-openwrt-linux-musl-ld -shared" all
make[3]: Entering directory '/hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909'
make -C src linux
make[4]: Entering directory '/hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/src'
make all-unix PLAT=linux
make[5]: Entering directory '/hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909/src'
powerpc-openwrt-linux-musl-gcc -Os -pipe -mcpu=8540 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909:luasocket-3.0-rc1-20130909 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include/fortify -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include -std=gnu99  -I/usr/include/lua/5.1 -DLUASOCKET_NODEBUG -DLUA_NOCOMPAT_MODULE -DLUASOCKET_API='__attribute__((visibility("default")))' -DUNIX_API='__attribute__((visibility("default")))' -DMIME_API='__attribute__((visibility("default")))' -pedantic -Wall -Wshadow -Wextra -Wimplicit -O2 -ggdb3 -fpic -fvisibility=hidden   -c -o luasocket.o luasocket.c
powerpc-openwrt-linux-musl-gcc -Os -pipe -mcpu=8540 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /hdd/openwrt/autobuild/dd/build_dir/target-powerpc_8540_musl-1.1.15/luasocket-3.0-rc1-20130909:luasocket-3.0-rc1-20130909 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/target-powerpc_8540_musl-1.1.15/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/usr/include -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include/fortify -I/hdd/openwrt/autobuild/dd/staging_dir/toolchain-powerpc_8540_gcc-5.3.0_musl-1.1.15/include -std=gnu99  -I/usr/include/lua/5.1 -DLUASOCKET_NODEBUG -DLUA_NOCOMPAT_MODULE -DLUASOCKET_API='__attribute__((vi [... cut ...]

 - OpenWrt: https://dev.openwrt.org/newticket
 - LEDE: https://bugs.lede-project.org/
 - Most LuCI packages: https://github.com/openwrt/luci/issues

Thanks for your contribution
Please remove this text (before ---) and fill the following template
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWRT/LEDE version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

yousong, 2016-12-12:
Seems to be missing linker flag `-lssp_nonshared`

yousong, 2016-12-12:
Ping luasocket's current maintainer @MikePetullo

karlp, 2017-10-26:
Still failing, but for different reasons now: http://downloads.lede-project.org/snapshots/faillogs/powerpc_8540/packages/luasocket/compile.txt

smartwires, 2017-11-08:
Does anyone have a fix for this issue I am experiencing the same problem?

lpapier, 2018-02-21:
Replacing `ld` by `gcc` fix the problem and should work on other arch (https://ubuntuforums.org/showthread.php?t=352642&p=10100263#post10100263).

```diff
--- a/src/makefile
+++ b/src/makefile
@@ -345,18 +345,18 @@ none:
 all: $(SOCKET_SO) $(MIME_SO)
 
 $(SOCKET_SO): $(SOCKET_OBJS)
-	$(LD) $(SOCKET_OBJS) $(LDFLAGS)$@
+	$(CC) $(SOCKET_OBJS) $(LDFLAGS)$@
 
 $(MIME_SO): $(MIME_OBJS)
-	$(LD) $(MIME_OBJS) $(LDFLAGS)$@
+	$(CC) $(MIME_OBJS) $(LDFLAGS)$@
 
 all-unix: all $(UNIX_SO) $(SERIAL_SO)
 
 $(UNIX_SO): $(UNIX_OBJS)
-	$(LD) $(UNIX_OBJS) $(LDFLAGS)$@
+	$(CC) $(UNIX_OBJS) $(LDFLAGS)$@
 
 $(SERIAL_SO): $(SERIAL_OBJS)
-	$(LD) $(SERIAL_OBJS) $(LDFLAGS)$@
+	$(CC) $(SERIAL_OBJS) $(LDFLAGS)$@
 
 install: 
 	$(INSTALL_DIR) $(INSTALL_TOP_LDIR)
```
[0301-Fix-mpc85xx-linking.patch.txt](https://github.com/openwrt/packages/files/1743507/0301-Fix-mpc85xx-linking.patch.txt)

MikePetullo, 2018-02-22:
Looks okay to me. Can you propose a merge request? Otherwise, I will add this during my next update.

lpapier, 2018-02-22:
@MikePetullo A merge request is available.

dibdot, 2018-02-24:
fix applied!

closed 2018-02-24 by dibdot