openwrt/packages #3595: unbound 1.5.10 crashes because of duplicate local-zone

vollkommenheit, 2016-11-29:
Current unbound - 1.5.10-3 of openwrt crashes repeatedly, if there're duplicate local-zone entries in unbound.conf:

/etc/unbound/unbound.conf

    local-zone: "doubleclick.net" redirect
    local-data: "doubleclick.net A 127.0.0.1"
    local-zone: "doubleclick.net" redirect
    local-data: "doubleclick.net A 127.0.0.1"

logread -f

>Tue Nov 29 10:27:40 2016 daemon.warn unbound: [7709:0] warning: duplicate local-zone
Tue Nov 29 10:27:40 2016 kern.info kernel: [382325.193855]
Tue Nov 29 10:27:40 2016 kern.info kernel: [382325.193855] do_page_fault(): sending SIGSEGV to unbound for invalid read access from 00000028
Tue Nov 29 10:27:40 2016 kern.info kernel: [382325.202662] epc = 77882640 in libc.so[7780a000+91000]
Tue Nov 29 10:27:40 2016 kern.info kernel: [382325.207848] ra  = 77882490 in libc.so[7780a000+91000]
Tue Nov 29 10:27:40 2016 kern.info kernel: [382325.213102]

This does not happen with the latest unbound 1.5.11 from https://github.com/jedisct1/unbound, running on Mac OS X. I did not test whether the latest 1.5.11 has the same issue on openwrt (due to lack of know-how), or whether the stable release of unbound 1.5.10 has the same issue on Mac OS X (due to lack of time).

hnyman, 2016-11-29:
cc @EricLuehrsen

EricLuehrsen, 2016-11-30:
I will attempt to repeat; though I use OpenWrt CC for stability. Any additional info like CPU type might be useful to cascade upstream, if I cant repeat. Its probably is a high level search/sort/duplicate issue running off an array, or dropping the dup' data and leaving the list pointer to nowhere behind, but maybe not.

vollkommenheit, 2016-11-30:
less /proc/cpuinfo

system type		: Atheros AR7161 rev 2
machine			: NETGEAR WNDR3700/WNDR3800/WNDRMAC
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 452.19
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

EricLuehrsen, 2016-11-30:
Issue also confirmed with net/unbound 1.5.10-4 in UCI manual mode with OpenWrt CC 15.5.1 on AR9344r2. No duplicate or similar condition bugs seem to be reported yet. Ticket [1167](https://www.nlnetlabs.nl/bugs-script/show_bug.cgi?id=1167) submitted upstream to [Unbound](https://unbound.net/).

EricLuehrsen, 2016-12-02:
Resolution pending official release upstream. Close okay?

vollkommenheit, 2016-12-02:
Absolutely. Feel free to close. I should have done a search on the commits descriptions in the unbound repository, but since it was only happening on openwrt for me, I thought the root cause might be something related to openwrt.

closed 2016-12-02 by hnyman

vollkommenheit, 2016-12-18:
Unbound 1.6.0 has been released. I hope to see it in openwrt Designated Driver in the near future.
https://www.unbound.net/pipermail/unbound-users/2016-December/004587.html

In addition to the bug fix relevant to this thread, there're a couple of new features I'd love to take advantage of (but can't yet, at least on openwrt).

hnyman, 2016-12-18:
Di you notice #3665 ????

vollkommenheit, 2016-12-18:
Nah, I either did not, or had forgotten about it. Looking forward to 1.6.0.