openwrt/packages #3607: upmpdcli 1.1.3 : crash when Python2 is not installed

schmurtzm, 2016-12-02:
Maintainer:  @ignisf @thess  @yousong  @psycho-nico
Environment: OpenWRT Bleeding Edge, 50013  - kernel 4.4.14

upmpdcli should run normally without Python2 (just radios should not be exposed) but it crash.
With Python2 installed no problem.
No crash problem with the old package (0.11.0)

For example :

---------------------------------------------------------------------------------
root@OpenWrt:/# upmpdcli -h 192.168.0.10 -p 6600 -i wlan0
[  343.002818]
do_page_fault(): sending SIGSEGV to upmpdcli for invalid read access from 2d6f7266
[  343.010628] epc = 2d6f7267 inra  = 7700b62f in libupnpp.so.2.3.0[76fba000+79000]
[  343.017477]
Segmentation fault
---------------------------------------------------------------------------------

Possible solutions to resolve the problem : 
- try to update to new code source from upmpdcli  (current version 1.2.10)
- roll back to old version that don't need Python2 (0.13.1)

Thanks

ignisf, 2016-12-02:
Thank you for the report, will take a look

thess, 2017-01-11:
ping @ignisf - Any update?

ignisf, 2017-01-11:
Ok, now I remember what I forgot to update during the holidays.

schmurtzm, 2017-01-11:
Hi ,
Some remarks :
- May be it's the good timing to update the package to the version 1.2.12 of upmpdcli which supports Google Play Music, Qobuz and Tidal. (and may be it will resolve the initial problem ?).
- It can be nice to indicate in the description of the package that Python is facultative (it just offers more functionalities like streaming services : openhome radios, Google Play Music, Qobuz and Tidal). 
- It can be nice to indicate in the description of the package Which version of python is needed because there are many package and the main python package is really heavy ! 
- By the way, as you're the most active people on these awesome packages, just a little promotion for [SC2MPD](https://www.lesbonscomptes.com/upmpdcli/sc2mpd.html) : this software works with upmpdcli and allows to send sound from PC (Windows/linux/Mac) to an openwrt device with a really good sound and latency. Really much better latency than airplay for example. I asked for the creation of [here](https://github.com/openwrt/packages/issues/2810) and [here](https://forum.openwrt.org/viewtopic.php?id=65393) but I think that nobody see the power of these software...

diizzyy, 2019-01-31:
@thess @hnyman 
Please close, it doesn't crash anymore

closed 2019-01-31 by hnyman