openwrt/packages #3617: dynamic_dns_lucihelper.sh: recent push flooding log

anomeome, 2016-12-04:
Dec  4 10:51:07 bsaedgy uhttpd[3508]: dynamic_dns_lucihelper.sh: command 'get_registered_ip': 'lookup_host' not set
Dec  4 10:51:07 bsaedgy uhttpd[3508]: Usage:
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  dynamic_dns_lucihelper.sh [options] -- command
Dec  4 10:51:07 bsaedgy uhttpd[3508]: Commands:
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  get_local_ip        using given INTERFACE or NETWORK or SCRIPT or URL
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  get_registered_ip   for given FQDN
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  verify_dns          given DNS-SERVER
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  verify_proxy        given PROXY
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  start               start given SECTION
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  reload              force running ddns processes to reload changed configuration
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  restart             restart all ddns processes
Dec  4 10:51:07 bsaedgy uhttpd[3508]: Parameters:
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -6                  => use_ipv6=1          (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -d DNS-SERVER       => dns_server=SERVER[:PORT]
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -f                  => force_ipversion=1   (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -g                  => is_glue=1           (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -i INTERFACE        => ip_interface=INTERFACE; ip_source="interface"
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -l FQDN             => lookup_host=FQDN
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -n NETWORK          => ip_network=NETWORK; ip_source="network"
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -p PROXY            => proxy=[USER:PASS@]PROXY:PORT
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -s SCRIPT           => ip_script=SCRIPT; ip_source="script"
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -t                  => force_dnstcp=1      (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -u URL              => ip_url=URL; ip_source="web"
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -S SECTION          SECTION to start
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -h                  => show this help and exit
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -L                  => use_logfile=1    (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -v LEVEL            => VERBOSE=LEVEL    (default 0)
Dec  4 10:51:07 bsaedgy uhttpd[3508]:  -V                  => show version and exit
...

chris5560, 2016-12-04:
@anomeome
Sorry, but please give me a chance to publish new corresponding luci-app-ddns.
Coming the next 2 or 3 days.

hnyman, 2016-12-05:
I have different output in system log, but apparently due to the same DDNS update. I wrote my comments to the original PR in https://github.com/openwrt/packages/pull/3616#issuecomment-264840413

chris5560, 2016-12-15:
Fixed ??
If Yes please close

anomeome, 2016-12-15:
Apologies, yes good for me.

closed 2016-12-15 by anomeome