openwrt/packages #3618: hd-idle: Setup for multiple hdd?

schovi, 2016-12-04:
Maintainer: @bobafetthotmail 
Environment: Turris omnia

Description: 

Can't find way, how to configure hd-idle to support more than one hdd. I opened issue there with some attempts https://github.com/openwrt/luci/issues/842 Last attempt with removing 'sdX' from section fail too.

/etc/config/hd-idle (I tried with names for section too)
```
config hd-idle
	option idle_time_unit 'minutes'
	option idle_time_interval '10'
	option disk 'sda'
	option enabled '1'

config hd-idle
	option idle_time_unit 'minutes'
	option idle_time_interval '10'
	option disk 'sdb'
	option enabled '1'
```

Result is only one running process `hd-idle -a sda -i 600`

bobafetthotmail, 2016-12-04:
lol nope. :)

`PKG_MAINTAINER:=Lim Guo Wei <limguowei@gmail.com>`

summoning this package's mantainer: @gwlim

schovi, 2016-12-04:
@bobafetthotmail I am dumb. I was looking for last contributor to that file. My bad :)

bobafetthotmail, 2016-12-04:
Hmm, by looking at the source here https://github.com/openwrt/packages/blob/master/utils/hd-idle/files/hd-idle.init

From a very superficial examination, it seems this thing now has code to react to more disks. See the "numdisk" variable.

By looking at the git history and the code you posted in your other issue, it seems you are having issues with the older init script, before dangowrt converted it to procd and added that code.

Might be worth it to try the hd-idle package from OpenWRT trunk or LEDE snapshot.

schovi, 2016-12-04:
I see problem there. I had some old version of init file. I was instantly thinking, that turris omnia is bullet proof, always up-to date router :)
copy & paste this init solve the problem.

closed 2016-12-04 by schovi

bobafetthotmail, 2016-12-04:
If you are on OpenWRT Chaos Calmer you get the old package by default, that's not an error. 

The commit adding this functionality was in September, it is way too new to be in the stable release.