openwrt/packages #3624: unbound - 1.5.10-4 reads from /var/lib/unbound/ by default?

vollkommenheit, 2016-12-07:
This seems a behavior change from 1.5.10-3. After updating to 1.5.10-4, even though /etc/unbound and a bunch of files within, such as /etc/unbound/root.key, got their ownership changed from "unbound" to "root" (I have "username: unbound" configure in /etc/unbound/unbound.conf), unbound now reads  /var/lib/unbound/unbound.conf instead. Is this new behavior intended?

hnyman, 2016-12-07:
cc maintainer @EricLuehrsen

EricLuehrsen, 2016-12-07:
The UID change is an old one, this was done for security 1.5.8 or so ago. named.cache hasnt been used in a longer time. The unbound validator (dnssec) performs RFC5011 for root DSKEY every hour or quicker. This beats up on flash. 

- all of /etc/unbound is root
- all of /etc/unbound is copied to /var/lib/unbound per file heirarchy stds
-- this is like Debian NetworkManager copying/editing isc-dhclient.conf to/in /var/lib/NetworkManager
- UCI will write a different unbound.conf (option)
- /var/lib/unbound is jailed (chroot) from inside Unbound
-- **the read/write jail is in tmpfs, important on embedded devices**
-- Unbound can write to tmpfs all it wants
- UCI optional scripts will copy root.key back to etc on a kinder gentler time scale.

You may use your own conf all you want (uci "manual_conf 1"). If you UID to "unbound", then use var/lib/unbound. The init s ripts make that a copy of etc/unbound. Recommend chroot to var... also for security.

vollkommenheit, 2016-12-07:
I'm fine with going without the non-built-in named.cache (root.hints), and moving to chrooting unbound in /var/lib/unbound. However, a few points are unclear:

The only file that got copied from /etc/unbound to /var/lib/unbound by the unbound 1.5.10-4 upgrade appears to be "root.key". Aside from the aforementioned /etc/unbound/named.cache that got deleted, none of the following (all owned by "unbound:unbound") got copied to /var/lib/unbound:

> unbound.check
unbound.conf
unbound.forward-zone.conf
unbound.local-zone.block.conf
unbound.local-zone.hosts.conf
unbound_control.key
unbound_control.pem
unbound_server.key
unbound_server.pem

Should these above have been copied automatically by the unbound 1.5.10-4 install scripts? The first two files (unbound.check and unbound.conf) exist in /var/lib/unbound, but these apparently came from 1.5.10-4 itself.

What is 
> uci "manual_conf 1"

?
Is this an additional step that users with existing unbound installation (and who want to continue keeping the customization, although from /var/lid/unbound) needs to contend with, post "opkg upgrade unbound", beginning with 1.5.10-4? Prior to that, I never have to deal with "uci" for the sake of running "unbound".

Migrating from /etc/unbound to /var/lib/unbound is a big deal, because users like myself have to be well aware of it, in order to timely update places such as /etc/upgrade.conf to preserve customized configs and files through pkg upgrades. Currently, I have /etc/unbound entered in /etc/upgrade.conf.

EricLuehrsen, 2016-12-08:
The package net/unbound 1.5.10-4 was explicitly intended to introduce UCI to Unbound. This gives more users with less time to learn each program access to a working recursive resolver. The full extent of the content is described in **[README.md](https://github.com/openwrt/packages/blob/master/net/unbound/files/README.md)**. For even easier access, **luci-app-unbound** is also provided in the LuCI repo.

Clearly people have there own configurations and there was no intention to hijack those. The UCI includes an option to disable itself. You can also check **[unbound.uci](https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.uci)** for a complete primer. Anyway one entry:
```
config unbound
	option manual_conf '1'
```
Summary of key problems solved in recent packages:
- Unbound DSKEY RFC5011 is hard on flash in /etc/unbound/root.key 
- DNSSEC needs time; NTP need DNS; and consumer routers have no power off clock
- Many services do not need to be 'root' and should drop privileges after port-bind
- Unbound doesn't bind to new interfaces always so a hot plug event may be needed
- Unbound doesn't manage root.hints like it does root.key
- chown in /etc/ occurs after first boot in init' and quickly expands jffs partition

Summary of behaviors to manage these:
- Copy /etc/unbound to /var/lib/unbound (tmpfs) and have unbound run normally there
  - Create tmpfs working dir and files with UID 'unbound'
  - Jail (chroot) to /var/lib/unbound for security
  - Read/write with wild abandon in tmpfs and no flash harm
  - Occasionally copy root.key back to /etc/unbound (UCI option)
  - Follows standard file system hierarchy and purpose
  - _Debian NetworkManager copying isc-dhclient.conf to /var/lib/NetworkManager_
- UCI only, DNSSEC runs at half strength during boot (time check excluded)
  - Hot plug on NTP update switches to full strength
  - /var/lib/unbound/unbound.time is flag-file for this one time only event
  - Common with lede/packages/network/service/dnsmasq "dnssec no time check"
- Hot plug on iface to force Unbound to rebind
  - UCI generated conf binds to **all** interfaces
  - UCI Access allowed to only subnets with an interface
  - Common with lede/packages/network/service/dnsmasq "local service"
- named.cache or root.hints hopeful per upstream ticket [Unbound ^837](https://nlnetlabs.nl/bugs-script/show_bug.cgi?id=837)

Brief History:
- 1.5.8 Remove named.cache = c0e5d5e (Jun2016)
- 1.5.9 UID 'unbound' = 414eaac (Jun2016)
- 1.5.9 procd init style = 28945ea (Jun2016)
- 1.5.10 /var/lib/unbound = cb56829 (Oct2016), note already package '-3'

Bug at 1.5.10-4
- Not all of /etc/unbound is copied to /var/lib/unbound
- **README.md** needs this redirection instruction for manual conf files

EricLuehrsen, 2016-12-08:
Remind me to never put a github issue in a commit again. Can we cull this noise?

vollkommenheit, 2016-12-08:
It seems /var/lib/unbound/unbound.conf always gets reset to the base version that 1.5.10-4 auto-generates, regardless of "uci set unbound.manual_conf=1" or "uci set unbound.manual_conf=0". This behavior jives with what unbound.uci says:

> option manual_conf '0'
    Boolean. Skip all this UCI nonsense. Manually edit the
    configuration. Make changes to /etc/unbound/unbound.conf.

So color me confused--if customized unbound.conf is supposed to remain /etc/unbound/unbound.conf, then what is the point of needing to copy everything in /etc/unbound (including /etc/unbound/unbound.conf, which gets reinitialized after each "/etc/init.d/unbound restart") to /var/lib/unbound? 

I don't pretend to understand how chroot works, but how is the stated goal of jailing unbound in /var/lib/unbound/ going to work, when customized config is going to remain in /etc/unbound? What should my customized /etc/unbound/unbound.conf have, for the "directory" and "chroot" directives?

>   directory: "/etc/unbound"
  chroot: "/etc/unbound"

or

>  directory: "/var/lib/unbound"
  chroot: "/var/lib/unbound"

or nothing?

Also, everytime I update custom include files such as "unbound.local-zone.block.conf", do I need to worry about pushing a copy to both /etc/unbound and /var/lib/unbound?

Currently, after "uci set unbound.manual_conf=1", it seems unbound 1.5.10-4 is honoring the configs of /etc/unbound/unbound.conf, IF I start unbound manually from the CLI. However, if started with /etc/init.d/unbound, it seems to only read the barebone self-generated /var/lib/unbound/unbound.conf.

In summary: Can a user have the benefits of BOTH customized unbound (using /etc/unbound) AND jailing in /var/lib/unbound + keeping root.key writes on tmpfs (using /var/include/unbound)?

EricLuehrsen, 2016-12-08:
Unbound internal jail (chroot) only works on one directory, so it all has to be there. `procd` jails are functional but create a lot of mount point noise for the OS to deal with. The whole idea is that /etc/unbound is cloned to /var/lib/unbound and life goes on. Anyway I just tested this.
```
$ vi /etc/config/unbound
:  config unbound
:    option manual_conf '1'
$ /etc/init.d/unbound restart
$ more /var/lib/unbound/unbound.conf
```
Also, Unbound was use the settings I put there to probe using unbound-control.

vollkommenheit, 2016-12-08:
The auto-generated /var/lib/unbound/unbound.conf I was referring to seems to be created by /usr/lib/unbound/unbound.sh / UCI:

> # /var/lib/unbound/unbound.conf generated by UCI Wed Dec  7 23:06:32 EST 2016
>
> remote-control:
  control-enable: no
>
> server:
  username: unbound
  num-threads: 1
  msg-cache-slabs: 1
  rrset-cache-slabs: 1
  infra-cache-slabs: 1
  key-cache-slabs: 1
> ...
> edns-buffer-size: 1280

grep 1280 /usr/lib/unbound/unbound.sh
> UNBOUND_N_EDNS_SIZE=1280
  config_get UNBOUND_N_EDNS_SIZE "$cfg" edns_size 1280
    UNBOUND_N_EDNS_SIZE=1280

EricLuehrsen, 2016-12-08:
When using the command line UCI, did you commit? Did you more/less /etc/config/unbound to check?

```
root@OpenWrt:~# uci set unbound.manual_conf=1
root@OpenWrt:~# uci commit unbound
root@OpenWrt:~# /etc/init.d/unbound restart
```

vollkommenheit, 2016-12-08:
Yeah, I did at least one commit

uci get unbound.manual_conf
uci set unbound.manual_conf=1
uci commit unbound
uci get unbound.manual_conf

EricLuehrsen, 2016-12-08:
I usually `$ vi /etc/config/...` so I didn't realize how UCI was actually handling this. Apparently an unnamed and uninstantiated section still is a section. It has a really odd syntax.

```
root@OpenWrt:~# uci get unbound.@unbound[0].manual_conf
0
root@OpenWrt:~# uci set unbound.@unbound[0].manual_conf=1
root@OpenWrt:~# uci commit unbound
root@OpenWrt:~# uci get unbound.@unbound[0].manual_conf
1
root@OpenWrt:~# 
```

vollkommenheit, 2016-12-08:
So, adding "option manual_conf '1'" to /etc/config/unbound would do the trick?

EricLuehrsen, 2016-12-08:
You need the "config unbound" section and the option within the section.

/etc/config/unbound  
```
config unbound
  option manual_conf '1'
```

vollkommenheit, 2016-12-08:
Yup. Actually, it worked even with two conflicting entries:

> option manual_conf '0'
option manual_conf '1'

EricLuehrsen, 2016-12-08:
Glad to help and sorry for the confusion. Actually it is the syntax for unnamed sections in UCI CLI that got both of us. The `config unbound` becomes `@unbound[0]`. The format is `file.section.option=value` and unnamed sections are `file.@type[0].option=value`
```
$ uci show unbound
...
unbound.@unbound[0].manual_conf=1
...
```
Anyway, you did find a bug and I have a pull request to fix it. Please look it over.

EricLuehrsen, 2016-12-08:
@hnyman - I see you merged #3631, we can close this issue. A new issue if 1.5.10-5 causes difficulty elsewhere.

vollkommenheit, 2016-12-08:
Thanks. 

Eeek. As I suspected, I don't see 1.5.10-5 after "opkg update". So I can't test it, because I don't have the toolchains/compiler set up to compile for openwrt ipkg from source code. I do see the part of the changes that copies everything from /etc/unbound to /var/lib/unbound.

At this point, my concerns are more around keeping unbound configs updated going forward, through future unbound upgrades:
1. IIRC, I see remarks in the upstream bug filed with NLlabs (concerning root.key thrashing ROM) confirming what I suspected/feared, that stuff in /tmp (tmpfs) are lost after reboots just like on desktop OS such as macOS. So does that mean I'd need to ensure my custom files get copied from /etc/unbound to /var/lib/unbound after every reboot, or is this automation provided by openwrt or unbound via UCI or otherwise? 

2. I see /etc/config/unbound has "option root_age '28'" that seems to take care of copying "root.key" from /var/lib/unbound back to /etc/unbound. Well, I was hoping I could ditch /etc/unbound entirely, in favor of /var/lib/unbound. So something that would automatically and/or on-demand copy everything from /var/lib/unbound back to /etc/unbound would seem ideal for me.

Basically, because I want to reap the benefits of the UCI/tmpfs-based unbound, I need to figure out my new post-upgrade/post-reboot workflow for it, if /var/lib/unbound does need to be rebuilt after each reboot.

EricLuehrsen, 2016-12-08:
The scripts handle it all. The intent is keep /etc/[application] the way it should be, per standards. The intent is to copy to /var/lib/[application] to handle modifications, lock out, jail and such, also standard. (Debian NetworkManager is my best example of a long standing and common user feature package that does this).

Write your files in /etc/. Know they are copied to /var/ every init.d-start. Set chroot and working dir options to /var/. The rest should just work.

If you dont username:unbound (:root) or you dont jail (no chroot:) then unbound doesnt care. You can access both /etc/ and /var/ and  everything ... which is the security point. Its Unbounds single path chroot that makes this necessary. Its the messy mounting behavior that keeps `procd` from being an attractive alternative.

Builbot should batch up a job soon now.

vollkommenheit, 2016-12-18:
I think this issue can be closed (I know I can close it myself). I've been running unbound 1.5.10-5 and 1.5.10-6 problem free.

closed 2016-12-31 by hnyman