openwrt/packages #3629: softethervpn - error compile/build to apm821xx/ppc (WNDR4700)

build000, 2016-12-07:
LEDE: CURRENT-r2443-ee55a19_powerpc_464fp

I indicate with no doubt in PR that PPC with a version of the Makefile as it is, simply has no chance to properly compile the main cryptographic library in this program. ( https://github.com/openwrt/packages/pull/3498#issuecomment-259507541 )

make -j8 BUILD_LOG=1
cat logs/package/feeds/softethervpn/compile.txt

```
Applying ./patches/103-noeucjp.patch using plaintext: 
patching file src/Mayaqua/Internat.c

Applying ./patches/105-nosslv3.patch using plaintext: 
patching file src/Mayaqua/Network.c
make[4]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613'
rm -f tmp/objs/Mayaqua/Cfg.o tmp/objs/Mayaqua/Encrypt.o tmp/objs/Mayaqua/FileIO.o tmp/objs/Mayaqua/Internat.o tmp/objs/Mayaqua/Kernel.o tmp/objs/Mayaqua/Mayaqua.o tmp/objs/Mayaqua/Memory.o tmp/objs/Mayaqua/Microsoft.o tmp/objs/Mayaqua/Network.o tmp/objs/Mayaqua/Object.o tmp/objs/Mayaqua/OS.o tmp/objs/Mayaqua/Pack.o tmp/objs/Mayaqua/Secure.o tmp/objs/Mayaqua/Str.o tmp/objs/Mayaqua/Table.o tmp/objs/Mayaqua/TcpIp.o tmp/objs/Mayaqua/Tick64.o tmp/objs/Mayaqua/Tracking.o tmp/objs/Mayaqua/Unix.o tmp/objs/Mayaqua/Win32.o
rm -f tmp/objs/Cedar/Account.o tmp/objs/Cedar/Admin.o tmp/objs/Cedar/AzureClient.o tmp/objs/Cedar/AzureServer.o tmp/objs/Cedar/Bridge.o tmp/objs/Cedar/BridgeUnix.o tmp/objs/Cedar/BridgeWin32.o tmp/objs/Cedar/Cedar.o tmp/objs/Cedar/CedarPch.o tmp/objs/Cedar/Client.o tmp/objs/Cedar/CM.o tmp/objs/Cedar/Command.o tmp/objs/Cedar/Connection.o tmp/objs/Cedar/Console.o tmp/objs/Cedar/Database.o tmp/objs/Cedar/DDNS.o tmp/objs/Cedar/EM.o tmp/objs/Cedar/EtherLog.o tmp/objs/Cedar/Hub.o tmp/objs/Cedar/Interop_OpenVPN.o tmp/objs/Cedar/Interop_SSTP.o tmp/objs/Cedar/IPsec.o tmp/objs/Cedar/IPsec_EtherIP.o tmp/objs/Cedar/IPsec_IKE.o tmp/objs/Cedar/IPsec_IkePacket.o tmp/objs/Cedar/IPsec_IPC.o tmp/objs/Cedar/IPsec_L2TP.o tmp/objs/Cedar/IPsec_PPP.o tmp/objs/Cedar/IPsec_Win7.o tmp/objs/Cedar/Layer3.o tmp/objs/Cedar/Link.o tmp/objs/Cedar/Listener.o tmp/objs/Cedar/Logging.o tmp/objs/Cedar/Nat.o tmp/objs/Cedar/NativeStack.o tmp/objs/Cedar/NM.o tmp/objs/Cedar/NullLan.o tmp/objs/Cedar/Protocol.o tmp/objs/Cedar/Radius.o tmp/objs/Cedar/Remote.o tmp/objs/Cedar/Sam.o tmp/objs/Cedar/SecureInfo.o tmp/objs/Cedar/SecureNAT.o tmp/objs/Cedar/SeLowUser.o tmp/objs/Cedar/Server.o tmp/objs/Cedar/Session.o tmp/objs/Cedar/SM.o tmp/objs/Cedar/SW.o tmp/objs/Cedar/UdpAccel.o tmp/objs/Cedar/UT.o tmp/objs/Cedar/VG.o tmp/objs/Cedar/Virtual.o tmp/objs/Cedar/VLan.o tmp/objs/Cedar/VLanUnix.o tmp/objs/Cedar/VLanWin32.o tmp/objs/Cedar/WaterMark.o tmp/objs/Cedar/WebUI.o tmp/objs/Cedar/WinUi.o tmp/objs/Cedar/Wpc.o
rm -f tmp/objs/vpnserver.o
rm -f tmp/as/vpnserver.a
rm -f bin/vpnserver/vpnserver
rm -f bin/vpnserver/hamcore.se2
rm -f tmp/objs/vpnclient.o
rm -f tmp/as/vpnclient.a
rm -f bin/vpnclient/vpnclient
rm -f bin/vpnclient/hamcore.se2
rm -f tmp/objs/vpnbridge.o
rm -f tmp/as/vpnbridge.a
rm -f bin/vpnbridge/vpnbridge
rm -f bin/vpnbridge/hamcore.se2
rm -f tmp/objs/vpncmd.o
rm -f tmp/as/vpncmd.a
rm -f bin/vpncmd/vpncmd
rm -f bin/vpncmd/hamcore.se2
rm -f tmp/hamcorebuilder
rm -f src/bin/BuiltHamcoreFiles/unix/hamcore.se2
make[4]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613'
make[4]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613'
cc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Cfg.c -o tmp/objs/Mayaqua/Cfg.o
In file included from src/Mayaqua/Cfg.c:125:0:
./src/Mayaqua/Mayaqua.h:285:31: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
                               ^
compilation terminated.
Makefile:64: recipe for target 'tmp/objs/Mayaqua/Cfg.o' failed
make[4]: *** [tmp/objs/Mayaqua/Cfg.o] Error 1
make[4]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613'
Makefile:131: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.21-9613/.built] Error 2
```

fededim, 2016-12-07:
Uhm well I forgot about this, libreadline-dev must be installed on build machine.

build000, 2016-12-07:
no "host-compile" ???

fededim, 2016-12-07:
No host-compile in makefile if you mean that

build000, 2016-12-07:
I'll check, but frankly I doubt whether readline.h (quite critical library) system / platform-type intel 64bit architecture is suitable for ppc-32bit - the effect may be such that they will be too big throws when encryption and router just starts choke or restart ...

fededim, 2016-12-07:
See https://github.com/el1n/OpenWRT-package-softether/issues/9, it is needed both on build host and toolchain

hnyman, 2016-12-07:
Buildbot fails in build, too:
http://downloads.lede-project.org/snapshots/faillogs/mips64el_mips64/packages/softethervpn/compile.txt

```
In file included from src/Mayaqua/Cfg.c:125:0:
./src/Mayaqua/Mayaqua.h:285:31: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
                               ^
compilation terminated.
Makefile:64: recipe for target 'tmp/objs/Mayaqua/Cfg.o' failed
make[4]: *** [tmp/objs/Mayaqua/Cfg.o] Error 1

```

build000, 2016-12-07:
@hnyman - solved problem https://github.com/openwrt/packages/issues/3629#issuecomment-265477532 ( "for new:" ) - However, it seems to me that this is not the correct approach to the problem and ultimately can be problems with operating the application ...

fededim, 2016-12-07:
>Question:
>
>make[4]: warning:  Clock skew detected.  Your build may be incomplete.
>???

Never noticed, but from http://stackoverflow.com/questions/3824500/compiling-c-on-remote-linux-machine-clock-skew-detected-warning

>That message is usually an indication that some of the files have modification times later than the >current system time. Since make decides which files to compile when performing an incremental build >by checking if a source files has been modified more recently than its object file, this situation can >cause unnecessary files to be built, or worse, necessary files to not be built.
>
>However, if you are building from scratch (not doing an incremental build) you can likely ignore this >warning without consequence.

It could make sense because in the makefile I change on purpose the modification date in future of the generated resource hamcore archive (hamcore.se2 file) through the host compiled hamcorebuild tool in order to avoid cross-compiling the hamcorebuild tool (it must run on host). I will test the .ipk generated if they work we can forget about the warning.

build000, 2016-12-07:
Overall, in my humble opinion, PPC will not work with such a compilation (probably incomplete as the show logs and based on inadequate .h files).

hnyman, 2016-12-07:
> solved problem https://github.com/openwrt/packages/issues/3629#issuecomment-265477532 ( "for new:" ) - However, it seems to me that this is not the correct approach to the problem and ultimately can be problems with operating the application ...

Well, manually installing something to the buildhost is not a real solution. The package should define its host build dependencies etc. in Makefile, so that all necessary tools are present for the compilation.

Several packages define libreadline both as PKG_BUILD_DEPENDS and as DEPENDS. E.g.
https://github.com/openwrt/packages/blob/master/net/atftp/Makefile
https://github.com/openwrt/packages/blob/master/libs/sqlite3/Makefile

Something like that might be the actual solution.

fededim, 2016-12-07:
Well I believed that PKG_BUILD_DEPENDS configured packages only on toolchain, but not also on the build host. I can add PKG_DEPENDS but I do believe that without libreadline on buildhost it won't compile.

build000, 2016-12-07:
Makefile:
```
(...)
 DEPENDS:=+libpthread +librt +libreadline +libopenssl +libncurses +kmod-tun +zlib $(ICONV_DEPENDS)
(...)
```
... after already.

fededim, 2016-12-07:
Ehm I did not understand...what after already ?!?

build000, 2016-12-07:
++libreadline

fededim, 2016-12-07:
Well take your time and tell me more precisely what you would like to change in the Makefile

hnyman, 2016-12-07:
> I can add PKG_BUILD_DEPENDS if we want to try but I do believe that without libreadline on buildhost it won't compile.

There is also HOST_BUILD_DEPENDS but I am not sure if that would work as you have avoided having a host build section in Makefile. You likely need to juggle around a bit and test how to get the needed library installed to the host. Might be that you need to create a new rather empty host build section just to get the libreadline installed into host before the actual build. 
https://wiki.openwrt.org/doc/devel/packages#host_tools_required

(hopefully somebody more experienced with host tools can provide a definite answer)

@build000 
Plain +libreadline in DEPENDS is not enough...

fededim, 2016-12-07:
>@build000
Plain +libreadline in DEPENDS is not enough...

Why ?

hnyman, 2016-12-07:
> > Plain +libreadline in DEPENDS is not enough...
> Why ?

Based on your own comment:
> libreadline-dev must be installed on build machine because softether compiles (before crosscompiling) an archiver tool called hamcorebuilder which needs it.

If you really need libreadline on the build host then I think that you need something more than DEPENDS, which is just about having the cross-compiled library package available in the final firmware.

fededim, 2016-12-07:
Well it is enough if libreadline is installed on build server

hnyman, 2016-12-07:
> Well it is enough if libreadline is installed on build server

But that is just the point: libreadline is not a prerequisite for Openwrt & LEDE, and it is not installed e.g. in the buildbots. So, the package Makefile needs to make sure that the host version of it gets built and gets installed to the host tools used in build.

fededim, 2016-12-07:
> There is also HOST_BUILD_DEPENDS but I am not sure if that would work as you have avoided having a host build section in Makefile. You likely need to juggle around a bit and test how to get the [...]
https://wiki.openwrt.org/doc/devel/packages#host_tools_required

>(hopefully somebody more experienced with host tools can provide a definite answer)

Ok so I was right about PKG_BUILD_DEPENDS. Never heard of HOST_BUILD_DEPENDS. Can anyone tell me about any packages using HOST_BUILD_DEPENDS ? Just to get an idea...the examples section in the documentation of Openwrt is missing for this hidden HOST_BUILD_DEPENDS.

hnyman, 2016-12-07:
>  Can anyone tell me about any packages using HOST_BUILD_DEPENDS ? 

Well, you can use Github search to search for that in this repo...

One example from here: https://github.com/openwrt/packages/blob/265f9595ed5fc57c867e5ef0e4de7bdbed727a98/lang/python-cffi/Makefile
One from Openwrt: https://github.com/openwrt/openwrt/blob/5ce38c486c525b28d9784f83e839537aa0149123/package/utils/bsdiff/Makefile

You might be able to get it done just by adding rather empty host section definitions (like in Wiki) and having libreadline/host as the item.

(The whole make process of this package might actually be simpler (or more clear) if you used the host/compile section for the first step, instead of doing it all in the cross-compile compile section.)

fededim, 2016-12-07:
>You might be able to get it done just by adding rather empty host section definitions (like in Wiki) and having libreadline/host as the item.

Ok I'll try thanks

hnyman, 2016-12-07:
I just tried that in my own build system and got 
`make[1]: *** No rule to make target 'package/libs/libreadline/host/compile'`
So it might not be that easy :-(

And as you use direct make commands in Makefile, you will probably pass the correct library directories etc. in any case.

@jow- & others
 any hints for @fededim how to solve this? I am out of my league with host tools

jow-, 2016-12-07:
@hnyman - I checked, there currently is no host build of `libreadline` available. Instead of packaging that I would suggest to patch `softethervpn` to not require it. I doubt that the host build parts of it actually require readline runtime functionality.

jow-, 2016-12-07:
As expected, the libreadline is not actually needed by the hamcorebuilder binary so we can stub it to make the linker happy while avoid the need to actually host-build a full libreadline.

While I checked the Makefile I also took a stab at converting it to use host-build.mk for the hamcore.se2 build phase: https://gist.github.com/jow-/7960548998fec72e4a7df7f06dc872e9

hnyman, 2016-12-08:
Thanks @jow- 
I tested your Makefile and softethervpn compiles ok with it. But I am not using the package, so maybe @fededim and @build000 could test the VPN compiled from https://gist.githubusercontent.com/jow-/7960548998fec72e4a7df7f06dc872e9/raw/5cba0984fe93055c78b2eb448e4bcd4930e1736f/Makefile and then you could commit it to the repo to fix things.

fededim, 2016-12-09:
Ok I'll try this makefile, but I'll need some time

hnyman, 2016-12-13:
> Ok I'll test this makefile, but I'll need some time

Have you been able to test jow's Makefile?

Currently the package is broken as it does not compile without extra installations, so it would be great to commit the fix.

build000, 2016-12-13:
@hnyman 
Well - first, please do commit and then will test a new solution - such tests outside the main branch is not very reliable - too many variables and possible, mutual impilkacji to finally acknowledge the commit is "broken" or not.
Anyway, with this application we talk about diametrically return of 180 degrees in terms of methodology to build, so no difference at this time (where the application is still with the status of "broken") or the latest Makefile of jow proves once again "broken" or not ...
Among other things, therefore, I refrained so far in testing this application - as I mentioned too many possible variables ...

fededim, 2016-12-13:
I can test it only after xmas.

hnyman, 2016-12-22:
@jow- 
Having a buildable version would be great in the light of the possible branching ahead. Currently softethervpn does not compile for any platform by the buildbot build machines. You should maybe commit your version into the repo in any case.

fededim, 2016-12-29:
I tested it, but the build does not work. It builds correctly on the host generating the hamcorebuilder binary and the hamcore.se2 archive, but then it exits the compilation without crosscompiling. Any idea why ?

fededim, 2016-12-31:
False alarm :) The compilation was fine, I was stopping it just after I saw it entering and exiting immediately the softether build directory.  If you make the compilation continue, it goes back again to softether folder later and compiles everything (has anybody an idea of this strange behaviour ?!?). Anyway I made a new PR #3718 and updated also softether to last version.

build000, 2016-12-31:
@fededim 
last your commit/update ( https://github.com/openwrt/packages/commit/b13e3f8fd27a387b08fce7981b8032e46e3830b7 ) build/compile error
make -j8 BUILD_LOG=1
cat logs/package/feeds/packages/softethervpn/compile.txt:
```

Applying ./patches/103-noeucjp.patch using plaintext: 
patching file src/Mayaqua/Internat.c
make[4]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634'
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Cfg.c -o tmp/objs/Mayaqua/Cfg.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Cfg.c:125:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Encrypt.c -o tmp/objs/Mayaqua/Encrypt.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Encrypt.c:146:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/FileIO.c -o tmp/objs/Mayaqua/FileIO.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/FileIO.c:123:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Internat.c -o tmp/objs/Mayaqua/Internat.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Internat.c:123:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Kernel.c -o tmp/objs/Mayaqua/Kernel.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Kernel.c:125:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Mayaqua.c -o tmp/objs/Mayaqua/Mayaqua.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Mayaqua.c:124:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Memory.c -o tmp/objs/Mayaqua/Memory.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Memory.c:124:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
[... cut ...]
src/makefiles/linux_32bit.mak:318: recipe for target 'tmp/hamcorebuilder' failed
make[4]: *** [tmp/hamcorebuilder] Error 1
make[4]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634'
Makefile:151: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634/.built' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634/.built] Error 2
```

hnyman, 2016-12-31:
I think that the reason may be that PR #3718 also deleted the patch 105-nosslv3.patch that disabled SSLv3. That should likely be restored/updated.

(sounds like a similar SSLv3 issue as with emailrelay in  #3185 )

fededim, 2016-12-31:
Yes I removed the patch because the new version of softether introduced by itself defines to skip SSL3, if you see the Makefile now I set the option -DSSL_OP_NO_SSLv3...on my linux I tested it and it compiled successfully, but I fear it does not recompile openssl (I tested it with SSL3 support and then without)....let me do a check

hnyman, 2016-12-31:
I looked briefly at the source code at it looks like they changed a code a bit and moved part of the SSLv3 behind that define you mentioned, but sadly left one SSLv3_method call outside the defines: https://github.com/SoftEtherVPN/SoftEtherVPN/blob/master/src/Mayaqua/Network.c#L13018

fededim, 2016-12-31:
Yes, you are right. I though that the new defines were sufficient since softether compiled successfully on my system (probably I have some problems with openssl not being rebuilt). I made a new PR #3724, this time it should work.

closed 2016-12-31 by hnyman

build000, 2016-12-31:
ok - in this time problem fixed ( https://github.com/openwrt/packages/commit/07052a6b9c4624588c0aa324e5820165dd392c18 )
```
Applying ./patches/103-noeucjp.patch using plaintext: 
patching file src/Mayaqua/Internat.c

Applying ./patches/105-nossl3.patch using plaintext: 
patching file src/Mayaqua/Network.c
make[4]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634'
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Cfg.c -o tmp/objs/Mayaqua/Cfg.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Cfg.c:125:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Encrypt.c -o tmp/objs/Mayaqua/Encrypt.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Encrypt.c:146:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/FileIO.c -o tmp/objs/Mayaqua/FileIO.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/FileIO.c:123:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Internat.c -o tmp/objs/Mayaqua/Internat.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Internat.c:123:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Kernel.c -o tmp/objs/Mayaqua/Kernel.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Kernel.c:125:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Mayaqua.c -o tmp/objs/Mayaqua/Mayaqua.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Mayaqua.c:124:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^
powerpc-openwrt-linux-musl-gcc -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char -c src/Mayaqua/Memory.c -o tmp/objs/Mayaqua/Memory.o
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from ./src/Mayaqua/Mayaqua.h:271:0,
                 from src/Mayaqua/Memory.c:124:
/home/sj/Pulpit/Projekty/WNDR4700/LEDE/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
[... cut ...]
rstrip.sh: /home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634/ipkg-powerpc_464fp/softethervpn/usr/libexec/softethervpn/vpnclient: executable
Packaged contents of /home/sj/Pulpit/Projekty/WNDR4700/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/v4.22-9634/ipkg-powerpc_464fp/softethervpn into /home/sj/Pulpit/Projekty/WNDR4700/LEDE/bin/packages/powerpc_464fp/packages/softethervpn_4.22-9634-1_powerpc_464fp.ipk
```
I close this issue - gdby revealed something new in relation to the new method of compilation, and in particular re- garding the action/practical use let you know in this thread.

fededim, 2017-01-01:
Well test it for some days, but then please close this issue. Should there be any running issue, you shuold reopen a new one