openwrt/packages #3643: munin-lite: fetch_: not found

Strykar, 2016-12-10:
Unable to fetch plugin data from munin-lite-node

```
$ uname -a
Linux apu2 4.4.27 #0 SMP Thu Oct 27 13:29:36 2016 x86_64 GNU/Linux

$ telnet FQDN 4949
Trying xxx.xxx.82.204...
Connected to FQDN.
Escape character is '^]'.
munin node at FQDN
list
df cpu if_pppoe_wan if_wlan0 if_eth2 if_eth1 if_br_lan if_eth0 if_err_pppoe_wan if_err_wlan0 if_err_eth2 if_err_eth1 if_err_br_lan if_err_eth0 load memory processes swap uptime interrupts irqstats ntpdate
fetch
/usr/sbin/munin-node: line 660: fetch_: not found
.
munin node at FQDN
fetch_df
Unknown command. Try list, nodes, config, fetch, version or quit
fetch_load
Unknown command. Try list, nodes, config, fetch, version or quit
version
munins node on FQDN version: 1.0.4 (munin-lite)
nodes
FQDN
.
config
/usr/sbin/munin-node: line 660: config_: not found
.
quit
Connection closed by foreign host.

$ grep fetch /usr/sbin/munin-node
fetch_df() {
fetch_cpu() {
fetch_if() {
fetch_if_err() {
fetch_load() {
    echo "swap_cache.info A piece of memory that keeps track of pages that have been fetched from swap but not yet been modified."
fetch_memory() {
fetch_processes() {
fetch_swap() {
fetch_netstat() {
fetch_uptime() {
fetch_interrupts() {
fetch_irqstats() {
fetch_ntpdate() {
do_fetch() {
    fetch_$1
      eval "fetch_if_${INTERRES}() { fetch_if $INTER $@; };"
      eval "fetch_if_err_${INTERRES}() { fetch_if_err $INTER $@; };"
        eval "fetch_${MYPLUGINNAME}() { $MYPLUGIN ; };"
FUNCTIONS="list nodes config fetch version quit"
```

yousong, 2016-12-12:
@Strykar please follow the template and at least `@` the maintainer, otherwise he/she may never notice the existence of your issue here.

jmccrohan, 2017-01-10:
Hi @Strykar ,

I can't seem to reproduce this.
Can you provide more information about your configuration? Are you using a self compiled image or have you downloaded the opkg file?

Thanks,
Jon

diizzyy, 2017-06-09:
@hnyman @thess 
Please close due to reporter timeout.

closed 2017-06-09 by hnyman