openwrt/packages #3645: dns-scripts: Hotplug script block the ifup procedure

hatogit, 2016-12-11:
Maintainer: Christian Schoenebeck <christian.schoenebeck@gmail.com>
Environment: 
router:WNDR3800
OpenWRT version:LuCI Master (git-16.345.61910-ac5bd8c) / OpenWrt Designated Driver 50017

Description:
If for some reason, the ddns update failed, then the script will wait there, block the finishing of the hotplug event. In my setup, because the event is actually caused (called) by mwan3 script, and before the hotplug even finished, the mwan3 setup cannot finish. Before the mwan3 setup finishes, there's no internet connection (also both wan connections and the vpn connections are all ready) thus the ddns update always fail. Before commit 1c20dcb there was no such issue.
Let the dynamic_dns_updater.sh in the hotplug script (95-ddns) run in background seems to fix this. And I think "-v 0" should also be added to that command there to silent the screen output in the case I mentioned above.

Thanks.

hnyman, 2016-12-11:
cc maintainer: @chris5560

chris5560, 2016-12-11:
@hatogit
there where already 3 patches published and merged.
your problem should already be fixed.
please give them a try.

hatogit, 2016-12-12:
Hi, chris
Thanks for you work.
I'm using ddns-script 2.7.6-3. The only change 2.7.6-4 introduced was about the result of changeip.com (I do use changeip). But in my case, the internet connection is "not ready" from the router's traffic point of view, the ddns-script can't even get the registered the IP address of the domain. So it would keep trying which would hold the hotplug event. But mwan3 script depends on that even to end to actually make the internet connection available.
And again, letting the dynamic_dns_updater.sh in the hotplug script (95-ddns) run in background (just add a "&" at the end of that line) seems fix this. And I think this is also how the old version works (although in the hotplug script, the command didn't work in background, but the actual updating function it called was actually run in the background). I'm not an expert on programming. I could be wrong.

PR #3651 (2016-12-12): ddns-scripts: set default verbose=0 so scripts run in background
Maintainer: me
Compile tested:
Run tested: LEDE r2438-a245772

Description:
fix #3645 and https://github.com/lede-project/source/pull/596#issuecomment-266395456
commit 13d1169d (2016-12-12): ddns-scripts: set default verbose=0 so scripts run in background

fix #3645 and https://github.com/lede-project/source/pull/596#issuecomment-266395456
closed 2016-12-12 by hnyman